earlephilhower/arduino-pico #1191: Add Bluetooth Library for Pico W

maxgerhardt, 2023-02-15:
When the upgrade to SDK 1.5 is completed per https://github.com/earlephilhower/arduino-pico/pull/1167 we can think about adding an Arduino library exposing the Bluetooth stack for the CYW43438 in an Arduino-friendly (or even API compatible with some other popular library?) way.

earlephilhower, 2023-02-15:
I'm a newcomer to Bluetooth, so if you have any suggested libraries or APIs that would be very helpful.

sanjay900, 2023-02-16:
I know the SDK is using https://github.com/bluekitchen/btstack, and you can get that working on the sdk15 branch pretty easily with a few tweaks, but i wonder if there is anything a little more friendly around, maybe something easier to use was written for the ESP32?

elral, 2023-02-26:
I have written a bluetooth (BLE) application for an ESP32 and NRF52840.
Both API's are usefull, but the Adafruit Bluefruit52 lib was some kind better and had little more functions.
Without thema a bluetooth application would be to complex for me...

earlephilhower, 2023-02-28:
Fixed by #1167 .  Add'l libraries will come as they come...

closed 2023-02-28 by earlephilhower