earlephilhower/arduino-pico #1201: Possible double byte flip with SPI LSB first on reads?

caternuson, 2023-02-17:
LSBFIRST with SPI is rare, but the PN532 is an example:
https://www.adafruit.com/product/364

Ran into this while working on the Adafruit Arduino Library:
https://github.com/adafruit/Adafruit-PN532

Not sure how to provide a simple non-PN532 test sketch for testing locally. But for the PN532 case, can see it in the return of "status ready" query, which is a SPI write of 0x02 (status ready request) followed by a read, with 0x01 indicating "ready". Here's a scope trace of that transaction:
[image]

The SPI analyzer configuration for ref:
[image]

The 0x02 is being clocked out as expected on MOSI (i.e., writing works as expected).
The 0x01 is being returned as expected by the PN532 on MISO - this is what is being read back in.
However, printing the actual return value in Arduino sketch after the transfer shows `0x80` instead of `0x01`.

I think the double flip is resulting from this line:
https://github.com/earlephilhower/arduino-pico/blob/97e787e48a51da1ebe12aa4af6556c49d0227e10/libraries/SPI/src/SPI.cpp#L138
since it has already been flipped in the `transfer()` function here: 
https://github.com/earlephilhower/arduino-pico/blob/97e787e48a51da1ebe12aa4af6556c49d0227e10/libraries/SPI/src/SPI.cpp#L114
**So writing is working as expected, but the reads come back byte flipped from expected.**

I can comment out line 138 in `SPI.cpp` and code runs as expected.

earlephilhower, 2023-02-17:
Looks like a good catch.  Want to make a RP with the fix?

caternuson, 2023-02-17:
Thanks for responding. Sure, happy to make a PR for above. Do you agree it's just a removal of line 138? Just want to make sure that won't break something else I may have missed. This was mainly found by code inspection of `SPI.cpp` and then the PN532 test to "confirm".

earlephilhower, 2023-02-17:
Yeah, I think it's just dropping line 138.  It's pretty obvious that the bitreverse is being done in the byte-by-byte bit so this 2nd flip is redundant and wrong.  Your testing w/real HW is just icing on the cake. 👍

commit ba413bb7 (2023-02-17): Remove extra SPI byte flip (#1202)

Fixes #1201
closed 2023-02-17 by earlephilhower