earlephilhower/arduino-pico #1203: Serial1 and Serial2 does not work in multicore.

chester2400, 2023-02-17:
I have a program that one core dose the computing and second core takes care of display with TFT_eSPI library.
In loop0 i have a serial1 sending data to other device and there is no signal on the scope. I have to put core1 to sleep to see the outgoing data. I tryed using SepialPIO there is no diferance.  The SPI on second core makes the Serial1 have no signal. Without SPI it works ok.

earlephilhower, 2023-02-17:
Please make and post a small MCVE sketch that reproduces the failure w/o the extra 3rd party libraries.  Without that, there's nothing we can even look at here.

chester2400, 2023-02-17:
Its not the multi core problem. Its the TFT_eSPI library. It pulls pin 0 high.

earlephilhower, 2023-02-17:
You can change the pins using `Serial1/2.setRX/setTX` to move it them to different pins.  Be sure the selected pins are legal for the hardware (see https://learn.adafruit.com/assets/99339 )

closed 2023-02-17 by earlephilhower