earlephilhower/arduino-pico #1206: Error expanding macro for PIN_LED: initializer element is not constant

oddstr13, 2023-02-19:
Regression caused by #1204

```
In file included from /home/oddstr13/Arduino/hardware/pico/rp2040/cores/rp2040/Arduino.h:30,
                 from /tmp/arduino_build_702994/sketch/constants.c:1:
/home/oddstr13/Arduino/hardware/pico/rp2040/variants/rpipicow/pins_arduino.h:9:24: error: initializer element is not constant
    9 | #define PIN_LED        (__isPicoW ? 32u : 25u)
      |                        ^
/home/oddstr13/Arduino/hardware/pico/rp2040/variants/rpipicow/../generic/common.h:9:21: note: in expansion of macro 'PIN_LED'
    9 | #define LED_BUILTIN PIN_LED
      |                     ^~~~~~~
/tmp/arduino_build_702994/sketch/constants.c:5:11: note: in expansion of macro 'LED_BUILTIN'
    5 | int led = LED_BUILTIN;
```

Target board: Raspberry Pi Pico W

earlephilhower, 2023-02-19:
Fixed by #1208.  CI didn't catch it because that use case (setting a global) wasn't in any of the WiFi-related sketches.  I've manually tested it w/Blink and BlinkWithoutDelay which does properly work on both Pico/PicoW (and uses a global int set to `LED_BUILTIN`)

oddstr13, 2023-02-22:
I've made a PR for a minimal reproduction, do with it as you wish.

I'm personally not entirely sure of all the implications of handling all of this at runtime as implemented in #1208, I think it has the potential to cause hard to track down bugs (the `PIN_LED 32 -> 25` mapping may not hold true on all boards, Pin 32 may have other meanings on some non-wifi boards).

Might not be an issue with current boards, but it makes me feel a bit uneasy.

closed 2023-02-23 by earlephilhower