earlephilhower/arduino-pico #1215: Should I use SPI or SPI1 in the program?

jinyistudio534, 2023-02-21:
https://shop.sb-components.co.uk/products/raspberry-pi-pico-hat-expansion?_pos=9&_sid=155b6beff&_ss=r
[image]

When I want to use GP2~GP5 (SPI,PICO side) in the figure, Should I use SPI or SPI1 in the program?

earlephilhower, 2023-02-21:
When in doubt about pinouts, check this image to verify: https://learn.adafruit.com/assets/99339
[image]

GP2-5 are `SPI`, not `SPI1`.

closed 2023-02-21 by earlephilhower

jinyistudio534, 2023-02-21:
`SPI.setCS(5);
  SPI.setSCK(2);
  SPI.setRX(4);
  SPI.setTX(3);`

I have connect a spi device to GP2~GP5. Why do i need to specify the pin position in the above program when using SPI in the program? If not specified, SPI cannot communicate

earlephilhower, 2023-02-22:
It depends on your board.  Those pins may not be the default ones for it (many peripherals can come out on many different pins on the RP2040).