earlephilhower/arduino-pico #1216: Feather RP2040 Scorpio pins not working as they should

whimsee, 2023-02-21:
Hi! I just migrated my Feather RP2040 project to its SCORPIO variant and I'm running into some trouble with the I2C and SPI pins not being where they should be since my OLED wasn't connecting on the I2C, as does the SD card using SPI.

I did some digging and comparing with the pinout sheet and pins_arduino.h file for both boards and it looks like the pins aren't lining up right with the bus.

To check: I did go ahead and change the I2C pins from--

SCORPIO:
#define PIN_WIRE0_SDA  (2u)
#define PIN_WIRE0_SCL  (3u)
#define PIN_WIRE1_SDA  (31u)
#define PIN_WIRE1_SCL  (31u)

to

BASE FEATHER RP2040:
#define PIN_WIRE0_SDA  (24u)
#define PIN_WIRE0_SCL  (25u)
#define PIN_WIRE1_SDA  (2u)
#define PIN_WIRE1_SCL  (3u)

And it worked for the OLED (though looking at it now, SDA and SCL should be 16 and 17 respectively). But I'm kind of stuck with the SPI not working for the SD card since it's looking at a different bus (ie. the Base Feather RP2040's SCK0 is now SCK1 in SCORPIO)

earlephilhower, 2023-02-21:
The Scorpio variant was built by ladyada and team at Adafruit:  https://github.com/earlephilhower/arduino-pico/commit/913dfad1ad3ef48f7270f26e2351ede7f8bc09de .   If there's an issue with the pinout, can you make a PR to fix it?

As for the SPI busses, just use this `SD::begin` to tell it to use SPI1 not SPI0:
https://github.com/earlephilhower/arduino-pico/blob/a97d5eab22ea0edecd01193ada120733ea898e75/libraries/SD/src/SD.h#L38

whimsee, 2023-02-21:
Thanks for the reply! Since I was using the SdFat library with SPI, I did end up using its SoftwareSPI class instead, which worked nicely with some fiddling. Though I'm grateful for that solution as well in case anyone else runs into this issue.

And sure, I wasn't certain if I could submit a PR because I've never done it before but I'd be glad to. I'm fairly positive about the I2C pins now but I'm still not 100% about the SPI pins; more like I know where SCK should be for both buses, but not so much about MOSI, MISO, and SS.

earlephilhower, 2023-02-21:
I guess for now we can close this.  If someone else runs into the same issue we can see about changing the pinout.  Unfortunately, I don't have that particular board myself so can't really chime in one way or the other.  Thx!

closed 2023-02-21 by earlephilhower