earlephilhower/arduino-pico #1230: Causing rp2040.reboot from an open terminal doesn't reconnect correctly

obdevel, 2023-02-26:
I have a mini serial command processor and sending a '*' causes rp2040.reboot() to be called.

However, when restarting, the startup messages are not always printed to the IDE serial terminal. e.g. I have the common serial startup code in setup():

```
  unsigned long t1 = millis();

  Serial.begin(115200);
  while (!Serial && millis() - t1 < 5000);
  Serial.printf("> xxx\n");
```

It seems setup() is waiting the 5 seconds before continuing but nothing is printed, and the terminal is unresponsive until I close and open it again. This happens about 50% of the time. I'm using IDE v 1.8.19 on macOS 12.3.1.

I have tried it with another terminal program (CoolTerm on Mac) and it works correctly 100% of the time.

So, this may well be the IDE's problem. I will raise an issue with that project if it's something you've seen before. And it's documented here for future searchers.

earlephilhower, 2023-02-26:
Your problem is most likely the reboot causes the terminal device node to change since the USB is run in SW on the Pico.  When the port goes away, the IDE will not close it's connection and hold on to the file pointer to the original terminal devnote.  When the "new" serial port comes alive after the Pico boots up the OS can't reuse the old name, since someone has it open, and comes up with a new one.  Closing/reopening on the IDE may end up picking the right name.

In any case, there's nothing in the core we can do about this.  It varies by OS (even different Linux kernels behave differently) so there's no one-size-fits-all answer.

As for the Arduino IDE bugs, I  don't believe they are taking any 1.x reports anymore.  Good luck, but I think they're going to suggest moving to 2.x.

closed 2023-02-26 by earlephilhower

obdevel, 2023-02-26:
I seem to have fixed it by inserting a yield() call into the while !Serial loop. I guess it is some kind of timing/race condition, or the processor needs some time to be able to startup the serial USB stack.

Hopefully this is useful to someone else. Thanks anyway.

earlephilhower, 2023-02-26:
FWIW, unless you're using TinyUSB, `yield()` is a complete no-op (well, other than a call/ret):
https://github.com/earlephilhower/arduino-pico/blob/264b9efb36dfeb3151549a606ebb9f21a67b9157/cores/rp2040/delay.cpp#L49-L54