earlephilhower/arduino-pico #1231: MacOS: Board Info does not show correct VID & PID (and serial monitor is not working as well)

Papirosnik, 2023-02-26:
- MacOS Ventura 13.2.1
- Apple M1 Pro
- Arduino IDE 2.0.3
- Arduino-picp v.2.7.3

[image]
[image]

Here is my simple sketch. It definitely works (the led is blinking at least) and is supposed to  write to serial port as well. 
But it either does not or I cannot see its ouput.
[image]

Screen command gives an empty screen
[image]

So, can someone confirm that Serial facilities are working with Pico on Mac?
Any ideas how to treat that?

earlephilhower, 2023-03-03:
As far as the IDE issue, that's something that works fine in 1.8.x and got broken when they moved to the new IDE.  It's not a big issue but if you do have a solution, a PR would be welcome.

I'm not a Mac user, but for the USB port, if you cannot manually choose the existing USB port from the IDE menu then I would guess you have a permission problem.  Either the app or the user isn't allowed to see the port, even though it's `rw` for everyone.  You'll need to find a Mac user to get more info on setting their equivalent of `udev` rules/etc. if such a thing exists.

earlephilhower, 2023-03-07:
No response in a week, closing.

closed 2023-03-07 by earlephilhower

Papirosnik, 2023-03-07:
Hah!
I've just read these messages on already closed thread and decided to try one more time.
I read this [stuff](https://www.stoneitech.com/raspberry-pi-pico-tutorial/#:~:text=The%20default%20baud%20rate%20is,similar%20to%20COM1%20in%20windows.) and thought about that `enable_uart=1` option, although had no idea if it is pico-related actually.
And suddenly and magically everything started working as expected :)
Probably, the issue was in my connected peripheria. This time I tried without any pins connected - the micro-usb cable only.
And voila!
[image]
And serial port is working properly and I can see its output.