earlephilhower/arduino-pico #1233: Bricking the Pico trying to produce a 5Hz squarewave, then need to switch IDEs to reprogram it

heliophagus, 2023-02-27:
I am using Arduino IDE 2.0.3 & 2.0.4. This is a slightly mixed question, having to do with (a) using the pico/stdlib.h library to generate a 5Hz squarewave from a GPIO, which bricks the Pico, & (b) the resulting circus involved with resuscitating the bricked Pico for reprogramming, which requires loading "Blink" using Arduino IDE 8.1.19. It's curious that one IDE can reprogram the Pico but the other cannot.

TL/DR: How do I generate a 5Hz squarewave without bricking the Pico? And, having bricked the Pico & then put it into boot mode, is there any way to avoid having to use the old IDE to re-enable uploading?

See issue #187 for the need for a routine that generates a simple low frequency squarewave (in my case, exactly 5Hz with minimal jitter: lower than available with the PWM commands). I adapted this code from @Bodmer's reply.

```
#include <stdio.h>
#include "pico/stdlib.h"

bool repeating_timer_callback(struct repeating_timer *t) {
  boolean wut = digitalRead(LED_BUILTIN);
  digitalWrite(LED_BUILTIN,!wut);
  return true;
}

void setup() {
  Serial.begin(115200);
  pinMode(LED_BUILTIN, OUTPUT);
  delay(1000);
  Serial.println("Starting timer");
  struct repeating_timer timer;
  add_repeating_timer_ms(100, repeating_timer_callback, NULL, &timer);
  delay(30000);
}

void loop() {
  delay(1);
}
```
The sketch produces the 5Hz squarewave as requested, then bricks the Pico after the "sleep_ms(30000);". If "sleep_ms(30000);" is omitted the board is bricked instantly with no squarewave produced.

Now the circus begins! If I try uploading this or any other sketch in the 2.0.x IDE after putting the Pico into boot mode (I'll wear out the USB socket at this rate), I get: 

> Resetting COM5
> Converting to uf2, output size: 137216, start address: 0x2000
> Scanning for RP2040 devices
> No drive to deploy.
> Failed uploading: uploading error: exit status 1

Trying again:

> Failed uploading: no upload port provided

It won't stop expecting a non-existent upload port until I start Arduino IDE 1.8.19, then load & program "blink", which the old IDE happily does.

Now I can program the Pico in the new IDE, but again, on this 5Hz example, it bricks. Incidentally, it also randomly bricks in < 1 minute if I try producing a 5Hz squarewave using @khoih-prog's RPI_PICO_TimerInterrupt library. (And if I program it using the old IDE, it bricks too.)

A precise 5Hz squarewave is essential for this project. I'm at my wit's end! It "should" be simple to do. Any help or constructive suggestions would be greatly appreciated.

Bodmer, 2023-02-28:
struct repeating_timer timer;

Only exists in the scope of setup().
When loop() is entered it no longer exists, hence the crash. Make it global.

earlephilhower, 2023-02-28:
Thanks, @Bodmer !

closed 2023-02-28 by earlephilhower

heliophagus, 2023-02-28:
Beautiful! Will try, and thank you!

earlephilhower, 2023-02-28:
When you crash the Pico, the USB goes away since the Pico SW has to do the USB stack.  Once the Pico isn't crashing, then uploads are pretty seamless. :)

heliophagus, 2023-02-28:
Thanks for the explanation. Moral of the story: Don't crash the Pico! At least it's recoverable if that happens.