earlephilhower/arduino-pico #1247: Serial.print() issue when using TinyUSB as USB Stack

hlcmOxO, 2023-03-03:
I have the same problem in #1231  and I was using 
Windows 10 profedssional 21H2
Arduino IDE 2.0.3
Arduino-picp v.3.0.0.
.
[image]
.
.
.
And I'm unable to make Serial.print() works properly when using TinyUSB as USB Stack.
.
I tried the sketch provided in that issue.
when I selected Pico SDK for USB Stack, the sketch worked fine.
[image]
Properly outputting "1" and "0" for light on and off with 1000ms delay.
.
.
But when I selected Adafruit_TinyUSB for USB Stack. It still outputted. But it didn't output "1" then, after 1000ms, output "0". Instead, it outputted "10" together (no delay between 1 and 0). While the light still blinking with 1000ms delay. (as gif below)
[image]
.
.
When I tried adding another Serial.print() into the loop() (Still using TinyUSB as USB Stack)
![%4A6X@LRU)
AH8@VADY%`5Q](https://user-images.githubusercontent.com/103294894/222634967-8ca6dad4-3e0d-4ba3-b373-208111a55196.png)
it started to outputted "021" together with no delay between "1", "0" and "2".
[image]

Any ideas how to treat that?

earlephilhower, 2023-03-03:
As discussed in #642, you need to use `Serial.flush()` to cause the Adafruit TinyUSB to pump all the characters down the pipe.  That's just the way they've implemented their stack.  The built-in USB is somewhat different, so unless you really need TinyUSB to do what you're trying to do I suggest just going with that.

closed 2023-03-03 by earlephilhower

hlcmOxO, 2023-03-03:
Thanks for your help! Adding Serial.flush(); solves the problem.