earlephilhower/arduino-pico #1251: setup_default_uart() error

Dj-EKI, 2023-03-03:
In both versions. 2.7.3 and 3.0.0
`setup_default_uart();`
Generates this error.
```
c:/users/reder/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\reder\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.0.0/lib/libpico.a(stdlib.c.obj): in function `setup_default_uart':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/src/rp2_common/pico_stdlib/stdlib.c:94: undefined reference to `stdio_uart_init'
collect2.exe: error: ld returned 1 exit status
```

Maybe I'm using the function incorrectly. The sleep_run_from_xosc(); uses this function and produces the same error. sleep_run_from_xosc(); is from Pico-extras.

earlephilhower, 2023-03-03:
You cannot call `stdio_uart_init` in this core.  The Arduino `Serial/Serial1/Serial2` ports already take control of the UART and USB.

Use `Serial1` for UART0 and `Serial2` for UART1.  `Serial` is USB.  Then `Serial.printf("dddd");` will just work, no SDK stuff needed.

earlephilhower, 2023-03-03:
One more thing, if you set the Tools->Debug Port->Serial/1/2, then plain `printf()` will go to that specific port (i.e. printf is not Arduino-like and normally only a debug thing).

closed 2023-03-03 by earlephilhower