earlephilhower/arduino-pico #1264: Challenger RP2040 WiFi examples didnt compile

Buddha56, 2023-03-04:
Hi.
I have a Problem to compile with the Challenger RP2040. None examples with wifi work . Here is the error Code from Arduino IDE : 

In file included from ...AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.0.0\libraries\lwIP_CYW43\src/lwIP_CYW43.h:3,
                 from ...AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.0.0\libraries\lwIP_Ethernet\src/LwipEthernet.h:28,
                 from ...AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.0.0\libraries\WiFi\src\WiFiClass.cpp:33:
..AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.0.0\libraries\lwIP_Ethernet\src/LwipIntfDev.h:59:29: error: 'SS' was not declared in this scope
   59 |     LwipIntfDev(int8_t cs = SS, SPIClass& spi = SPI, int8_t intr = -1) :
      |                             ^~

earlephilhower, 2023-03-04:
Sorry, the `WiFi` only works with the CYW43 chip on the PicoW.  You will need to use the Challenger's own WiFi libraries to use their onboard networking.

closed 2023-03-04 by earlephilhower

sonnny, 2023-03-06:
[youtube demo](https://www.youtube.com/watch?v=Ncc9SWQs5zs)