earlephilhower/arduino-pico #1268: Setting RX and TX pins on Serial 2 causes the pico to hang up

drakejest, 2023-03-05:
Arduino IDE v 2.0.4
Core v 3.0.0

Hello i have been scratching my head on why my code was not working, and have narrowed it down to these specific lines. setRX and setTX.

If i uncomment those line and upload it to the pico, after successful upload the pico will become not recognizable by the computer. I have to use BOOTSEL button  in order to upload new code. I can also infer that the pico hangs up or gets stuck in those 3 lines of code because I do not see the internal LED blink so it was not able to reach the loop statement .

What makes me wonder is v2.6.0, and v2.7.0 this also happens. So this has not been caught for a long time, making me think that i am most probably doing something wrong.

Any advice? Thanks 
 
```
#define device2_TX 4
#define device2_RX 5
#define device2_RST 6

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  //Serial2.setRX(device2_TX);
  //Serial2.setTX(device2_RX);
  //Serial2.setFIFOSize(512);

}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(200);
  digitalWrite(LED_BUILTIN, LOW);
  delay(200);
}
```

earlephilhower, 2023-03-05:
Those are not legal UART1 pins, caused the Pico to panic.  Check the datasheeet for the permitted pin outs.

closed 2023-03-05 by earlephilhower

drakejest, 2023-03-05:
@earlephilhower  Well this is a bit confusing to me.... What  numbering does the core use? GPIO number or pin number? 

The documentation [here ](https://arduino-pico.readthedocs.io/en/latest/serial.html) did not mention what pin numbering to use or what pins are permitted for UART0 and UART1. Can you please link to me  the documentation for the permitted pin outs?

I used 0 and 1 for UART0. And it works, so i assumed its the GPIO numbering

```
  Serial1.setRX(0);
  Serial1.setTX(1);
  Serial1.setFIFOSize(512);
```

GPIO 4 and 5 are listed here as UART1.

[image]

earlephilhower, 2023-03-05:
Yes, it takes GP numbers in the calls.  You've got all you need in the picture you added.  The outer purple where it says UART1-TX is legal for `Serial2.setTX()` (i.e. 4, 8).  Same for RX.

drakejest, 2023-03-05:
@earlephilhower  I made a mistake in giving you the wrong example code. The code i posted has a flipped RX and TX parameter, but its not what i ment by wrong. I was trimming the code to be as short as possible to be posted here. on the actual code the setTX and setRX are given valid pin numbers buts still cant proceed .

After 12 hours of investigating (since before i posted this thread). I finally have narrowed it more down, calling setTx and setRx of serial2 inside a class is causing problems..

```
class uartDevice {

  public:

    uartDevice(int serialDeviceNumber, uint8_t GPIO_TX, uint8_t GPIO_RX, uint8_t GPIO_RST, uint32_t FIFO_SIZE = 512) {

      deviceNumber = serialDeviceNumber;
      txPin = GPIO_TX;
      rxPin = GPIO_RX;
      resetPin = GPIO_RST;
      bufferSize = FIFO_SIZE;

      pinMode(resetPin, OUTPUT);
      digitalWrite(resetPin, LOW);

      switch (serialDeviceNumber) {
        case 1:
          Serial1.setTX(txPin);
          Serial1.setRX(rxPin);
          Serial1.setFIFOSize(bufferSize);
          serialDevice = &Serial1;
          break;
        case 2:
          Serial2.setTX(4); // Same exact code as in the setup() but this time is hangs up
          Serial2.setRX(5);
          Serial2.setFIFOSize(512);
          serialDevice = &Serial2;
          break;
        default:
          return;
      }
    }

  private:
    uint8_t deviceNumber;
    uint8_t txPin;
    uint8_t rxPin;
    uint8_t resetPin;
    uint32_t bufferSize;

    HardwareSerial* serialDevice = nullptr;
};

uartDevice dev2(2,  4,  5,  6, 512); // <- this line causes the freeze

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  //    Serial2.setTX(4); // When executed here works no problem
  //    Serial2.setRX(5);
  //    Serial2.setFIFOSize(512);

}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(200);
  digitalWrite(LED_BUILTIN, LOW);
  delay(200);
}
```

drakejest, 2023-03-05:
Its getting even weirder... It would seem that the constructor parameters is whats making it not work...

This code works : 

```
class uartDevice {
  public:
    uartDevice() {
      Serial2.setTX(4); // Same exact code as in the setup() but this time is hangs up
      Serial2.setRX(5);
      Serial2.setFIFOSize(512);
    }
};

uartDevice dev2();

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  //    Serial2.setTX(4); // When executed here works no problem
  //    Serial2.setRX(5);
  //    Serial2.setFIFOSize(512);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(200);
  digitalWrite(LED_BUILTIN, LOW);
  delay(200);
}
```

While this does not. The constructor parameters are not even used.

```
class uartDevice {
  public:
    uartDevice(int serialDeviceNumber, uint8_t GPIO_TX, uint8_t GPIO_RX, uint8_t GPIO_RST, uint32_t FIFO_SIZE = 512) { 
      Serial2.setTX(4); // Same exact code as in the setup() but this time is hangs up
      Serial2.setRX(5);
      Serial2.setFIFOSize(512);
    }
};

uartDevice dev2(2,  4,  5,  6, 512); //the parameters are not even used

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);

  //    Serial2.setTX(4); // When executed here works no problem
  //    Serial2.setRX(5);
  //    Serial2.setFIFOSize(512);
}

void loop() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(200);
  digitalWrite(LED_BUILTIN, LOW);
  delay(200);
}
```

earlephilhower, 2023-03-05:
You are running into a C++ issue, not a core one here.  The order that constructors are called is not guaranteed, so you cannot call another global object safely.  That's why there are the `::begin` methods which do most of the work in the Arduino core.  The object constructors just set a couple entries and call no other objects.

In the case which "does nothing" your constructor is called before the Serial2 one, so the `Serial2::Serial2()` constructor overwrites the values you just set.

drakejest, 2023-03-05:
> You are running into a C++ issue, not a core one here. The order that constructors are called is not guaranteed, so you cannot call another global object safely. That's why there are the `::begin` [...]

I see.... so my lesson to take away here is not to do anything on the constructor aside from assigning variables. Moving those lines of code into the begin() does seem to fix the problem.

I have coded several C++ programs for windows and have not so far encountered this problem. I would assume this is a Arduino specific issue?

earlephilhower, 2023-03-05:
Nope, it's a C++ thing and I think you just got lucky.  Global object constructors have no guarantees whatsoever for ordering.  https://www.google.com/search?q=c%2B%2B+global+constructor+ordering+site:stackoverflow.com has lots of discussion about this.

At the end of the day, there's just an array of constructor functions to be called by the runtime init routines.  What order the fcns are in that table? Could be anything as long as they're all there.

earlephilhower, 2023-03-05:
(To be clear, I'm not talking about C++ inherited/base class ordering, that **is* preserved for global objects.  It's just between 2 global objects that you can't guarantee anything)

drakejest, 2023-03-05:
> (To be clear, I'm not talking about C++ inherited/base class ordering, that *_is_ preserved for global objects. It's just between 2 global objects that you can't guarantee anything)

Yes i quite understand, now that i have think of it, i have not coded on global objects that called on each other. majority of what i coded are inherited/base classes

Thank you very much for taking the time to help me. i was just so lost on this problem it was soo weird to me.

Again thank you so much for the work on the core, enjoying v3.0.0 so far, it has really come a long way from the v1 days where i had problems with :)