earlephilhower/arduino-pico #1273: Pico-SDK not initializing

PhilippMolitor, 2023-03-05:
I just tried to get some timer stuff running and noticed that the documentation has no such thing in it (something like Timer1/2/3 in Arduino).

When using `#include <pico/stdlib.h>` and `stdio_init_all()` to start a timer using the pico-sdk,  the following compilation error shows up:
```
/Users/phil/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: .pio/build/debug/src/main.cpp.o: in function `setup':
/Users/phil/Projects/truncated/src/main.cpp:117: undefined reference to `stdio_init_all'
```

I am using the most recent version with PlatformIO.

earlephilhower, 2023-03-05:
The Pico SDK stdio is not supported since the USB and UART ports are owned by the Arduino Serial/1/2 ports.

The Arduino way is `Serial.printf()`/etc.  You can also use `printf()` if you set the Tools->Debug port to the desired output.

closed 2023-03-05 by earlephilhower

PhilippMolitor, 2023-03-07:
Is there any way to interact with the timers without accessing the raw timer registers?

earlephilhower, 2023-03-07:
There's the `alarm` portion of the SDK you can look at.  It's got a higher level API than register-level twiddling.