earlephilhower/arduino-pico #1274: basicHTTPclient freezes when end point does not exist

manlyfoodcoop, 2023-03-06:
My code scans IP addresses on startup to look for the aircontioner conroller. It used to work but I think i accepted an update and now when the IP address points to a non existent host it never returns. The debug stops at ": ref 1". I tried setting low timeout but no luck. The issue can be re-produced using the sample basicHttpClient with a non-existent end point:

```
/**
   BasicHTTPClient.ino

   Created on: 24.05.2015

*/

#include <Arduino.h>
#include <WiFi.h>
#include <HTTPClient.h>

#ifndef STASSID
#define STASSID "mysid"
#define STAPSK "mypwd"
#endif

const char *ssid = STASSID;
const char *pass = STAPSK;

WiFiMulti WiFiMulti;

void setup() {

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFiMulti.addAP(ssid, pass);
}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

    HTTPClient http;

    Serial.print("[HTTP] begin...\n");
    if (http.begin("http://192.168.1.135/skyfi/aircon/get_sensor_info?lpw")) {  // HTTP

      Serial.print("[HTTP] GET...\n");
      // start connection and send HTTP header
      int httpCode = http.GET();

      // httpCode will be negative on error
      if (httpCode > 0) {
        // HTTP header has been send and Server response header has been handled
        Serial.printf("[HTTP] GET... code: %d\n", httpCode);

        // file found at server
        if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
          String payload = http.getString();
          Serial.println(payload);
        }
      } else {
        Serial.printf("[HTTP] GET... failed, error: %s\n", http.errorToString(httpCode).c_str());
      }

      http.end();
    } else {
      Serial.printf("[HTTP} Unable to connect\n");
    }
  }

  delay(10000);
}`
```

earlephilhower, 2023-03-06:
Thanks for the simple description and reproduction info, that makes it much easier to look into!

Could you please revert to 2.7.3 and try the failing code in your setup and report back?  The 3.0.0 build didn't really change the WiFi code used, but it did hand over the WiFi background IRQ processing to the Pico-SDK handler (before I had my own, painfully developed, but with the Bluetooth infrastructure it's no longer tenable to maintain my own version).

If the Serial port doesn't disappear when this happens, I think we have a mutex deadlock in the new CYW43/LWIP stuff.  If it does crash, then I think we have a LWIP timer issue, as reported by others with the 1.5 SDK version in the PicoSDK repo...

earlephilhower, 2023-03-06:
Quick testing at lunch shows this is related to the new async_context infrastructure and `sys_check_timeouts` (the first option above).

manlyfoodcoop, 2023-03-07:
I can confirm reverting to 2.7.3 has resolved the issue. Praise for your "painfully developed" original stable code!

earlephilhower, 2023-03-07:
Yeah, there is some instability in the 3.0.0 + PicoSDK background processing.  In 2.7.3 I was able to pound the AdvancedWebServer from 2 machines holding down F5 to refresh constantly on both.

In the SDK 1.5 version I can get a failure in under 30 seconds in the same test with corruption of the LWIP internals.  That implies a locking issue, sadly.  This is gonna be somewhat involved... :sob:

manlyfoodcoop, 2023-03-08:
What is support like for you getting changes / fixes in the PICO SDK? Or is it up to people like you to identify and make the actual fixes directly?

earlephilhower, 2023-03-08:
Just to be clear, this could be all on me not understanding the new async context stuff.  So I'm not saying they have a problem, just that translating what I had before based on their old interrupt driven stuff to the new framework isn't working well (yet).  For low intensity workloads, it's fine.  I ran over 16hrs of AdvancedWebServer before merging (update from the PC E every 5 seconds) without incident.

That said, the Raspberry Pi guys are great.  But I obviously need to do my homework before bugging them.  I've submitted bugs and patches and they've all been given a fair shake.

closed 2023-03-08 by earlephilhower

earlephilhower, 2023-03-08:
Looks like it was me, after all.  There were ordering guarantees that were no longer valid when moving to the new async infrastructure.  With #1286 I can bang AdvancedWebServer with F5 refreshes for 1000s of cycles without any issues.  It also removes the special-casing causing the example you posted here to hang (it now times out as expected).

If you still see stuff, please do open a new issue with MCVE/etc.

PR #1286 (2023-03-08): Adjust LWIP intf to avoid crashes with null PCBs
It seems possible now for TCP connection _pcbs to disappear while being processed, due to the new async context configuration.  This would cause LWIP to hang with a panic when a NULL pcb was passed in.

Check for and avoid passing in null PCBs in the ClientContext.

Undo special-casing of sys_check_timeouts wrapper

AdvancedWebServer with heavy F5-refresh and #1274 test both pass.

Fixes #1274
commit e6c7b97b (2023-03-08): Adjust LWIP intf to avoid hangs/crashes under load (#1286)

It seems possible now for TCP connection _pcbs to disappear while being
processed, due to the new async context configuration.  This would cause
LWIP to panic when a NULL pcb was passed in.

Check for and avoid passing in null PCBs in the ClientContext.

Undo special-casing of sys_check_timeouts wrapper

AdvancedWebServer with heavy F5-refresh and #1274 test both pass.

Fixes #1274