earlephilhower/arduino-pico #1276: UART using SerialPIO produces 2 unknown characters at the end

drakejest, 2023-03-06:
Im using SerialPIO for UART but when i print the response from the UART device serial monitor shows two questionmark characters "??". I know that its not supposed to be there because if i transfer it a hardware UART say Serial1 or Serial2 the question marks dont show up in the serial monitor

[image]

The code:

```
SerialPIO dev1(8, 9, 512);

void setup() {

  pinMode(10, OUTPUT);
  digitalWrite(10, HIGH);
  
  dev1.begin(9600);
  Serial.begin(9600);

  while(!Serial);
  
  Serial.println("Device 3: ");
  dev1.println("AT");
  dev1.flush();
  delay(2000);

  Serial.println(dev1.readString());
}

void loop() {        
}
```

I went on investigating further by doing:

```
  //Serial.println(dev1.readString());
  while (dev1.available()) {
    Serial.println(dev1.read(),HEX);
  }
```
[image]

moving over the device to a hardware uart its response is this:

[image]

The last two characters are different, i checked with the ascii table what those ment its a non standard character, The hardware UARTs response is the correct one 0x0D 0x0A.

earlephilhower, 2023-03-06:
The SerialPIO and the UART handle mis-framed bytes differently.  Can you grab a logic analyzer trace of your setup?  Without your hardware, it's impossible to reproduce this since you're having trouble with what some other chip is sending you.

What it looks like is you have a clock mismatch.  0x86 * 2 = 0x0d, 0x85 * 2 = 0x0a.  So the 9600bps speed (which is seldom exactly 9600 due to clock limitations on both sides) is off, with the external chip slower than the PIO expects.

drakejest, 2023-03-06:
[image]

screen shot, sorry but i dont have a UART module. I will update the image after i figure out how to install the uart module of the analyzer

drakejest, 2023-03-06:
[image]

Updated screenshot, and it confirms that PIO is recieving wrong data.

I have 6 UART device 2 hardware and 4 PIOs. all 4 PIOS has the last 2 characters wrong. If it is what you say a clock mismatch, what can be done?

earlephilhower, 2023-03-06:
If you could get some measures of bit times that would help, both for PIO and for the external chip.  Then you could adjust the PIO baud up/down to better match the sender.

drakejest, 2023-03-06:
for the PIO (sender)

[image]

name	                type	start_time	duration	                data
Async Serial [1]	data	0.799333958	0.000989583333	0x41
Async Serial [1]	data	0.800480125	0.000989583333	0x54
Async Serial [1]	data	0.801626292	0.000989583333	0x0D
Async Serial [1]	data	0.802772458	0.000989583333	0x0A

For the device (response)

[image]

name	        type	start_time	duration	                data
Async Serial	data	0.808673792	0.000989583333	0x41
Async Serial	data	0.809707792	0.000989583333	0x54
Async Serial	data	0.810741833	0.000989583333	0x0D
Async Serial	data	0.811775792	0.000989583333	0x0D
Async Serial	data	0.81280975	0.000989583333	0x0A
Async Serial	data	0.81384375	0.000989583333	0x4F
Async Serial	data	0.814877792	0.000989583333	0x4B
Async Serial	data	0.815911792	0.000989583333	0x0D
Async Serial	data	0.81694575	0.000989583333	0x0A

Im not sure if this is what you mean. I used an image to make the formatting easier. I also included a text form incase you want to copy and paste it somewhere

drakejest, 2023-03-06:
Does the serial PIO accepts awkward baudrates like  9610 for example?

earlephilhower, 2023-03-06:
> Does the serial PIO accepts awkward baudrates like 9610 for example?

Yes, it doesn't have any special speed requirements.

drakejest, 2023-03-06:
okay got the correct data now at baud 9620, but at baud 9800 data is also correct. Should i go higher?  what is generally a good number here?

drakejest, 2023-03-06:
Thank you for the help as always :)

earlephilhower, 2023-03-06:
No worries.  The bugger is this should be an easy thing for the PIO to grab because technically you're to sample at the midpoint of a bit time.  So you have a **lot** of clock cycles to move fwd and back before you slip a bit like this.  Maybe we're grabbing at the end of a bit so it's hairy.  Worth investigating at some point...

maxgerhardt, 2023-03-07:
Does the PIOSerial do like a 8-times oversampling and majority vote on the bit value?

earlephilhower, 2023-03-07:
No, the PIO can't do anything so complicated.

Now, running at Fsys, the PIO waits for a 1->0 transition.  It then waits for 1/2 a bit time's worth of Fsys clks and captures 18(? from memory, may be 19) samples every 1/2 bit time and eventually pushes it to the core FIFO.  The ARM then decodes it and checks framing/parity/etc.

It should be possible to add a debug 1-bit GPIO sideset to indicate the sample clock to experimentally verify things.  What is weird here is that the PIO is being used at much higher frequencies by folks (and I tested at up to 1MHz IIRC).  At 9600bps there should be tons of margin.

It could also be a function of stop bit processing since the earlier chars seem to come through OK.  But I would have thought if that would cause the read data to shift left, not right like in this case.

drakejest, 2023-03-07:
for the record, the device i am using is the SIM800L V2. Im using 6 of these connected to a single pico board. I have 4 serialPIO instance and all 4 of them wrongly reads the last 2 character.

drakejest, 2023-03-11:
I bought new pico and pico W.  Its still does this, adjusting the baud rate solves the problem as proposed by @earlephilhower  :)

earlephilhower, 2023-03-18:
@drakejest can you please reduce the BPS in your app back to 9600 exactly, and try the 1-character patch in #1315.  You can manually change the `34` to `35` in that one source line if you have trouble using the GIT version or managing PR pulls.

Essentially it looks like the PIO UART was capturing every 1/2 cycle, starting at the middle of the start bit.  But, that means that the first *2* bits captured (middle of start + end start/begin bit0) should be thrown out, not *1*.  Depending on hold times it could "just work", and speeding up the BPS (like you did to make it work) would also push that sample closer to the center of the bit and be more solid.

drakejest, 2023-03-29:
Sorry took so long to reply, my modules are currently deployed. Will come back with an update when i get the chance to take it down. 

Thanks for fixing the problem :)

drakejest, 2023-04-08:
@earlephilhower  so i tested out the new edit and there is a problem, all the characters that were correct is now wrong and the last two that were wrong is now correct 

[image]

reverting back to `decode >>= 33 - _rxBits;` goes back to the original problem

[image]

earlephilhower, 2023-04-08:
Weird.  Loopback from another serial port on the pico showed no errors with this patch (or the unmodified one).  I'll close this and the PR since there's something special about that chip...

closed 2023-04-08 by earlephilhower

Apehaenger, 2024-05-16:
Had the same issue with an [WT901 IMU](https://witmotion-sensor.com/products/9-axis-accelerometer-tilt-sensor-wt901-high-accuracy-acceleration-gyroscope-angle-magnetometer-with-kalman-filtering-triaxial-mpu9250-ahrs-imu-iic-ttl-200hz-for-pc-android-arduino).
Everything worked great with HardwareSerial but when trying to drive it via SerialPIO I got a couple of non-sense data. After I tried everything I googled around and (luckily) found this post. Couldn't believe at first, but tried with 9650 baud and got immediately correct data. Tried up to 9800 baud and all worked, but not 9600!