earlephilhower/arduino-pico #1278: Raspberry Pico W with Arduino-Pico framework failing on install

jimemo, 2023-03-06:
I’m developing for a Raspberry Pico W using the PlatformIO IDE. This has been working fine with thse instructions for a few months: [Using this core with PlatformIO — Arduino-Pico 3.0.0 documentation](https://arduino-pico.readthedocs.io/en/latest/platformio.html)
Since a third party library melted my existing installation I am doing a reinstall of PlatformIO and the arduino-pico framework. Does anybody recognise these error and have a fix or workaround?
`fatal: Failed to recurse into submodule path ‘pico-sdk/lib/tinyusb’`
Here is my platformio.ini that worked before I started the reinstall.

```
[env:rpipicow]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipicow
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
```
Here is the build output which is where the arduino-pico framwork is being installed. It is failing at the same point when downloading parts of the pico-sdk:

```
Submodule 'hidapi' (https://github.com/signal11/hidapi) registered for path 'pico-sdk/lib/tinyusb/tools/uf2/hidapi'
Cloning into 'C:/Users/jimem/.platformio/.cache/tmp/pkg-installing-5v0888hr/pico-sdk/lib/tinyusb/tools/uf2/hidapi'...
Submodule path 'pico-sdk/lib/tinyusb/tools/uf2/hidapi': checked out 'a6a622ffb680c55da0de787ff93b80280498330f'
fatal: Failed to recurse into submodule path 'pico-sdk/lib/tinyusb'
fatal: Failed to recurse into submodule path 'pico-sdk'
Submodule path 'tools/libbearssl/bearssl': checked out 'f294aa022f82833941ffdf3b0f306c2072e3a26c'
Submodule path 'tools/pyserial': checked out '0e7634747568547b8a7f9fd0c48ed74f16af4b23'
Error: VCS: Could not process command ['git', 'clone', '--recursive', 'https://github.com/earlephilhower/arduino-pico.git', 'C:\\Users\\jimem\\.platformio\\.cache\\tmp\\pkg-installing-5v0888hr']
```
Does anybody recognise these error and have a fix or workaround?

earlephilhower, 2023-03-06:
Look at the latest version of the PIO config file from @maxgerhardt  https://github.com/platformio/platform-raspberrypi/pull/36

It's got a comment about enabling long paths on Windows, and it looks like what you're hitting.

maxgerhardt, 2023-03-06:
Yep.  I also opened an issue about that in TinyUSB.

earlephilhower, 2023-03-06:
Would it be possible to do a shallow `git clone`?  We only need the core's submodules and the 1st level of submodules in `pico-sdk` (like described in the README.md) to build.  i.e.

````
<at arduino-pico top level>
git submodule update --init
cd pico-sdk
git submodule update --init
cd ..
````

That will get rid of the extra LWIP and a bunch of board support packages in TinyUSB.  Much faster and smaller.

earlephilhower, 2023-03-06:
Did you use the latest (updated yesterday) PIO script?  There is a `git config` command that was added and I think is required to be run to make the git executable allow long paths.

Also, your files look like it's some other app or toolchain.  Is this the same thing or something else?  I know there's no Linux patches in the repos we use...

jimemo, 2023-03-06:
I found that git config setting after I posted and reran everything. Ran this as administrator in powershell and rebooted:
`git config --system core.longpaths true`
To summarise you'll need both the windows registry setting and the git config to solve this.

closed 2023-03-06 by jimemo

jimemo, 2023-03-06:
> Also, your files look like it's some other app or toolchain. Is this the same thing or something else? I know there's no Linux patches in the repos we use...

Here is that section of the build output:
```
Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/ti': checked out '143ed6cc20a7615d042b03b21e070197d473e6e5'
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0001-RTT-VMM-implement-dual-system-running-on-realview-pb.patch: Filename too long
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0002-arm-gic-correct-the-cpu-map-on-gic_raise_softirq-for.patch: Filename too long
fatal: Unable to checkout '17761f5cf9dbbf2dcf665b7c04934188add20082' in submodule path 'pico-sdk/lib/tinyusb/hw/mcu/wch/ch32v307'
Submodule path 'pico-sdk/lib/tinyusb/lib/CMSIS_5': checked out '20285262657d1b482d132d20d755c8c330d55c1f'
```

maxgerhardt, 2023-03-07:
>Would it be possible to do a shallow git clone? We only need the core's submodules and the 1st level of submodules in pico-sdk (like described in the README.md) to build. i.e.

Sadly when I put the .git link to Arduino-Pico in the platform.json, PlatformIO will always perform a `--recursive` clone with no further options to control it. And if I pointed at the .zip file (https://github.com/earlephilhower/arduino-pico/archive/refs/heads/master.zip), the entire pico-sdk folder is empty, this would probably fail compilation with TinyUSB and a few other things :/ 

Also tracked in https://github.com/hathach/tinyusb/issues/1939.

If that isn't fixed soon (and somehow these changes cascade through all layer layers) I'll need to put a big warning somewhere that Win32 Long Paths and ` git config --system core.longpaths true` is needed for Windows now.