earlephilhower/arduino-pico #1285: Hostname resolution sometimes fails

joba-1, 2023-03-08:
Hi,

I try to post a http request to an influx database with this code

```cpp
    WiFiClient wifiHttp;
    HTTPClient http;
    http.begin(wifiHttp, "job4", 8086, "/write?db=my_db&precision=s");
    http.setUserAgent("test");
    int influx_status = http.POST("some data");
```

on this core:
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipicow

I did a debug session and found it fails to resolve the hostname in WifiClass.cpp#577: WiFiClass::hostByName(const char* aHostname, IPAddress& aResult, int timeout_ms). The correct IP would be 192.168.1.4 but instead it gives me 185.116.114.6

I also use this same host as syslog server and mqtt broker. The Syslog and PubSubClient libraries can resolve the hostname just fine.

joba-1, 2023-03-08:
I did some more tests: it seems it starts to fail only after some successful calls.

joba-1, 2023-03-08:
A more simple test program like this works just fine even after 500 resolves of "job4". Strange...

```cpp
/* 
WiFi on a Raspberry Pico W: Serial1 stops on WiFi.begin()!?
*/

#include <Arduino.h>

#define HEALTH_LED_PIN 9

#include <WiFi.h>

// Get wlan data from eeprom (written by PicoWlan firmware)
#include <EEPROM.h>
const uint32_t eeprom_magic = 0xeeda4aee;  // as written by PicoWlan
struct eeprom_data {
  uint32_t magic;
  char ssid[32];
  char pass[32];
  uint32_t sum;  // PicoWlan simply adds bytes of magic, ssid and pass 
} data;

void setup() {
    pinMode(HEALTH_LED_PIN, OUTPUT);
    digitalWrite(HEALTH_LED_PIN, HIGH);

    Serial1.begin(BAUDRATE);
    Serial1.println("\r\nStarting " PROGNAME " v" VERSION " " __DATE__ " " __TIME__);

    EEPROM.begin(512);
    EEPROM.get(0, data);
    EEPROM.end();

    for (uint8_t *ptr=(uint8_t *)&data; ptr<(uint8_t *)&data.sum; ptr++) {
        data.sum -= *ptr;
    }

    if (data.magic != eeprom_magic || data.sum) {
        Serial1.println("EEPROM data invalid. Write with PicoWlan firmware");
    }

    WiFi.mode(WIFI_STA);
    String host(HOSTNAME);
    host.toLowerCase();
    WiFi.setHostname(host.c_str());

    Serial1.printf("Connecting '%s' to ssid '%s'\r\n", WiFi.getHostname(), data.ssid);
    WiFi.begin(data.ssid, data.pass);
    while (WiFi.waitForConnectResult(10000) != WL_CONNECTED) {
        Serial1.println("Connection Failed! Rebooting...");
        delay(100);
        rp2040.restart();
    }
    Serial1.printf("Got IP %s\r\n", WiFi.getHostname(), WiFi.localIP().toString().c_str());

    digitalWrite(HEALTH_LED_PIN, LOW);
}

void loop() {
    const char host[] = "job4";
    IPAddress ip;
    int rc = WiFi.hostByName(host, ip, 1000);

    static unsigned count = 0;
    count++;
    Serial1.printf("%u: Alive rc %d ip %s\r\n", count, rc, ip.toString().c_str());
    digitalWrite(HEALTH_LED_PIN, digitalRead(HEALTH_LED_PIN) ? LOW : HIGH);
    delay(1000);
}
```

earlephilhower, 2023-03-08:
> I did a debug session and found it fails to resolve the hostname in WifiClass.cpp#577: WiFiClass::hostByName(const char* aHostname, IPAddress& aResult, int timeout_ms). The correct IP would be 192.168.1.4 but instead it gives me 185.116.114.6

Can you clarify, are you saying the DNS response to "job4" **did not come back**, or that it **came back as different**?

DNS is common throughout LWIP, so I think it's possible you have 2 DNS servers specified on your connection.  When the 1st one works, it gives your local CNAME A address, but when the 1st one times out or is slow the 2nd one is queried and it returns the other address which is someone else's IP.

Have you tried manually fixing it to only have 1 DNS server, pointing only to your internal one?

joba-1, 2023-03-08:
The POST() comes back with connection failed (-1).
The hostByName() comes back with the wrong IP (I have not checked the int returncode ... yet, but I dont really see a reason why it should fail after succeeding twice with the same hostname). 
I use only one dns server in my local net (my local openwrt wlan router). What is maybe a bit unusual is that it is not the same as the default gateway. But the net works perfectly fine this way since over a year without config changes for all other clients, including many ESPs. 
I'll push the actual code to my github repo soon in case you want to have a look...

joba-1, 2023-03-08:
This fails:

https://github.com/joba-1/PicoPresence/blob/master/src/wifi.cpp#L101

Producing this error message:
```
Post job4:8086/write?db=PicoPresence&precision=s status=-1 line='Wifi,Host=picopresence-1,Version=0.1 BSSID="28:d1:27:4e:6b:9d",
IP="192.168.1.134",RSSI=-46' response=''
```

if I change the influx hostname from "job4" to "192.168.1.4" it works!

earlephilhower, 2023-03-08:
Thanks for the link, but that's something that can't run outside of your network and it's got a ton of dependencies.

DNS is fundamental to the operation of the WiFi core and if broken WiFi would be basically unusable for folks.  That doesn't seem to be the case, because your one app seems to be the only one hitting this.  I'm inclined to think this is something within your app or network and not something we can help with here.

If you could capture the actual DNS query and response (wireshark/etc.), the actual returned value/error/etc. from the DNS lookup, etc. when it fails in your environment that would help.  Also, please manually set a single DNS server (`WiFi.setDNS(xxx)`) to ensure you're not leaking onto another server somehow via DHCP-set options.

FWIW DNS lookups are stored in a cache in the LWIP state, so it's quite possible one read could hit the cache and the next loop, due to some other operation using DNS, could go back out on the wire.

joba-1, 2023-03-08:
I added a print statement to hostByName() 

```cpp
int WiFiClass::hostByName(const char* aHostname, IPAddress& aResult, int timeout_ms) {
    int rc = _wifi.hostByName(aHostname, aResult, timeout_ms);
    Serial1.printf("WiFiClass::hostByName(%s, %s) = %d\n", aHostname, aResult.toString().c_str(), rc);
    return rc;
}
```

and get this (the resolve fails, after it succeeded twice) :

```
Starting PicoPresence v0.1 Mar  8 2023 21:26:09
Connecting to 'job.fritz.ssid'
Host 'picopresence-1' IP: 192.168.1.134
WiFiClass::hostByName(fritz.box, 192.168.1.221) = 1
Firmware update on http://picopresence-1:80/update
WiFiClass::hostByName(job4, 192.168.1.4) = 1
Connected to MQTT broker job4:1883 using topic PicoPresence/1
WiFiClass::hostByName(job4, 192.168.1.4) = 1
{"Version":0.1,"Hostname":"picopresence-1","Wifi":{"BSSID":"28:d1:27:4e:6b:9d","IP":"192.168.1.134","RSSI":-48}}
WiFiClass::hostByName(job4, (IP unset)) = 0
WiFiClass::hostByName(job4, (IP unset)) = 0
Post job4:8086/write?db=PicoPresence&precision=s status=-1 line='Wifi,Host=picopresence-1,Version=0.1 BSSID="28:d1:27:4e:6b:9d",
IP="192.168.1.134",RSSI=-48' response=''
...
```
now fails every time until restart...

earlephilhower, 2023-03-08:
Are you absolutely sure your router adds the proper local domain to the name lookup?  `job4` is not a FQDN AIUI.

What could have happened is that your code always failed, but because the RAM storing that address was set by some other call the leftover data was "correct."  So you'd still go where you expected.  But, now due to the stack/heap layout being changed that's no longer the case.

joba-1, 2023-03-08:
> I'm inclined to think this is something within your app or network

I understand that.  

WiFi.setDNS(IPAddress(192, 168, 1, 236)); called right after connect gives the same result :(
The log above shows every single call to hostByName(). 

I can't see how it can succeed twice and then always fail other than my DNS server or the  _wifi.hostByName() freak out. Since my DNS works fine for everything else but also see this should be more common for other pico users if it is the wifi core I won't bet on anything...

I'll try the wireshark/tcpdump route.

joba-1, 2023-03-08:
> Are you absolutely sure your router adds the proper local domain to the name lookup? `job4` is not a FQDN AIUI.

I also tried with the fqdn job4.job. Same result

earlephilhower, 2023-03-08:
DNS lookup is handled here, basically completely inside LWIP:

https://github.com/earlephilhower/arduino-pico/blob/e6c7b97b5e395f967d6d317c2509d902fccf5846/libraries/lwIP_Ethernet/src/LwipIntfDev.h#L154-L203

https://www.nongnu.org/lwip/2_1_x/group__dns.html

The core sends an async request to LWIP which will then either, on success call the callback fcn which does the actual IP address set , or on failure return false and not touch the IP address pointer passed in.

joba-1, 2023-03-08:
This might explain why it works if I pass in an ip address string :)

https://github.com/earlephilhower/arduino-pico/blob/e6c7b97b5e395f967d6d317c2509d902fccf5846/libraries/lwIP_Ethernet/src/LwipIntfDev.h#L172-L175

joba-1, 2023-03-09:
dns_gethostbyname(job4.job, (IP unset)) = -1  => ERR_MEM (Out of memory error)

earlephilhower, 2023-03-09:
Interesting!  LWIP has a fixed memory buffer assigned to it that it manages itself (since it's IRQ based, it can't use malloc()).

I've added 50% more RAM (24K, up from 16K) to #1288.  Can you give it a try?

joba-1, 2023-03-09:
I‘ll check soon, once I know how to use pr’s as platform

earlephilhower, 2023-03-09:
Also, please make sure you're on latest `git` head before applying because there was a LWIP functionality update, too. (#1286)

joba-1, 2023-03-10:
Unfortunately unchanged behavior. These are all calls to dns_gethostbyname() since boot:
```
dns_gethostbyname(>fritz.box, <192.168.1.221) = 0
dns_gethostbyname(>job4, <192.168.1.4) = 0
dns_gethostbyname(>job4, <192.168.1.4) = 0
dns_gethostbyname(>job4, <(IP unset)) = -1
...
```

For me this looks like a problem with the cache algo.
Imo it should not query the same host twice, and when it does (and maybe even adds the second result to the cache?), things finally break.

Still strange that it happens to just me...

I used these entries in platformio.ini to build (i.e. just added the platform_packages line):
```
[env]
framework = arduino
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages = framework-arduinopico @ https://github.com/earlephilhower/arduino-pico.git#ipfreely
board = rpipicow
...
```

I made sure the ~/platformio/packages/framework-arduinopico is the only pico framework, contains your changes for ram and pcb and I removed all previous build artifacts (rm -rf .pio).

earlephilhower, 2023-03-10:
Can you also check the TTL on the DNS response for `job4`?  Maybe it's 0 or something weird that's not laying well and hence the multiple calls for the same address?

joba-1, 2023-03-10:
I definitely use just core 0. 
I plan to use core 1 for sensor measurement once the infrastructure is settled, but for now setup1() and loop1() are commented out completely.

TTL from pico is 255 and back is 64

Did some more testing before you wrote and can confirm: It is not the lwip dns cache as such, the 4th "real" resolve in lwip always fails.

I wrote my own cache in framework-arduinopico/libraries/lwIP_Ethernet/src/LwipIntfDev.h to make sure each hostname is only resolved once inside lwip -> My code using just two hostnames worked fine.

Now I used different hostnames for mqtt, syslog and influx, so with ntp 4 hostnames -> 3 hosts are resolved and the connections using them work fine. The 4th resolve does not lead to a tcpdump entry and fails. All other hostnames still worked after that using ips from my cache.

The modified function:
```cpp
template<class RawDev>
int LwipIntfDev<RawDev>::hostByName(const char* aHostname, IPAddress& aResult, int timeout_ms) {
    ip_addr_t addr;
    aResult = static_cast<uint32_t>(0xffffffff);

    if (aResult.fromString(aHostname)) {
        // Host name is a IP address use it!
        return 1;
    }

    // local host=>ip cache for testing
    static const size_t HOST_SIZE = 10;
    static const size_t HOST_ENTRIES = 5;
    typedef struct host_ip { char host[HOST_SIZE]; ip_addr_t ip; } host_ip_t;
    static host_ip_t hosts[HOST_ENTRIES] = { 0 };
    static size_t host_ips = 0;
    if (aHostname[0] && strlen(aHostname) < HOST_SIZE) {
        for (size_t i=0; i<host_ips; i++) {
            if (strcmp(aHostname, hosts[i].host) == 0) {
                aResult = hosts[i].ip;
                Serial1.printf("!!! use host #%u: '%s' with ip '%s'\n", i, hosts[i].host, aResult.toString().c_str());
                return 1;
            }
        }
    }

    _dns_cb_t cb = { &aResult, this };
#if LWIP_IPV4 && LWIP_IPV6
    err_t err = dns_gethostbyname_addrtype(aHostname, &addr, &_dns_found_callback, &cb, LWIP_DNS_ADDRTYPE_DEFAULT);
#else
    err_t err = dns_gethostbyname(aHostname, &addr, &_dns_found_callback, &cb);
#endif
    if (err == ERR_OK) {
        aResult = IPAddress(&addr);
    } else if (err == ERR_INPROGRESS) {
        _dns_lookup_pending = true;
        uint32_t now = millis();
        while ((millis() - now < (uint32_t)timeout_ms) && _dns_lookup_pending) {
            sys_check_timeouts();
            delay(10);
        }
        _dns_lookup_pending = false;
        if (aResult.isSet()) {
            err = ERR_OK;
        }
    }

    if (err == ERR_OK) {
        // add host=>ip entry to cache
        if (host_ips < HOST_ENTRIES && aHostname[0] && strlen(aHostname) < HOST_SIZE) {
            strcpy(hosts[host_ips].host, aHostname);
            hosts[host_ips].ip = (ip_addr_t)aResult;
            Serial1.printf("!!! add host #%u: '%s' with ip '%s'\n", host_ips, hosts[host_ips].host, IPAddress(hosts[host_ips].ip).toString().c_str());
            host_ips++;
        }

        return 1;
    }

    return 0;
}
```

I also did another test sketch that just resolves a variable amount of hosts (up to 11) in a loop -> all fine 

I added a http get to that sketch to have some real traffic similar to the influx request besides dns -> still all fine

I don't get it...

joba-1, 2023-03-10:
> can confirm: It is not the lwip dns cache

Just to clarify it is still strange that one host is queried twice. This doesn't happen when my host-cache is active.

joba-1, 2023-03-10:
I noticed this checksum error from the dns to the pico in tcpdump:
```
19:42:03.939638 IP (tos 0x0, ttl 255, id 11, offset 0, flags [none], proto UDP (17), length 50)
    picopresence-1.job.22144 > ax1.53: [udp sum ok] 46512+ A? job4. (22)
19:42:03.939911 IP (tos 0x0, ttl 64, id 58446, offset 0, flags [DF], proto UDP (17), length 66)
    ax1.53 > picopresence-1.job.22144: [bad udp cksum 0x8502 -> 0xbf58!] 46512* q: A? job4. 1/0/0 job4. [0s] A 192.168.1.4 (38)
```
So I switched to another dns.

The checksum error is gone, the second resolve of job4 is also gone. But the resolve of further requests fails anyways.
Looks like we were dealing with two different errors.

joba-1, 2023-03-10:
The checksum error of the dnsmasq dns on the openwrt box is also present for all other devices, but they don't bother. 

Wireshark is less harsh (and probably more precise) with its interpretation: 

    Header Checksum: 0x0ba7 [validation disabled]

```
Frame 6: 80 bytes on wire (640 bits), 80 bytes captured (640 bits)
    Encapsulation type: Ethernet (1)
    Arrival Time: Mar  8, 2023 23:27:54.486177000 CET
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1678314474.486177000 seconds
    [Time delta from previous captured frame: 0.000228000 seconds]
    [Time delta from previous displayed frame: 0.000228000 seconds]
    [Time since reference or first frame: 0.155732000 seconds]
    Frame Number: 6
    Frame Length: 80 bytes (640 bits)
    Capture Length: 80 bytes (640 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: eth:ethertype:ip:udp:dns]
    [Coloring Rule Name: UDP]
    [Coloring Rule String: udp]
Ethernet II, Src: BeijingX_70:e3:ef (28:d1:27:70:e3:ef), Dst: Raspberr_07:6a:f0 (28:cd:c1:07:6a:f0)
    Destination: Raspberr_07:6a:f0 (28:cd:c1:07:6a:f0)
        Address: Raspberr_07:6a:f0 (28:cd:c1:07:6a:f0)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Source: BeijingX_70:e3:ef (28:d1:27:70:e3:ef)
        Address: BeijingX_70:e3:ef (28:d1:27:70:e3:ef)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Type: IPv4 (0x0800)
Internet Protocol Version 4, Src: 192.168.1.236, Dst: 192.168.1.134
    0100 .... = Version: 4
    .... 0101 = Header Length: 20 bytes (5)
    Differentiated Services Field: 0x00 (DSCP: CS0, ECN: Not-ECT)
        0000 00.. = Differentiated Services Codepoint: Default (0)
        .... ..00 = Explicit Congestion Notification: Not ECN-Capable Transport (0)
    Total Length: 66
    Identification: 0xaa41 (43585)
    Flags: 0x40, Don't fragment
        0... .... = Reserved bit: Not set
        .1.. .... = Don't fragment: Set
        ..0. .... = More fragments: Not set
    ...0 0000 0000 0000 = Fragment Offset: 0
    Time to Live: 64
    Protocol: UDP (17)
    Header Checksum: 0x0ba7 [validation disabled]
    [Header checksum status: Unverified]
    Source Address: 192.168.1.236
    Destination Address: 192.168.1.134
User Datagram Protocol, Src Port: 53, Dst Port: 5321
    Source Port: 53
    Destination Port: 5321
    Length: 46
    Checksum: 0x8502 [unverified]
    [Checksum Status: Unverified]
    [Stream index: 2]
    [Timestamps]
        [Time since first frame: 0.000228000 seconds]
        [Time since previous frame: 0.000228000 seconds]
    UDP payload (38 bytes)
Domain Name System (response)
    Transaction ID: 0x5b43
    Flags: 0x8580 Standard query response, No error
        1... .... .... .... = Response: Message is a response
        .000 0... .... .... = Opcode: Standard query (0)
        .... .1.. .... .... = Authoritative: Server is an authority for domain
        .... ..0. .... .... = Truncated: Message is not truncated
        .... ...1 .... .... = Recursion desired: Do query recursively
        .... .... 1... .... = Recursion available: Server can do recursive queries
        .... .... .0.. .... = Z: reserved (0)
        .... .... ..0. .... = Answer authenticated: Answer/authority portion was not authenticated by the server
        .... .... ...0 .... = Non-authenticated data: Unacceptable
        .... .... .... 0000 = Reply code: No error (0)
    Questions: 1
    Answer RRs: 1
    Authority RRs: 0
    Additional RRs: 0
    Queries
        job4: type A, class IN
            Name: job4
            [Name Length: 4]
            [Label Count: 1]
            Type: A (Host Address) (1)
            Class: IN (0x0001)
    Answers
        job4: type A, class IN, addr 192.168.1.4
            Name: job4
            Type: A (Host Address) (1)
            Class: IN (0x0001)
            Time to live: 0 (0 seconds)
            Data length: 4
            Address: 192.168.1.4
    [Request In: 5]
    [Time: 0.000228000 seconds]
```

earlephilhower, 2023-03-10:
Maybe this is why it's requesting DNS resolve on `job4` twice?  TTL = 0?
````
Answers
        job4: type A, class IN, addr 192.168.1.4
            Name: job4
            Type: A (Host Address) (1)
            Class: IN (0x0001)
=======>    Time to live: 0 (0 seconds)
            Data length: 4
            Address: 192.168.1.4
````

joba-1, 2023-03-10:
ah, I only looked at packet ttl, didn‘t know dns has another one. Could well be the (good) reason for another query

joba-1, 2023-03-13:
I put some print messages in the lwip code and get this:

```
LwipIntfDev<RawDev>::hostByName(fritz.box): dns_gethostbyname_addrtype(fritz.box): dns_lookup(fritz.box): NOT found dns_enqueue(fritz.box): in progress 
LwipIntfDev<RawDev>::hostByName(fritz.box): -> ok
LwipIntfDev<RawDev>::hostByName(job4): dns_gethostbyname_addrtype(job4): dns_lookup(job4): NOT found dns_enqueue(job4): in progress 
LwipIntfDev<RawDev>::hostByName(job4): -> ok
LwipIntfDev<RawDev>::hostByName(job4.): dns_gethostbyname_addrtype(job4.): dns_lookup(job4.): NOT found dns_enqueue(job4.): in progress 
LwipIntfDev<RawDev>::hostByName(job4.): -> ok
LwipIntfDev<RawDev>::hostByName(job4.): dns_gethostbyname_addrtype(job4.): dns_lookup(job4.): NOT found dns_enqueue(job4.): no pcb -> stop 
LwipIntfDev<RawDev>::hostByName(job4.): -> error
LwipIntfDev<RawDev>::hostByName(job4.job): dns_gethostbyname_addrtype(job4.job): dns_lookup(job4.job): NOT found dns_enqueue(job4.job): no pcb -> stop 
LwipIntfDev<RawDev>::hostByName(job4.job): -> error
LwipIntfDev<RawDev>::hostByName(job4.): dns_gethostbyname_addrtype(job4.): dns_lookup(job4.): NOT found dns_enqueue(job4.): no pcb -> stop 
LwipIntfDev<RawDev>::hostByName(job4.): -> error
```
It always starts failing at the 4th request, which is the same hostname as already found -> should use cache but doesn't. Instead tries another dns_enqueue and fails.

I have not investigated on these results yet. I thought maybe you already have an idea what is going on there?

joba-1, 2023-03-13:
Couldn't find how to properly use LWIP debug messages and put them on Serial1, so I cooked my own. Maybe it helps you to understand the exact origin of the messages, if needed.

```diff
diff --git a/libraries/lwIP_Ethernet/src/LwipIntfDev.h b/libraries/lwIP_Ethernet/src/LwipIntfDev.h
index 23c133b..4298570 100644
--- a/libraries/lwIP_Ethernet/src/LwipIntfDev.h
+++ b/libraries/lwIP_Ethernet/src/LwipIntfDev.h
@@ -53,6 +53,8 @@
 
 extern "C" void cyw43_hal_generate_laa_mac(__unused int idx, uint8_t buf[6]);
 
+extern "C" void push(const char *msg);
+
 template<class RawDev>
 class LwipIntfDev: public LwipIntf, public RawDev {
 public:
@@ -166,11 +168,16 @@ void LwipIntfDev<RawDev>::_dns_found_callback(const char *name, const ip_addr_t
 
 template<class RawDev>
 int LwipIntfDev<RawDev>::hostByName(const char* aHostname, IPAddress& aResult, int timeout_ms) {
+    push("LwipIntfDev<RawDev>::hostByName(");
+    push(aHostname);
+    push("): ");
+
     ip_addr_t addr;
     aResult = static_cast<uint32_t>(0xffffffff);
 
     if (aResult.fromString(aHostname)) {
         // Host name is a IP address use it!
+        push("name is ip -> ok\n");
         return 1;
     }
 
@@ -182,6 +189,7 @@ int LwipIntfDev<RawDev>::hostByName(const char* aHostname, IPAddress& aResult, i
 #endif
     if (err == ERR_OK) {
         aResult = IPAddress(&addr);
+        push("from cache ");
     } else if (err == ERR_INPROGRESS) {
         _dns_lookup_pending = true;
         uint32_t now = millis();
@@ -195,10 +203,16 @@ int LwipIntfDev<RawDev>::hostByName(const char* aHostname, IPAddress& aResult, i
         }
     }
 
+    push("\nLwipIntfDev<RawDev>::hostByName(");
+    push(aHostname);
+    push("): ");
+    
     if (err == ERR_OK) {
+        push("-> ok\n");
         return 1;
     }
 
+    push("-> error\n");
     return 0;
 }
```
```diff
diff --git a/src/core/dns.c b/src/core/dns.c
index e84bab24..d5fb8047 100644
--- a/src/core/dns.c
+++ b/src/core/dns.c
@@ -97,6 +97,8 @@
 
 #include <string.h>
 
+void push(const char *msg);
+
 /** Random generator function to create random TXIDs and source ports for queries */
[... 175 lines cut ...]
 }
 
+#include <queue>
+
+std::queue<String> msgs;
+
+extern "C" {
+    void push(const char *msg) {
+        msgs.push(msg);
+    }
+}
 
 void loop() {
     bool health = true;
@@ -415,8 +423,8 @@ void loop() {
     MDNS.update();
     handle_reset();
 
+    while(!msgs.empty()) {
+        Serial1.print(msgs.front());
+        msgs.pop();
+    }
 }

```

earlephilhower, 2023-03-13:
Enabling LWIP debug needs editing the `lwipopts.h` in `tools/libpico` and rebuilding the libs, so what you did is a good compromise.

If I quickly trace the logic in DNS what seems to be happening in your case is the DNS connection PCBs (statically allocated to == # of DNS cache entries == 2, IIRC) are set but not cleared (in time? or never? not sure) so that the DNS logic can't go out on the wire after a few tries.

Are all your DNS queries (HTTP connection requests in this case since `job4` can't be DNS cached) right after each other, or are they separated by many seconds?  There may be a timeout like the TCP reuse timeout under the hood that's not firing before you start over.

That would seem to imply, though, that eventually you'd get a successful one or two requests every now and then (after one of the 2 DNS PCBs time out), and not always 100% of the time forever fail.

joba-1, 2023-03-13:
During startup queries to all hostnames are fired quickly. Later in the loop only if data changes or every 10s (on average every 2s)

joba-1, 2023-03-13:
I did not notice successful requests - I think I should have noticed at least sporadic syslog messages, but I didn't

joba-1, 2023-03-13:
Some new info: 

I wrote a very stripped down program that only does name resolution on 4 hosts every second. No network traffic except for the dns requests. Only fritz.box gets cached, no matter if it is the first or last entry. All others (job4.job, job4, ...) query the server every time (and always successful).

I didn't notice this until now, because in the other program the first request is the ntp server fritz.box which is never used again.

From the code I got the impression the cache should hold 4 entries.
#define DNS_TABLE_SIZE 4

earlephilhower, 2023-03-13:
> All others (job4.job, job4, ...) query the server every time (and always successful).

This is expected as the DNS TTL = 0, so caching the response isn't allowed.

In any case, it seems like all the small samples you build pass, and only the full app is dying, yes?

joba-1, 2023-03-13:
yes.

joba-1, 2023-03-14:
Back to the big app: I think there is a relation to udp (or even ip in general) and maybe not specific to dns or name caching:

I use udp with dns, ntp (core) and syslog (external lib) and tcp with mqtt and http -> this group causes the error

if I deactivate syslog -> all fine (syslog is the most used network service)
if I activate another ntp connection (with external library NTPClient) -> error is back even without syslog (problems start with ntp.begin())
If I do or don't do mqtt or http requests seems to make no difference.

joba-1, 2023-03-14:
recompiled with `#define MEMP_NUM_UDP_PCB            6` (default 4) in lwipopts.h and all is fine :)

earlephilhower, 2023-03-14:
That's great news. I'll do a rebuild for the next release.  I'm still not sure it's fixing vs. masking a problem, but it's a microscopic memory impact.

joba-1, 2023-03-14:
I agree, it should at least recover on further retries - and it is strange it failed already after the third request when it should have four pcbs. Looks more like masking.

earlephilhower, 2023-03-14:
Unfortunately, given the unreproducibility of the issue outside of one app I'm not sure we can get much better than masking.  In any case, the PR doesn't do anything hackish, and adds only 10s of bytes of RAM, so it's not going to cause any trouble later on no matter what.

PR #1304 (2023-03-14): Increase UDP PCBs to avoid DNS OOM error
Fixes #1285 (or at least works well enough for now)
commit ebe5bfbc (2023-03-14): Increase UDP PCBs to avoid DNS OOM error (#1304)

Fixes #1285 (or at least works well enough for now)
closed 2023-03-14 by earlephilhower