earlephilhower/arduino-pico #1287: Possibility to reserve BCLK and LRCLK order

ruifig, 2023-03-08:
Right now, I2S requires BCLK followed by LRCLK.

Seems like some devices need this in reverse. LRCLK followed by BCLK.
E.g, Rev2.1 of Pico-Audio expansion module ([https://www.waveshare.com/wiki/Pico-Audio]()
There is some confusion in the documentation, but looking at their examples, it does seem they use reserve order.

Fyi, Rev2.1 of the board does require MCLK, which I am trying to use (from issue #647 ).
Once I have some free time I'll set it up on a breadboard (instead of stacked) and manually swap those pins to see if that's indeed the case.

earlephilhower, 2023-03-08:
Sorry, I don't quite parse your request.  What do you mean by LRCLK following BCLK?  Can you post some timing diagrams with dependencies?

I2S is a well-spec'd protocol so if they've moved the clock edge for things, or number of clocks, then it's not I2S...

ruifig, 2023-03-08:
Zero experience with PIO and electronics in general.
I'm a software engineer that recently (ish) started learning electronics as a hobby.
I'll try  to get it working as-is on a breadboard first, and go from there. Might give it a go with PIO for the first time. :)

ruifig, 2023-03-09:
Got the Pico-Audio board working on a breadboard with the MCLK's gist and reversing the LRCLK and BCLK pins.
I'll give it a go at changing the PIO programs.

earlephilhower, 2023-03-09:
It's trivial to hack it in.  Just reverse the `side AB` bits at the end of the insn in the PIO program to `side BA`, re-`pioasm` it, and off you go.

https://github.com/earlephilhower/arduino-pico/blob/e6c7b97b5e395f967d6d317c2509d902fccf5846/libraries/I2S/src/pio_i2s.pio#L29-L42

Making it into an API (i.e. `I2S.swapClocks()`) is also pretty mechanical.

ruifig, 2023-03-09:
Yes, that did the trick. 
Quickly skimmed over the RP2040 and I2S protocol documentation to actually understand the parts I needed.

earlephilhower, 2023-03-11:
@ruifig can you give #1289 a try and report back?

ruifig, 2023-03-13:
For a moment you lost me there. You linked to another issue. :)
But I see you have a pull request: https://github.com/earlephilhower/arduino-pico/pull/1298
I'll take a look tonight after work.

ruifig, 2023-03-15:
Finally got around to test https://github.com/earlephilhower/arduino-pico/pull/1298 .
It works fine. :)

earlephilhower, 2023-03-15:
Ah, thanks for the update!

closed 2023-03-15 by earlephilhower

earlephilhower, 2024-01-12:
Call it before the `begin`.

DoomHammer, 2024-01-12:
So my code looks like this:

```

// GPIO pin numbers
#define pDOUT 22
#define pBCLK 27
#define pWS (pBCLK+1)
#define pMCLK 26  // optional MCLK pin

const int frequency = 440; // frequency of square wave in Hz
const int sampleRate = 16000; // minimum for many i2s DACs
const int bitsPerSample = 16;
const int amplitude = 1 << (bitsPerSample - 2); // amplitude of square wave = 1/2 of maximum

#define MCLK_MUL  256     // depends on audio hardware. Suits common hardware.
```

then

```
  i2s.swapClocks();
  i2s.setBCLK(pBCLK);
  i2s.setDATA(pDOUT);
  i2s.setMCLK(pMCLK); // must be run before setFrequency() and i2s.begin()
  i2s.setMCLKmult(MCLK_MUL); // also before i2s.begin()
  i2s.setBitsPerSample(16);
```

(everything else like in example)

And it doesn't work. Same Waveshare board V2.1 as @ruifig mentioned.

ruifig, 2024-01-12:
@DoomHammer I haven't touched that in a very long time unfortunately. For me, electronics is a hobby, and my work load since ~June meant no electronics at all. :(
If that wasn't enough, I'm not sure I actually have that work saved. I got attacked by Ransomware back in early December and lost some stuff I didn't have backed up. Lesson learned is I shouldn't leave RDP open. ;)
I'll take a look at some point this weekend to see if I still have that code.

DoomHammer, 2024-01-13:
No worry. Whenever you can check, that'd be fine. I'm just trying to make it working and you seem to be the only person who was able to do so :D

Sorry about the attack.

ruifig, 2024-01-14:
> No worry. Whenever you can check, that'd be fine. I'm just trying to make it working and you seem to be the only person who was able to do so :D
> 
> Sorry about the attack.

@DoomHammer , looks like I had a copy on my laptop.
I've commit all I had to https://github.com/ruifig/czmicroaudio/commits/dev/   (the `dev` branch)

But I'm fairly certain it was under heavy refactoring, because I was porting one of my old audio libraries (from the 90s, hehe). 
But some things you can look at:
`src\scratchpad.h`
`src\MCLK\*`
`lib\src\crazygaze\microaudio\output\arduino`

Sorry I can't be of more help. :(
I don't remember the state of that, so very unlikely you'll figure it out from my code either.

DoomHammer, 2024-01-15:
Thanks, that should help me get some direction!

PR #1298 (2023-03-15): Add I2S::swapClocks() for boards w/reversed pins
Allow users of boards like the Pico-Audiom where the LRCLK comes before the BCLK pin, to swap the BCLK/LRCLK of the I2S interface.

Fixes #1287
commit 3dbe5cf9 (2023-03-15): Add I2S::swapClocks() for boards w/reversed pins (#1298)

Allow users of boards like the Pico-Audiom where the LRCLK comes
before the BCLK pin, to swap the BCLK/LRCLK of the I2S interface.

Fixes #1287