earlephilhower/arduino-pico #1289: Pico crashing with Serial1.readString()

drakejest, 2023-03-09:
My pico is crashing and i could not figure out why, the only culprit i can blame is Serial1.readString(). Its not just a normal crash, its a total crash, core 2 stops also, and often on some crash pico becomes usb unrecognizable, repowering the pico (unplugging the usb then pluggin again) will not make it recover. It has to be uploaded with new code through BOOTSEL button.

At first serial monitor stop recieving prints from the pico and  the led from core 2 also stops blinking either stuck in on or off position.
For now this is the most minimum code i can make it reliably crash. the number it prints is also not the same with each crash.

```
void setup1() {
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
}

void loop1() {
  digitalWrite(LED_BUILTIN, HIGH);
  delay(250);
  digitalWrite(LED_BUILTIN, LOW);
  delay(250);
}

void setup() {
  pinMode(resetPin, OUTPUT);
  Serial.begin(115200);

  while (!Serial)
    ;

  Serial1.setTX(0);
  Serial1.setRX(1);
  Serial1.setFIFOSize(512);
  Serial1.setTimeout(1000);
  Serial1.begin(9600);

  digitalWrite(resetPin, LOW);
  delay(1000);
  digitalWrite(resetPin, HIGH);
  delay(1000);

  Serial.println("   starting  : ");
}

int num = 0;

void loop() {

String str = Serial1.readString();
Serial.println(num);
num++;
}
```

Usually with this code mine crashes at 4,5,6,7 as its last print, no further. I also attempted to just not open the serial monitor and see if the LED stops blinking, and so far it does not stop after 5minutes. Thus me concluding that either `print` or `readstring()` is causing the crash. I commented out `Serial.println(num);` (I still have to open serial monitor in order to let it proceed in the loop). and yes it crashed. So `readstring()` is the only one left to blame

Core v3.0.0
Arduino IDE 1.18.19 and Arduino IDE 2.0.4

earlephilhower, 2023-03-09:
Looking at the code there doesn't seem to be any obvious connection between the cores.  It also looks like you need external HW pumping in bits on Serial1, so probably not reproducible outside of your setup.

Can you connect a picoprobe after the crash and dump `thread 0` `where` `thread 1` `where` to give a stack dump for both cores to help pinpoint things?

Also, FWIW, a 512 byte serial FIFO's pretty enormous.  Any particular reason?  I don't think there are any assumptions about the actual size in the code but it's odd to see something so large...

drakejest, 2023-03-09:
Now that you mentioned it i tried disconnecting the hardware on GP 0,1,GND, and 2 and it does still crash  but at a higher number (in the image 35 is the last one). but core 2 does not, it continues to blink. I currently have no second pico, so hopefully you can recreate it with just the code. will try to buy one sometime tomorrow)

[image]

 Weird thing is there is no data going in or out Serial1. With the device back on gp 0,1,gnd, and 2.

I modified loop to this: 

```
void loop() {
String str = Serial1.readString();
Serial.print(num);
Serial.print("\t");
Serial.println(str.length());
num++;
}
```

And it did not crash...

drakejest, 2023-03-09:
I still did make it crash at with the original code. (the count this time was 73)

I have jlink mini edu which can do SWD can this get what you need ?

earlephilhower, 2023-03-09:
As long as you can use it with `openocd` then it should be fine.  Unfortunately I have no idea what the incantation for that specific device would be.  If you have a spare Pico and 3 wires, you can breadboard/jumper things very easily (the picoprobe FW is OSS and available from the RPI folks).

Alternatively, if you have no input on `Serial1` (so that, IIRC, the `Serial1.readline()` will timeout), does that fail too?  If so, I can run that locally, no problem.

drakejest, 2023-03-09:
Can a Pico W be used as the probe? as that would be a much more convenient buy for me

earlephilhower, 2023-03-09:
Sure, a Pico or a PicoW can be used.  The RPI repo has all the info.  Wiring up a picoprobe was the 2nd thing I did once I got a set of 4 Picos a couple years ago. :)

https://github.com/raspberrypi/picoprobe

Although, again, if your SWD device can talk to `openocd` then it should work, too.

earlephilhower, 2023-03-10:
@drakejest with your setup, if you don't connect anything to Serial1.RX pin, does it still crash?  In my own testing w/floating input on Serial1.RX, it runs forever.

drakejest, 2023-03-11:
Sorry fo the late update the picos just arrived, so i bought 5 picos, 2 of them are pico W and 3 ordinary picos. So i uploaded the sketch on all 5 of them, i could not replicate them on the new ones, the old pico in question is still crashing though. The conclusion i can get from this is that, this old picos has had its time (bought it and been used since release). 

I can still try to chase whats wrong with that specific pico if you want, but at the end of the day there seem to be nothing wrong with your core :) its just my pico that is broken

closed 2023-03-11 by drakejest