earlephilhower/arduino-pico #1300: HTTPClient GET hangs

jazztunes, 2023-03-11:
I've got a problem where HTTPClient GET will hang randomly.  I've boiled it down this minimal reproduceable scenario.

This sketch here repeatedly pulls down a ~5.6k file printing the getFreeHeap() value on each iteration. It will stop running in about 20-30 minutes usually. I don't think it's a memory problem because I wrote a version that would malloc and then free 30k before the GET, and the GET would still hang. I've tried different file sizes on various servers and all have the same problem. I have this same basic code running on an ESP32 based board and it's never hung.

Any help tracking this down would help! I'm using v3.0.0 on an RP2040 Pico W.

```
#include <WiFi.h>
#include <HTTPClient.h>

const char* ssid = "xxxx";
const char* password = "xxxx";

WiFiMulti WiFiMulti;
HTTPClient http;

int i=0;

void setup() {
  Serial.begin(38400);
  WiFiMulti.addAP(ssid, password);
}

void loop() {
  Serial.printf("\n---LOOP: %d-----------------------\n",i++);
  if (WiFiMulti.run() != WL_CONNECTED) {
    delay(1000);
    printf("Connecting to Wi-Fi\n");
    return;
  }

  Serial.print("[HTTP] begin...\n");
  Serial.println("FreeHeap begin: "+(String)rp2040.getFreeHeap());
  if (http.begin("https://jsonplaceholder.typicode.com/users")) {
      http.setInsecure();
      Serial.print("[HTTP] GET...");
      
      int httpCode = http.GET();
      Serial.println("OK");

      if (httpCode > 0)
      {
          Serial.printf("[HTTP] Response: %d\n", httpCode);

          if (httpCode == HTTP_CODE_OK)
          {
            String payload = http.getString();              
            Serial.printf("Payload OK: %d bytes\n", payload.length());
          }
      }
      else
      {
          Serial.printf("[HTTP] GET failed, error: %s\n", http.errorToString(httpCode).c_str());
      }

      http.end();
  }
  else
  {
      Serial.print("[HTTP} Unable to connect\n");
  }
  delay(2000);
}
```

Serial output:
```
15:59:05.208 -> ---LOOP: 122-----------------------
15:59:05.208 -> [HTTP] begin...
15:59:05.208 -> FreeHeap begin: 183316
15:59:05.208 -> [HTTP] GET...OK
15:59:06.129 -> [HTTP] Response: 200
15:59:06.223 -> Payload OK: 5645 bytes
15:59:08.235 -> 
15:59:08.235 -> ---LOOP: 123-----------------------
15:59:08.235 -> [HTTP] begin...
15:59:08.235 -> FreeHeap begin: 183316
15:59:08.235 -> [HTTP] GET...OK
15:59:09.171 -> [HTTP] Response: 200
15:59:09.264 -> Payload OK: 5645 bytes
15:59:11.260 -> 
15:59:11.260 -> ---LOOP: 124-----------------------
15:59:11.260 -> [HTTP] begin...
15:59:11.260 -> FreeHeap begin: 183316
15:59:11.260 -> [HTTP] GET...
```

It's clearly getting hung on `int httpCode = http.GET();`.

jazztunes, 2023-03-12:
A little more info. I enabled serial debugging level "all", and I've captured another hang scenario with some more detail. The sz=-1 and the abort seem to be a clue.

```
---LOOP: 183-----------------------
[HTTP] begin...
FreeHeap begin: 182804
[HTTP] GET...:ref 1
:rcl pb=0 sz=-1
:abort
```

Appears that this is coming from `libraries/WiFi/src/include/ClientContext.h`
```
err_t _recv(tcp_pcb* pcb, pbuf* pb, err_t err) {
    (void) pcb;
    (void) err;
    if (pb == 0) {
        // connection closed by peer
        DEBUGV(":rcl pb=%p sz=%d\r\n", _rx_buf, _rx_buf ? _rx_buf->tot_len : -1);
        _notify_error();
        if (_rx_buf && _rx_buf->tot_len) {
            // there is still something to read
            return ERR_OK;
        } else {
            // nothing in receive buffer,
            // peer closed = nothing can be written:
            // closing in the legacy way
            abort();
            return ERR_ABRT;
        }
    }
```

Looks like the connection is aborted because there's nothing in receive buffer.  Is there a way to detect and recover from this gracefully?

earlephilhower, 2023-03-12:
Have you run with the 3.1.0 release yet?  There was a change in the LWIP processing which fixed a few mutex deadlock cases which behave like an infinite loop...

jazztunes, 2023-03-12:
Just realized that ClientContext.h was changed recently to fix issues that may be related. I've pulled down 3.1.0 and I'm trying that now.

jazztunes, 2023-03-12:
Ok, it's still hanging with 3.1.0, but now the symptoms are a little different. It eventually hangs in the GET as before, however, I don't see ":abort" in the debug output. The last thing I get is a few ":rch" lines.

Here's the output from the last three iterations placed side by side for comparison. Loops 287 & 288 were successful and 289 never returned. Everything's basically the same up until line 053 where loop 289 does an ":rd" instead of an ":rch". Other runs had the same pattern.

I hope all this this helpful!

```
LN#  ---LOOP: 287--------------- ---LOOP: 288---------------- ---LOOP: 289-----------
001  [HTTP] begin...             [HTTP] begin...              [HTTP] begin...
002  FreeHeap begin: 183580      FreeHeap begin: 183580       FreeHeap begin: 183580
003  [HTTP] GET...:ref 1         [HTTP] GET...:ref 1          [HTTP] GET...:ref 1
004  :wr 233 0                   :wr 233 0                    :wr 233 0
005  :wrc 233 233 0              :wrc 233 233 0               :wrc 233 233 0
006  :ack 233                    :ack 233                     :ack 233
007  :rn 1460                    :rn 1460                     :rn 1460
008  :rch 1460, 1098             :rch 1460, 1099              :rch 1460, 1098
009  :rd 5, 2558, 0              :rd 5, 2559, 0               :rd 5, 2558, 0
010  :rdi 1460, 5                :rdi 1460, 5                 :rdi 1460, 5
011  :rd 91, 2558, 5             :rd 91, 2559, 5              :rd 91, 2558, 5
012  :rdi 1455, 91               :rdi 1455, 91                :rdi 1455, 91
013  :rd 5, 2558, 96             :rd 5, 2559, 96              :rd 5, 2558, 96
014  :rdi 1364, 5                :rdi 1364, 5                 :rdi 1364, 5
015  :rd 2328, 2558, 101         :rd 2328, 2559, 101          :rd 2328, 2558, 101
016  :rdi 1359, 1359             :rdi 1359, 1359              :rdi 1359, 1359
017  :c 1359, 1460, 2558         :c 1359, 1460, 2559          :c 1359, 1460, 2558
018  :rdi 1098, 969              :rdi 1099, 969               :rdi 1098, 969
019  :rd 5, 1098, 969            :rd 5, 1099, 969             :rd 5, 1098, 969
020  :rdi 129, 5                 :rdi 130, 5                  :rdi 129, 5
021  :rd 115, 1098, 974          :rd 116, 1099, 974           :rd 115, 1098, 974
022  :rdi 124, 115               :rdi 125, 116                :rdi 124, 115
023  :rd 5, 1098, 1089           :rd 5, 1099, 1090            :rd 5, 1098, 1089
024  :rdi 9, 5                   :rdi 9, 5                    :rdi 9, 5
025  :rd 4, 1098, 1094           :rd 4, 1099, 1095            :rd 4, 1098, 1094
026  :rdi 4, 4                   :rdi 4, 4                    :rdi 4, 4
027  :c0 4, 1098                 :c0 4, 1099                  :c0 4, 1098
028  :wr 42 0                    :wr 42 0                     :wr 42 0
029  :wrc 42 42 0                :wrc 42 42 0                 :wrc 42 42 0
030  :wr 6 0                     :wr 6 0                      :wr 6 0
031  :wrc 6 6 0                  :wrc 6 6 0                   :wrc 6 6 0
032  :wr 37 0                    :wr 37 0                     :wr 37 0
033  :wrc 37 37 0                :wrc 37 37 0                 :wrc 37 37 0
034  :ack 42                     :ack 42                      :ack 42
035  :ack 43                     :ack 43                      :ack 43
036  :rn 43                      :rn 43                       :rn 43
037  :rd 5, 43, 0                :rd 5, 43, 0                 :rd 5, 43, 0
038  :rdi 43, 5                  :rdi 43, 5                   :rdi 43, 5
039  :rd 1, 43, 5                :rd 1, 43, 5                 :rd 1, 43, 5
040  :rdi 38, 1                  :rdi 38, 1                   :rdi 38, 1
041  :rd 5, 43, 6                :rd 5, 43, 6                 :rd 5, 43, 6
042  :rdi 37, 5                  :rdi 37, 5                   :rdi 37, 5
043  :rd 32, 43, 11              :rd 32, 43, 11               :rd 32, 43, 11
044  :rdi 32, 32                 :rdi 32, 32                  :rdi 32, 32
045  :c0 32, 43                  :c0 32, 43                   :c0 32, 43
046  :wr 149 0                   :wr 149 0                    :wr 149 0
047  :wrc 149 149 0              :wrc 149 149 0               :wrc 149 149 0
048  :ack 149                    :ack 149                     :ack 149
049  :wr 46 0                    :wr 46 0                     :wr 46 0
050  :wrc 46 46 0                :wrc 46 46 0                 :wrc 46 46 0
051  :ack 46                     :ack 46                      :ack 46
052  :rn 1390                    :rn 1390                     :rn 1390
053  :rch 1390, 1390             :rch 1390, 1390              :rd 5, 1390, 0
054  :rch 2780, 1460             :rch 2780, 1460              :rdi 1390, 5
055  :rch 4240, 1320             :rch 4240, 1320              :rd 1385, 1390, 5
056  :rch 5560, 1244             :rch 5560, 1240              :rdi 1385, 1385
057  :rd 5, 6804, 0              :rd 5, 6800, 0               :c0 1385, 1390
058  :rdi 1390, 5                :rdi 1390, 5                 :rch 1390, 1390
059  :rd 1385, 6804, 5           :rd 1385, 6800, 5            :rch 2780, 1460
060  :rdi 1385, 1385             :rdi 1385, 1385
061  :c 1385, 1390, 6804         :c 1385, 1390, 6800
062  OK                          OK
[... 7 lines cut ...]
070  :rdi 1460, 5                :rdi 1460, 5
071  :rd 1385, 4024, 5           :rd 1385, 4020, 5
072  :rdi 1455, 1385             :rdi 1455, 1385
073  :rd 5, 4024, 1390           :rd 5, 4020, 1390
074  :rdi 70, 5                  :rdi 70, 5
[... cut ...]
083  :rdi 1239, 1213             :rdi 1235, 1209
084  :rd 5, 1244, 1218           :rd 5, 1240, 1214
085  :rdi 26, 5                  :rdi 26, 5
086  :rd 21, 1244, 1223          :rd 21, 1240, 1219
087  :rdi 21, 21                 :rdi 21, 21
088  :c0 21, 1244                :c0 21, 1240
089  Payload OK: 5645 bytes      Payload OK: 5645 bytes
090  :ur 1                       :ur 1
091  :dsrcv 0                    :dsrcv 0
092  :close                      :close
093  :del                        :del
```

earlephilhower, 2023-03-14:
I just let your example run for >400 times and was unable to see any hangs in my testing.  Since it's a 3rd party site I don't want them to get upset at the amount of traffic I was pulling so stopped there.

Does it reliably hang?  The code itself looks OK (good job putting the `HTTPClient` in the globals, that makes the SSL memory allocation persistent and reduces memory fragmentation!).

Do you have a Picoprobe?  On your hung system, can you get a `where` stack trace to help pinpoint what's going on?  OTW, given I can't repro there's not much I can suggest at this point.

jazztunes, 2023-03-14:
After a few shorter runs, my last test just did 33,758 iterations and ran for over 24 hrs without a hang 🤷. I can't reproduce it reliably but it does seem more stable in 3.1.0. I have a picoprobe on the way, so let's close this until I can get more info.

Thanks for the help on this and all your work on this project!

closed 2023-03-14 by earlephilhower