earlephilhower/arduino-pico #1302: Can't update board via Arduino 2.0.x (4)

jaszczurtd, 2023-03-13:
Please take a look on the attached screenshot(s).
The description of the rp2040 board package is so long, that somehow overlaps the buttons/controls for the install/update in Arduino 2.0.x IDE release. However, it seems that with tle older version of Arduino the issue does not exists.
Hint: there is no "More info" link, which indicates the possibility to jump onto github repository.
Tested on Arduino 2.0.4 on windows/osx.

[image]
[image]

jazztunes, 2023-03-13:
You can resize the boards flyout menu.

[image]

jaszczurtd, 2023-03-13:
lol, nice. :D

closed 2023-03-13 by jaszczurtd