earlephilhower/arduino-pico #1308: Package creation is too painful

daveythacher, 2023-03-15:
I am running into a lot of issues with this. Documentation is terrible.
- Fails to mention how to "publish". (I figured this out.)
- Gets 404 and fails in the action view. (Note you need to enable this.)
- For some reason it finally generates the JSON file, but bears out the URL for this repo.
- No mention of permissions or other configuration settings.

In summary I cannot test any changes without maybe a network proxy or relay.

earlephilhower, 2023-03-15:
There's no reason to actually do a package and/or publish to test local changes.  Please just use the `git` method described in the README.md which is how I do all my own development.

Make sure you aren't using a Boards Manager installation (forcibly remove it from the IDE Boards Manager to be 100% sure) and make sure the git clone is in the proper place and everything "just works."

daveythacher, 2023-03-15:
Now how do I integrate with the Arduino IDE?

earlephilhower, 2023-03-15:
If you've placed the `git` clone in the right spot, it just appears.  `~/Arduino/hardware/pico/rp2040` on Linux.  Under `Arduino/hardware/pico/rp2040` in the Windows `Documents` folder IIRC.  May need to restart the IDE if you change `boards.txt` which is where the menus come from.

daveythacher, 2023-03-15:
How does Arduino know to look in that folder? Does it just do this on its own?

Any chance we can get this in the readme?

earlephilhower, 2023-03-15:
Yes, it does it on its own.  There's nothing special about this vs. any other board package as far as the IDE is concerned.  If it's not picking things it, then the dir is in the wrong spot or the IDE is busted.

Don't use the Windows Store install (or flatpack/snap/etc. under Linux) also as discussed in the README.  Those prepackaged ones have funky access permissions and often can't communicate w/the board properly, so it would not surprise me if they also break standard IDE functionality.

daveythacher, 2023-03-15:
Any chance the comments on git and packages can be made in the README.md? That would close this out in my opinion.

earlephilhower, 2023-03-16:
I'm sorry, but I don't think there's much I can say.  It's pretty standard, but I can see how it's confusing for a first time Arduino contributor.  The `git` section in the README really does summarize things https://github.com/earlephilhower/arduino-pico#installing-via-git .

Good luck on the RAM-only mode.  You should also think about disabling the core freeze for flash operations in that config, since it's not needed (by definition both cores are running from RAM).

closed 2023-03-16 by earlephilhower