earlephilhower/arduino-pico #1313: Consistent gcc and gdb releases

rei-vilo, 2023-03-17:
When trying to debug against the Raspberry Pico on Visual Studio Code 1.76.2 with Cortex-Debug 1.10.0 using a PicoProbe, an error complains about `gdb`. 

> GDB could not start as expected. Bad installation or version mismatch. See if you can start gdb from a shell prompt and check its version (Must be >= 9)

Actually, the supplied `pqt-gcc` tool-chain includes `gcc` 10.3.0 but `gdb` 8.2.50.

```
$ Arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/arm-none-eabi-c++ -v
gcc version 10.3.0 (GCC) 
```

```
$ Arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/arm-none-eabi-gdb -v
GNU gdb (GDB) 8.2.50.20190202-git
```

Could `gdb` be changed to the same release 10.3.0 as `gcc`? 

I solved the issue by installing `arm-none-eabi-gcc` 10-2020-q4-major.

Thank you!

earlephilhower, 2023-03-17:
I don't think the GDB and GCC releases always coincide, but this seems a reasonable request.  I've set up the toolchain build for GCC-10.4 to pull in a `binutils` with gdb 10.x in it.  So, next minor toolchain release this should be in there.

One note of caution using 3rd party compiler packages.  Using GDB is perfectly fine, but if you use the compiler, too, there may be big issues with respect to the `libc`/`Newlib` libraries.  So, save yourself the gray hairs and don't do that,. :)

rei-vilo, 2023-03-17:
Thank you for the advice!

rei-vilo, 2023-05-13:
An alternative for a more recent `gdb`, as required by Visual Studio Code, is to use `gdb-multiarch`.

On Linux, simply run

``` 
sudo apt upgrade gdb-multiarch
```

earlephilhower, 2023-08-20:
#1650 related.  Should pull both changes in at the same time

commit ade74986 (2023-08-25): GCC 12.3, GDB 12, OpenOCD 0.12, Picotool 1.1.2 (#1670)

Major toolchain update including:
* GCC 12.3
* GDB 12
* OpenOCD 0.12
* Picotool 1.1.2

* Fix MDNS infinite recursion

* Remove legacy Picoprobe

Fixes #1313
Fixes #1650
closed 2023-08-25 by earlephilhower