earlephilhower/arduino-pico #1316: Unable to pull your repo into the TFT_eSPI library

mytechnotalent, 2023-03-18:
When trying to use your library within the https://github.com/Bodmer/TFT_eSPI library I am getting.

```
Error: VCS: Could not process command ['git', 'clone', '--recursive', 'https://github.com/earlephilhower/arduino-pico.git', 'C:\\Users\\mytec\\.platformio\\.cache\\tmp\\pkg-installing-ejwdlins']

 *  The terminal process "C:\Users\mytec\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'pico'" terminated with exit code: 1.
 *  Terminal will be reused by tasks, press any key to close it.
 ```

Here is my platform.ini file.  This is a Raspberry Pi Pico using a GC9a01 display.
```
;PlatformIO User notes:

;It is possible to load settings from the calling program rather than modifying
;the library for each project by modifying the "platformio.ini" file.

;The User_Setup_Select.h file will not load the user setting header files if
;USER_SETUP_LOADED is defined.

;Instead of using #define, use the -D prefix, for example:

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter, extra scripting
;   Upload options: custom port, speed and extra flags
;   Library options: dependencies, extra library storages
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
lib_deps = bodmer/TFT_eSPI@^2.5.21
; change microcontroller
board_build.mcu = rp2040

; change MCU frequency
board_build.f_cpu = 133000000L

build_flags =
  -Os
  -DUSER_SETUP_LOADED=1
  ; Define the TFT driver, pins etc here:
  -DTFT_PARALLEL_8_BIT=1
  -DRM68120_DRIVER=1
  -DRP2040_PIO_CLK_DIV=1
  -DTFT_DC=6
  -DTFT_WR=2
  -DTFT_RST=7

  -DTFT_D0=12
  -DTFT_D1=13
  -DTFT_D2=14
  -DTFT_D3=15
  -DTFT_D4=16
  -DTFT_D5=17
  -DTFT_D6=18
  -DTFT_D7=19

  -DTFT_BL=8
  -DTFT_BACKLIGHT_ON=HIGH

  -DLOAD_GLCD=1
  -DLOAD_FONT2=1
  -DLOAD_FONT4=1
  -DLOAD_FONT6=1
  -DLOAD_FONT7=1
  -DLOAD_FONT8=1
  -DLOAD_GFXFF=1
  -DSMOOTH_FONT=1

  ; Define the specific pin mappings here:
  -DSCR_PIN_DIN=3
  -DSCR_PIN_CLK=2
  -DSCR_PIN_CS=5
  -DSCR_PIN_DC=6
  -DSCR_PIN_RST=7
  -DSCR_PIN_BL=8
```

Any help would be greatly appreciated thanks in advance!

earlephilhower, 2023-03-18:
Err, you'd pull the library into the core, not the other way around normally.

Anyway, do you have `git` installed?  When you run 
````
git clone --recursive https://github.com/earlephilhower/arduino-pico.git
````
from a command line, does it succeed?  What's the output?

mytechnotalent, 2023-03-18:
@earlephilhower it does succeed when I `git clone --recursive https://github.com/earlephilhower/arduino-pico.git` as I cloned that into the root of the repo.  Would that need to be cloned in the `lib` dir instead as this is a platformio project?

earlephilhower, 2023-03-18:
Unfortunately I don't use Platform.IO.  @maxgerhardt might be able to point to some things to check,  since the basics here seem okay.

Did you follow the steps he outlined in https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst for starters?

maxgerhardt, 2023-03-18:
Close VSCode, delete `C:\Users\mytec\.platformio\.cache` and `C:\Users\mytec\.platformio\packages\framework-arduinopico` completely, reopen VSCode. Does it reload the project correctly?

mytechnotalent, 2023-03-18:
@maxgerhardt I did exactly that and still getting 
```
Error: VCS: Could not process command ['git', 'clone', '--recursive', 'https://github.com/earlephilhower/arduino-pico.git', 'C:\\Users\\mytec\\.platformio\\.cache\\tmp\\pkg-installing-xewrcnt6']```

maxgerhardt, 2023-03-18:
And you're sure you did `git config --system core.longpaths true` and enabled [Win32 Long Paths](https://www.microfocus.com/documentation/filr/filr-4/filr-desktop/t47bx2ogpfz7.html)?

mytechnotalent, 2023-03-18:
> git config --system core.longpaths true

I dont have the `gpedit.msc` within Windows 11 and which repo would I `git config --system core.longpaths true` in?

maxgerhardt, 2023-03-18:
Then go via [the registry instead](https://thegeekpage.com/make-windows-11-accept-file-paths-over-260-characters/). An the got config is system-wide, it does not need to be executed within a repo.

mytechnotalent, 2023-03-18:
@maxgerhardt I manually changed the value to 1 within the Windows Registry, closed VS Code and reloaded and same issue.

maxgerhardt, 2023-03-18:
>fatal: Failed to recurse into submodule path 'pico-sdk/lib/tinyusb'

Yeah exactly, that is a classical sign that it couldn't clone it due to path issues (https://github.com/hathach/tinyusb/issues/1939#issuecomment-1464737325).

Have you rebooted the computer already?

Your exact `platformio.ini` works on my Windows 10 machine with activated Git and Win32 long paths without issues.

[image]

mytechnotalent, 2023-03-18:
@maxgerhardt 
```Initialized empty Git repository in C:/Users/mytec/Desktop/display/.git/
PS C:\Users\mytec\Desktop\display> git config --system core.longpaths true
error: could not lock config file C:/Program Files/Git/etc/gitconfig: Permission denied```

I will try on my MAC M1 computer in the morning.  Thank you so much for the help I will let you know and I think it is safe to close out I will keep trying.

mytechnotalent, 2023-03-18:
Worked thank you so much Max!!!

closed 2023-03-18 by mytechnotalent