earlephilhower/arduino-pico #1322: Cannot install on PlatformIO

oofus, 2023-03-20:
Hi

I'm having problems installing on Platform IO. The Git update proceeds, but after quite a while of good progress I get ...

From https://github.com/STMicroelectronics/stm32u5xx_hal_driver
 * branch            2e1d4cdb386e33391cb261dfff4fefa92e4aa35a -> FETCH_HEAD

Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/st/stm32u5xx_hal_driver': checked out '2e1d4cdb386e33391cb261dfff4fefa92e4aa35a'
Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/st/stm32wbxx_hal_driver': checked out '2c5f06638be516c1b772f768456ba637f077bac8'
Submodule path 'pico-sdk/lib/tinyusb/hw/mcu/ti': checked out '143ed6cc20a7615d042b03b21e070197d473e6e5'
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0001-RTT-VMM-implement-dual-system-running-on-realview-pb.patch: Filename too long
error: unable to create file Application/8_uarts_server/RT-Thread/uart_server_rtt/rt-thread/components/vmm/linux_patch-v3.8/0002-arm-gic-correct-the-cpu-map-on-gic_raise_softirq-for.patch: Filename too long

fatal: Unable to checkout '17761f5cf9dbbf2dcf665b7c04934188add20082' in submodule path 'pico-sdk/lib/tinyusb/hw/mcu/wch/ch32v307'

Submodule path 'pico-sdk/lib/tinyusb/lib/CMSIS_5': checked out '20285262657d1b482d132d20d755c8c330d55c1f'
Submodule path 'pico-sdk/lib/tinyusb/lib/FreeRTOS-Kernel': checked out '2a604f4a2818b8354b5e1a39e388eb5e16cfbc1f'
Submodule path 'pico-sdk/lib/tinyusb/lib/lwip': checked out '159e31b689577dbf69cf0683bbaffbd71fa5ee10'
Submodule path 'pico-sdk/lib/tinyusb/lib/sct_neopixel': checked out 'e73e04ca63495672d955f9268e003cffe168fcd8'
Submodule path 'pico-sdk/lib/tinyusb/tools/uf2': checked out '19615407727073e36d81bf239c52108ba92e7660'
Submodule 'hidapi' (https://github.com/signal11/hidapi) registered for path 'pico-sdk/lib/tinyusb/tools/uf2/hidapi'

Cloning into 'C:/Users/chris/.platformio/.cache/tmp/pkg-installing-xy3w7c6z/pico-sdk/lib/tinyusb/tools/uf2/hidapi'...

Submodule path 'pico-sdk/lib/tinyusb/tools/uf2/hidapi': checked out 'a6a622ffb680c55da0de787ff93b80280498330f'
fatal: Failed to recurse into submodule path 'pico-sdk/lib/tinyusb'

fatal: Failed to recurse into submodule path 'pico-sdk'

Submodule path 'tools/libbearssl/bearssl': checked out 'f294aa022f82833941ffdf3b0f306c2072e3a26c'
Submodule path 'tools/pyserial': checked out '0e7634747568547b8a7f9fd0c48ed74f16af4b23'
Error: VCS: Could not process command ['git', 'clone', '--recursive', 'https://github.com/earlephilhower/arduino-pico.git', 'C:\\Users\\chris\\.platformio\\.cache\\tmp\\pkg-installing-xy3w7c6z']

Is it me, or is there an issue with the latest version ?

Thanks

earlephilhower, 2023-03-20:
See https://github.com/earlephilhower/arduino-pico/blob/129fb14c5380ec0a312132decefa109680085305/docs/platformio.rst#important-steps-for-windows-users-before-installing .  There's a TinyUSB path-depth issue you can fix using the instructions there.

closed 2023-03-20 by earlephilhower

oofus, 2023-03-20:
Ah, thanks. I was following these instructions, 

https://arduino-pico.readthedocs.io/en/latest/platformio.html

where that isn't mentioned !

earlephilhower, 2023-03-20:
Yeah, it just came up w/the Pico SDK 1.5 and the latest TinyUSB which added even more submodules.  The link you used will be updated once the doc update I wrote is confirmed.

oofus, 2023-03-20:
Success! Thanks for the rapid response.