earlephilhower/arduino-pico #1326: WaveShare Pico-ResTouch-LCD-2.8 display conflicting SPI

DennisSchm, 2023-03-23:
Hello and thank you for your very convenient port!

I am using the **raspberry pi pico w** board with the **WaveShare Pico-ResTouch-LCD-2.8** display and write my code in VS Code with PlatformIO. The display provides an additional SD card module I'd like to use. For now the default library will suffice. 

The issue I want to report is that I can't use this SD card since it uses pin 5 as CLK which belongs to SPI0 and pin 11 and 12 as MOSI and MISO which belong to SPI1. I need to pass SPI1 to SD.begin() but it can't mount the SD which is most likely because I can't assign an SPI0 CLK pin to SPI1 I think. 

Is there any elegant way to work around this issue? 

Thank you

earlephilhower, 2023-03-23:
Looking at the docs their choice of wiring seems kind of wacky.

If I read it correctly they use GP5 as SPI-CLK.  The Pico HW can only drive [SPI0-*chip-select* on GP5](https://www.raspberrypi.com/documentation/microcontrollers/images/pico-pinout.svg), not a clock.  So right off the bat the board can't be used with the integrated peripherals.

Your best/only option is to find a software SPI implementation.  If there is a generic one out there that uses standard Arduino `digitalWrite`, `digitalRead`, and `pinMode` then it should work here.  But there is no built-in support for a SW-based SPI implementation (with 2 HW peripherals and lots of IO options for them it seems not worth the time).

closed 2023-03-23 by earlephilhower

DennisSchm, 2023-03-28:
Thank you for your answer. I also had trouble reading their wiring. Using this diagram ( https://www.waveshare.com/w/upload/d/dc/Pico-ResTouch-LCD-2.8_Sch.pdf ) I was able to deduce, that all peripherals on this board use pin 10 as CLK, which is SPI1. So If you don't use SDIO feature and just use regular SD speeds you should be fine. I used the following User_Setup.h for TFT_eSPI: 

`#define USER_SETUP_INFO "User_Setup"

#define ST7789_DRIVER 

#define TFT_BACKLIGHT_ON HIGH

#define TOUCH_CS 16
// #define TOUCH_IRQ 17 <- not needed, the IRQ is used by default

#define TFT_WIDTH 240  
#define TFT_HEIGHT 320 

#define TFT_DC 8 // data command 
#define TFT_CS 9 // chip select 
#define TFT_SCLK 10
#define TFT_MOSI 11
#define TOUCH_MISO 12
#define TFT_RST 15 // reset pin 
#define TFT_BL 13  // LED back-light

#define LOAD_GLCD  // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2 // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4 // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6 // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7 // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8 // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts
 
// this will save ~20kbytes of FLASH
#define SMOOTH_FONT

#define TFT_SPI_PORT 1 // Set to 0 if SPI0 pins are used

#define SPI_FREQUENCY 27000000
#define SPI_READ_FREQUENCY 20000000
#define SPI_TOUCH_FREQUENCY 2500000`

main.h 

`#include <SPI.h>
#include <SD.h>

// Set REPEAT_CAL to true instead of false to run calibration
// again, otherwise it will only be done once.
// Repeat calibration if you change the screen rotation.
#define REPEAT_CAL false

 TFT_eSPI tft = TFT_eSPI();

    void touch_calibrate()
    {
        uint16_t calData[5];
        uint8_t calDataOK = 0;

        // check file system exists
        const int pinSdCs = 22;
        const int pinSdClk = 10;
        const int pinSdMosi = 11;
        const int pinSdMiso = 12;

        // check if calibration file exists and size is correct
        if (SD.exists(CALIBRATION_FILE))
        {
            if (REPEAT_CAL)
            {
                // Delete if we want to re-calibrate
                SD.remove(CALIBRATION_FILE);
            }
            else
            {
                File f = SD.open(CALIBRATION_FILE, "r+");
                if (f)
                {
                    if (f.readBytes((char *)calData, 14) == 14)
                        calDataOK = 1;
                    f.close();
                }
            }
        }

        if (calDataOK && !REPEAT_CAL)
        {
            // calibration data valid
            tft.setTouch(calData);
        }
        else
        {
            // data not valid so recalibrate
            tft.fillScreen(TFT_BLACK);
            tft.setCursor(20, 0);
            tft.setTextFont(2);
            tft.setTextSize(1);
            tft.setTextColor(TFT_WHITE, TFT_BLACK);

            tft.println("Touch corners as indicated");

            tft.setTextFont(1);
            tft.println();

            if (REPEAT_CAL)
            {
                tft.setTextColor(TFT_RED, TFT_BLACK);
                tft.println("Set REPEAT_CAL to false to stop this running again!");
            }

            tft.calibrateTouch(calData, TFT_MAGENTA, TFT_BLACK, 15);

            tft.setTextColor(TFT_GREEN, TFT_BLACK);
            tft.println("Calibration complete!");

            // store data
            File f = SD.open(CALIBRATION_FILE, "w+");
            if (f)
            {
                f.write((const unsigned char *)calData, 14);
                f.close();
            }
        }
    }

setup(){
 SPI1.setSCK(pinSdClk);
        SPI1.setTX(pinSdMosi);
        SPI1.setRX(pinSdMiso);
        SPI1.begin();
        
        while (!SD.begin(pinSdCs,SPI1))
        {
            Serial.println("Card Mount Failed");
            delay(500);
        }
    tft.init();
        tft.setRotation(1);
        tft.fillScreen(colorBlack);
        tft.setTextDatum(TC_DATUM);
        tft.setTextColor(colorWhite, colorBlack);
        touch_calibrate();
}

    uint16_t x, y;

loop(){
        bool touched = DisplayGt::tft.getTouch(&x, &y);
        if (touched){
            // touch code
        }
}
`