earlephilhower/arduino-pico #1329: PlatformIO document for using this core with Pico does not work when you use Adafruit TinyUSB as shown in document

alijani1, 2023-03-26:
[Document on this repository](https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst) provides great instructions but unfortunately when you try to use the build flags to override the USB stack from Pico SDK to Adafruit TinyUSB as shown in the document, you get compile errors with conflicting definitions.

Document says to use 'build_flags = -DUSE_TINYUSB`

But one gets 'error: conflicting declaration Adafruit_USBD_CDC UART_USB'

This might be some issue with platformio as the same sketeches work fine in Arduino IDE.

This is the [example sketch](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/master/examples/HID/hid_gamepad/hid_gamepad.ino) I tried to compile without any luck.

And here is the sample platformio.ini I used:
`[env:pico]
platform = raspberrypi

board = pico

framework = arduino

lib_archive = no

build_flags = -DUSE_TINYUSB
'

alijani1, 2023-03-26:
I was able to get past this by using the earlephilhower core with maxgerhardt updated platform.  I will notate this on the doc in the USB stack section.

platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board_build.core = earlephilhower

Also I am not sure why https://github.com/earlephilhower/arduino-pico would not work either with even more errors.

maxgerhardt, 2023-03-27:
Right, with
```ini
platform = raspberrypi
board = pico
framework = arduino
lib_archive = no
build_flags = -DUSE_TINYUSB
```
you're using ArduinoCore-mbed, not Arduino-Pico, so this repo was never involved in the first build.

>Also I am not sure why https://github.com/earlephilhower/arduino-pico would not work either with even more errors.

If you tried to use that URL as the `platform = ...` value that's wrong because this repo is not a PlatformIO platform. It's (also) a PlatformIO package and can be used in a `platform_packages` expression [as documented](https://arduino-pico.readthedocs.io/en/latest/platformio.html#selecting-a-different-core-version).

closed 2023-03-27 by earlephilhower