earlephilhower/arduino-pico #1332: Request/Question: Bare metal core without Arduino

jonathanlundstrom, 2023-03-28:
Hi there,

How hard would it be to create a simpler PlatformIO platform that only uses the Pico SDK but bundles the compilation, debug tools and other nice things from this project? The reason for my question being that we want to avoid the Arduino framework all together and only work with C and the Pico SDK.

Regards,
Jonathan

closed 2023-03-28 by jonathanlundstrom

CHUYRO, 2023-04-05:
Hi @jonathanlundstrom, sorry to bother but I was also looking forward to a reply on this topic even though it seems you closed it, did you get answered or just left it?

jonathanlundstrom, 2023-04-05:
I decided to close the issue as I realised this was not the right place to ask for it, seeing as how this repository is an Arduino one. I think ultimately one would have to write a library such as the Wiz-IO one, but based on the pure SDK and with full implementation of debug tools – which is something I would love to do, but currently do not have time for. Although if others are interested I would probably free up some time for it.

CHUYRO, 2023-04-05:
Understood, thanks for quick reply!