earlephilhower/arduino-pico #1336: active Wi-Fi feature (Pico W) has negative effect on USB CDC Host operation

lyusupov, 2023-03-29:
**Release**: 2.7.3
**Hardware**: Pico W
**Sketch**: [serial_host_bridge.ino](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/e1e922d2caf349f2d9f90446af13349751254740/examples/DualRole/serial_host_bridge/serial_host_bridge.ino)

When Wi-Fi SoftAP is **OFF** - the USB Host CDC demo sketch works nicely with connected USB CDC ACM client device.

However, when the sketch is changed a little fit to enable the AP -

```
--- serial_host_bridge.ino.orig 2023-03-29 20:38:37.686612638 +0300
+++ serial_host_bridge.ino      2023-03-29 20:39:13.730015196 +0300
@@ -48,12 +48,17 @@
 // Setup and Loop on Core0
 //--------------------------------------------------------------------+

+#include <WiFi.h>
+
 void setup() {
   Serial1.begin(115200);

   Serial.begin(115200);
   while ( !Serial ) delay(10);   // wait for native usb

+  WiFi.mode(WIFI_AP);
+  WiFi.softAP("picow", "12345678");
+
   Serial.println("TinyUSB Host Serial Echo Example");
 }
```
- this causes an internal infinite loop _(tested by watchdog timer activation)_.

### Expected behavior

The sketch should work the same way regardless of if the Wi-Fi is turned ON or OFF.

### To reproduce

Run the sketch on Pico W with and without the SoftAP feature,
Make sure that data is transferred out of the  connected CDC source. When there is no data flow - the issue may not reproduce this case .

earlephilhower, 2023-04-04:
No response in a week, closing.

closed 2023-04-04 by earlephilhower

lyusupov, 2023-04-14:
Tips from this report: https://github.com/sekigon-gonnoc/Pico-PIO-USB/issues/76
such as:

1.  claim of resources
```
  pio_sm_claim(pp->pio_usb_tx, pp->sm_tx);
  pio_sm_claim(pp->pio_usb_rx, pp->sm_rx);
  pio_sm_claim(pp->pio_usb_rx, pp->sm_eop);

  dma_claim_mask(1<<c->tx_ch);
```
2. CYW43 SPI PIO re-assignment
```
CYW43_SPI_PIO_PREFERRED_PIO=0
```
give positive effect necessary.