earlephilhower/arduino-pico #1344: Can't upload

sstaub, 2023-03-31:
Using Arduino IDE 2.0.4 on macOS Ventura. I can't upload to board.
Tools:
[image]
Error Message:
```
Using board 'rpipico' from platform in folder: /Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0
Using core 'rp2040' from platform in folder: /Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0
Detecting libraries used...
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/arm-none-eabi-g++ -I /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/core -c -Werror=return-type -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipico\"" -DTARGET_RP2040 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -DENABLE_PICOTOOL_USB -iprefix/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/ @/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/lib/platform_inc.txt -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/cores/rp2040 -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/variants/rpipico /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/sketch/sketch_mar31a.ino.cpp -o /dev/null
Generating function prototypes...
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/arm-none-eabi-g++ -I /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/core -c -Werror=return-type -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipico\"" -DTARGET_RP2040 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -DENABLE_PICOTOOL_USB -iprefix/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/ @/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/lib/platform_inc.txt -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/cores/rp2040 -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/variants/rpipico /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/sketch/sketch_mar31a.ino.cpp -o /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/preproc/ctags_target_for_gcc_minus_e.cpp
/Users/sstaub/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed/python3 -I /Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/tools/signing.py --mode header --publickey /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/.arduinoIDE-unsaved2023231-55285-1e7ehcd.ykao/sketch_mar31a/public.key --out /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/core/Updater_Signing.h
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/arm-none-eabi-g++ -I /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/core -c -Wall -Wextra -Werror=return-type -Wno-ignored-qualifiers -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipico\"" -DTARGET_RP2040 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -DENABLE_PICOTOOL_USB -MMD -iprefix/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/ @/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/lib/platform_inc.txt -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/include -fno-rtti -std=gnu++17 -g -pipe -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/cores/rp2040 -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/variants/rpipico /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/sketch/sketch_mar31a.ino.cpp -o /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/sketch/sketch_mar31a.ino.cpp.o
Compiling libraries...
Compiling core...
Using precompiled core: /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/cores/d0e9c18e5b3c10275db6a8a607a9f09c/core.a
Linking everything together...
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed/python3 -I /Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/tools/simplesub.py --input /Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/lib/memmap_default.ld --out /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/memmap_default.ld --sub __FLASH_LENGTH__ 2093056 --sub __EEPROM_START__ 270528512 --sub __FS_START__ 270528512 --sub __FS_END__ 270528512 --sub __RAM_LENGTH__ 256k
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/arm-none-eabi-gcc -Werror=return-type -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipico\"" -DTARGET_RP2040 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -DENABLE_PICOTOOL_USB -Os -u _printf_float -u _scanf_float -c /Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/boot2/boot2_w25q080_2_padded_checksum.S -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/pico-sdk/src/rp2040/hardware_regs/include/ -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/pico-sdk/src/common/pico_binary_info/include -o /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/boot2.o
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/arm-none-eabi-g++ -L/private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8 -Wall -Wextra -Werror=return-type -Wno-ignored-qualifiers -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipico\"" -DTARGET_ [... cut ...]
  Referenced from: <04C710E9-2AD6-3052-85E4-E74077148E1C> /Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-picotool/1.5.0-b-03f2812/picotool
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
Failed uploading: uploading error: signal: abort trap
```

sstaub, 2023-03-31:
I forgot to say. Using PlatformIO works without problems.

maxgerhardt, 2023-03-31:
>dyld[55433]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <04C710E9-2AD6-3052-85E4-E74077148E1C> /Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-picotool/1.5.0-b-03f2812/picotool
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
Failed uploading: uploading error: signal: abort trap

Looks to me like it can't find the libusb library. Try installing [brew](https://brew.sh/) and then `brew install libusb`.

It might have worked in PlatformIO because the picotool package has a `lib/` folder with `libusb-1.0.0.dylib`.

sstaub, 2023-03-31:
Installing with brew works but lead me to the next error

```
Using board 'rpipico' from platform in folder: /Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0
Using core 'rp2040' from platform in folder: /Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0
Detecting libraries used...
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/arm-none-eabi-g++ -I /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/core -c -Werror=return-type -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipico\"" -DTARGET_RP2040 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -DENABLE_PICOTOOL_USB -iprefix/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/ @/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/lib/platform_inc.txt -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/cores/rp2040 -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/variants/rpipico /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/sketch/sketch_mar31a.ino.cpp -o /dev/null
Generating function prototypes...
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/arm-none-eabi-g++ -I /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/core -c -Werror=return-type -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipico\"" -DTARGET_RP2040 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -DENABLE_PICOTOOL_USB -iprefix/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/ @/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/lib/platform_inc.txt -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/cores/rp2040 -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/variants/rpipico /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/sketch/sketch_mar31a.ino.cpp -o /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/preproc/ctags_target_for_gcc_minus_e.cpp
/Users/sstaub/Library/Arduino15/packages/builtin/tools/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed/python3 -I /Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/tools/signing.py --mode header --publickey /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/.arduinoIDE-unsaved2023231-55285-1e7ehcd.ykao/sketch_mar31a/public.key --out /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/core/Updater_Signing.h
/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/arm-none-eabi-g++ -I /private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/core -c -Wall -Wextra -Werror=return-type -Wno-ignored-qualifiers -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipico\"" -DTARGET_RP2040 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -DENABLE_PICOTOOL_USB -MMD -iprefix/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/ @/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/lib/platform_inc.txt -I/Users/sstaub/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.0/include -fno-rtti -std=gnu++17 -g -pipe -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 - [... cut ...]
Sketch uses 52324 bytes (2%) of program storage space. Maximum is 2093056 bytes.
Global variables use 8916 bytes (3%) of dynamic memory, leaving 253228 bytes for local variables. Maximum is 262144 bytes.
"/Users/sstaub/Library/Arduino15/packages/rp2040/tools/pqt-picotool/1.5.0-b-03f2812/picotool" load "/private/var/folders/vd/yv1mb7lx6gn7ymffgtjk6hx80000gn/T/arduino/sketches/45B3134C005BE0473484BADE2698D7B8/sketch_mar31a.ino.uf2" -f
ERROR: Unable to locate reset interface on the device
Failed uploading: uploading error: exit status 250
```

sstaub, 2023-03-31:
OK, after doing in boot loader mode it works.

maxgerhardt, 2023-03-31:
The docs could be improved to say that when Tools->Upload Method is selected as "Picotool", you need to make sure that on Mac `libusb` is installed. Currently [it doesn't mention that](https://arduino-pico.readthedocs.io/en/latest/install.html#uploading-sketches-with-picotool).

But also, if now a Arduino-Pico sketch is loaded, I'd recommend to change the upload method back to "Default (UF2)" so that a script will do a reset-into-bootloader-mode via a 1200BPS serial port open after which it can flash normally. So you won't need to reboot the board into BOOTSEL mode for every upload.

earlephilhower, 2023-03-31:
On the Mac, don't the DLLs in the executable directory automatically get picked up?  The Picoprobe OSX tarball has the libusb dynlib file included (as does the OpenOCD), in the same install directory.

Anyway, it sounds like we just need a note that on the Mac you want to `brew install libusb` (does it need `sudo`?) before using PicoProbe (or OpenOCD I guess, same issue), yes?

sstaub, 2023-03-31:
`brew install libusb `worked without `sudo`. I can remember that an earlier version (before 3.x) works  without a problem.
Normally I use PIO, but sometimes I use the Arduino IDE to check for compatibility issues.

PR #1355 (2023-04-04): Add MacOS Picotool upload help
Fixes #1344
commit ff0d794c (2023-04-04): Add MacOS Picotool upload help (#1355)

Fixes #1344
closed 2023-04-04 by earlephilhower