earlephilhower/arduino-pico #1356: BTstack examples won't compile

noahbaron, 2023-04-06:
Thanks for making an Arduino wrapper for the Pico W bluetooth functionality. I am trying to compile iBeaconScanner.ino in the BTstack folder with board v.3.1.0, which I understand is beta/alpha however I did note that it should compile. I haven't modified the example and in looking at BTstack.h I don't see any reason why these errors would occur.

Here are the errors:

/private/var/folders/2w/n4f1n28573345fr4zwssdkmc0000gn/T/.arduinoIDE-unsaved202335-73470-gxhntz.rwt4/iBeaconScanner/iBeaconScanner.ino:37:6: error: variable or field 'advertisementCallback' declared void
   37 | void advertisementCallback(BLEAdvertisement *adv) {
      |      ^~~~~~~~~~~~~~~~~~~~~
/private/var/folders/2w/n4f1n28573345fr4zwssdkmc0000gn/T/.arduinoIDE-unsaved202335-73470-gxhntz.rwt4/iBeaconScanner/iBeaconScanner.ino:37:28: error: 'BLEAdvertisement' was not declared in this scope
   37 | void advertisementCallback(BLEAdvertisement *adv) {
      |                            ^~~~~~~~~~~~~~~~
/private/var/folders/2w/n4f1n28573345fr4zwssdkmc0000gn/T/.arduinoIDE-unsaved202335-73470-gxhntz.rwt4/iBeaconScanner/iBeaconScanner.ino:37:46: error: 'adv' was not declared in this scope
   37 | void advertisementCallback(BLEAdvertisement *adv) {
      |                                              ^~~
/private/var/folders/2w/n4f1n28573345fr4zwssdkmc0000gn/T/.arduinoIDE-unsaved202335-73470-gxhntz.rwt4/iBeaconScanner/iBeaconScanner.ino: In function 'void setup()':
/private/var/folders/2w/n4f1n28573345fr4zwssdkmc0000gn/T/.arduinoIDE-unsaved202335-73470-gxhntz.rwt4/iBeaconScanner/iBeaconScanner.ino:17:3: error: 'BTstack' was not declared in this scope
   17 |   BTstack.setup();
      |   ^~~~~~~
/private/var/folders/2w/n4f1n28573345fr4zwssdkmc0000gn/T/.arduinoIDE-unsaved202335-73470-gxhntz.rwt4/iBeaconScanner/iBeaconScanner.ino:18:39: error: 'advertisementCallback' was not declared in this scope
   18 |   BTstack.setBLEAdvertisementCallback(advertisementCallback);
      |                                       ^~~~~~~~~~~~~~~~~~~~~
/private/var/folders/2w/n4f1n28573345fr4zwssdkmc0000gn/T/.arduinoIDE-unsaved202335-73470-gxhntz.rwt4/iBeaconScanner/iBeaconScanner.ino: In function 'void loop()':
/private/var/folders/2w/n4f1n28573345fr4zwssdkmc0000gn/T/.arduinoIDE-unsaved202335-73470-gxhntz.rwt4/iBeaconScanner/iBeaconScanner.ino:24:3: error: 'BTstack' was not declared in this scope
   24 |   BTstack.loop();
      |   ^~~~~~~
/private/var/folders/2w/n4f1n28573345fr4zwssdkmc0000gn/T/.arduinoIDE-unsaved202335-73470-gxhntz.rwt4/iBeaconScanner/iBeaconScanner.ino: At global scope:
/private/var/folders/2w/n4f1n28573345fr4zwssdkmc0000gn/T/.arduinoIDE-unsaved202335-73470-gxhntz.rwt4/iBeaconScanner/iBeaconScanner.ino:37:6: error: variable or field 'advertisementCallback' declared void
   37 | void advertisementCallback(BLEAdvertisement *adv) {
      |      ^~~~~~~~~~~~~~~~~~~~~
/private/var/folders/2w/n4f1n28573345fr4zwssdkmc0000gn/T/.arduinoIDE-unsaved202335-73470-gxhntz.rwt4/iBeaconScanner/iBeaconScanner.ino:37:28: error: 'BLEAdvertisement' was not declared in this scope
   37 | void advertisementCallback(BLEAdvertisement *adv) {
      |                            ^~~~~~~~~~~~~~~~
/private/var/folders/2w/n4f1n28573345fr4zwssdkmc0000gn/T/.arduinoIDE-unsaved202335-73470-gxhntz.rwt4/iBeaconScanner/iBeaconScanner.ino:37:46: error: 'adv' was not declared in this scope
   37 | void advertisementCallback(BLEAdvertisement *adv) {
      |                                              ^~~

exit status 1

Compilation error: variable or field 'advertisementCallback' declared void

earlephilhower, 2023-04-06:
To be clear, I just added the Arduino library that the BTStack folks wrote themselves (with simple compilation fixes) so I really do mean the  "unsupported" part of the commit.  I think it may have been abandoned at some point because the version that's distributed in their repo actually has the wrong number of parameters in a callback (fixed in the copy of the lib here) which is basically fatal to any GCC never than ~7.0.

How did you install the core?  I'm able to build all of the examples without incident via my own `git` version.  Maybe the ZIP (IDE boards manager) install is munged somehow.

lyusupov, 2023-04-06:
> I am trying to compile iBeaconScanner.ino in the BTstack folder with board v.3.1.0

https://github.com/earlephilhower/arduino-pico/releases/tag/3.1.1

noahbaron, 2023-04-06:
Same procedure as you have outlined. I have a screenshot showing all the settings with the error output pasted previously. I have also tried with a dfferent Windows 11 PC laptop using the latest Arduino v1.8 and v2.0.4. I verified the location of the header files is:

btstack.h
/Users/noahneibaron/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1/pico-sdk/lib/btstack/src/btstack.h

spi.h
/Users/noahneibaron/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1/libraries/SPI/src/SPI.h

Not sure what else to check. The other examples are compiling, just not btstack. I understand it's not supported, however I was hoping to use the Pico W as a customized iBeacon to WiFi and iBeacon to serial adapter.

[image]

earlephilhower, 2023-04-06:
@lyusupov are you doing this on a Mac, too?  There might be a case sensitive filename issue, but afaik only Win32 NTFS has that problem, not Unix derived OSes like Linux and MacOS.

I'm away from the computer now, but I think the Arduino lib has a CAP BTStack.h while the internal header is all lowercase.  For case sensitive OSesi might need to rename the lib to BTStackLib and adjust the header to match to avoid this problem...

noahbaron, 2023-04-06:
I tried different cases on my Mac laptop with latest OS v13.1 <btstack.h> vs. <BTstack.h> both seemed ok.

One thing I noticed is Arduino is referencing the location for <BTstack.h> in possibly the wrong location since on further inspection, it appeared to have the SDK version of this file that as you said won't compile.

/Users/noahneibaron/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1/pico-sdk/lib/btstack/src/btstack.h (This location is the SDK version and Arduino 2.0.4 is showing this file is being referenced)
[image]

/Users/noahneibaron/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1/pico-sdk/lib/btstack/port/arduino/btstack.h (This location has your Arduino wrapper version)

earlephilhower, 2023-04-06:
No, that's the problem.  Per https://discussions.apple.com/thread/251191099 "BTStack.h" (the library header) and "btstack.h" (the internal file you as a user should never include) will both resolve (incorrectly) to the same filename.

So, looks like I need to rename their library too (in addition to fixing all the compilation errors already done) to make it work anywhere but Linux and the odd Windows w/NTFS case-sensitivity turned on.

earlephilhower, 2023-04-06:
Also, as for your use case, I did test the iBeacon.ino sketch and detected the PicoW with my cell phone, but I didn't have any other beacons to test so did not try your iBeaconScanner side. to verify it could properly detect any modern beacons (the library is ~8yrs old IIRC!).

noahbaron, 2023-04-06:
Great to hear the iBeacon transmitter is working. iBeacon is 10 years old
at this point but is still an active standard and still useful (for example
Tesla model 3 transmits an iBeacon and the Pico W could potentially detect
it now). Most low cost iBeacon to WiFi capable boards are based on ESP32 so
the Pico W might be a great alternative to the popular Espressif ecosystem.

closed 2023-04-07 by earlephilhower

lyusupov, 2023-04-07:
> @lyusupov are you doing this on a Mac, too?

Debian Linux

commit a37676ee (2023-04-06): Rename BTstack lib to BTstackLib

Fixes #1356

Mac and Windows have case-insensitive filesystems, so the will find
the internal (all-lowercase) "btstack.h" and not the library's "BTstack.h",
causing compilation errors.

Rename the library and header to avoid the issue.
PR #1360 (2023-04-07): Rename BTstack lib to BTstackLib
Fixes #1356

Mac and Windows have case-insensitive filesystems, so the will find the internal (all-lowercase) "btstack.h" and not the library's "BTstack.h", causing compilation errors.

Rename the library and header to avoid the issue.
commit 07781e9c (2023-04-07): Rename BTstack lib to BTstackLib (#1360)

Fixes #1356

Mac and Windows have case-insensitive filesystems, so the will find
the internal (all-lowercase) "btstack.h" and not the library's "BTstack.h",
causing compilation errors.

Rename the library and header to avoid the issue.