earlephilhower/arduino-pico #1362: Auto switch to Pico debug fails

AndreasMainz, 2023-04-07:
I'm using Pico debug and according to your doc it says:
Under Windows and macOS, any user should be able to access pico-debug automatically..

but if Debug is started it reports: (using the 3. button next to "Download" Button)
_**[2023-04-07T07:02:36.304Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Users\\Andreas\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-openocd\\1.5.0-b-c7bab52/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "c:\\Users\\Andreas\\Documents\\Arduino\\Knx_RP_Test_GAs_01" -f "C:/Program Files/Arduino IDE/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f "C:\\Users\\Andreas\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\3.1.1/lib/picoprobe.tcl"
Open On-Chip Debugger 0.11.0-g228ede43d-dirty (2022-12-22-20:10)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : only one transport option; autoselect 'swd'
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Error: Failed to open or find the device
Error: Can't find a picoprobe device! Please check device connections and permissions.

[2023-04-07T07:02:36.371Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...**_

NOT clear where to switch to Pico-Debug, but Download via Pico Debug works.
Using Arduino 2.0.4

Many thanks for any advice..

earlephilhower, 2023-04-07:
This is an IDE 2.0 limitation AFAIK.

They do not support actually having multiple debug configs (even when specified in the platform.txt file) as far as I can see, and only use the first one defined which in this case is Picoprobe because that's the most common.  (To be fair, I think the debug stuff is/was still undocumented and not officially supported.)  You can see that it is trying to use a `picoprobe` interface, not `pico-debug` in the OCD error message:
````
 Error: Can't find a picoprobe device! Please check device connections and permissions.
````
Uploading, as you saw, does work and that's the best we can do right now until the platform.txt support is updated in the IDE, I believe, sorry!

closed 2023-04-07 by earlephilhower

AndreasMainz, 2023-04-07:
Does it work in IDE 1.x?

AndreasMainz, 2023-04-08:
Hi Earle,thanks for your reply even if the answer is no good news..Could you give me some advice where to change the call to the gdb?It must be located in some json files..so I could change the default settings to cmsis debugger.This would be very helpful.If we succeed I would publish the solution in the forum..Many thanks for your support and best regards from Germany Viele Grüße Andreas Kohl