earlephilhower/arduino-pico #1364: Feature Request: Support for Black Magic Probe (for PlatformIO)

Riffer, 2023-04-08:
Since release 1.8.0 BMP supports RP2040: https://github.com/blackmagic-debug/blackmagic/releases/tag/v1.8.0 - a BMP adapter can be easily made from a Blue (STM32F1) or Blackpill (STM32F4) and was nicely integrated into platformio for the STM32 platforms.

Please add support for the pico, as well.

I read somewhere it should not be too hard, but I have no clue - I am also not sure if this request is better here at https://github.com/maxgerhardt/platform-raspberrypi and who needs to do what.

maxgerhardt, 2023-04-08:
Yes, support of that needes to be added in platform-raspberrypi if the PlatformIO is supposed to benefit from this, but maybe the core here can also allow that as an upload method.

Do you have a precompiled binary for a bluepull (C8 or CB version I guess?) that I can test with? Which pins will SWDI+SWDCLK be on then? Is native USB-CDC used to create the serial port?

maxgerhardt, 2023-04-08:
I did a small addition based on platform-ststm32 code. Can you please try
```
[env]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git#bmp
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
monitor_speed = 115200

[env:rpipico]
board = rpipico
upload_protocol = blackmagic
```
as your `platformio.ini` and then do the project task Build, then Advanced->Verbose Upload and post the logs?

Riffer, 2023-04-09:
I just compiled BMP version 1.9 and uploaded the two files you need for the update.:
https://drive.google.com/file/d/1HB76_w14D6Vky2fsFd9YlcRu8dhHTcJn/view?usp=share_link

In the best case, you have an STLink for the flash job. Find an excellent manual [here](https://primalcortex.wordpress.com/2017/06/13/building-a-black-magic-debug-probe/).

I will come back as soon as possible with the requested log - thank you very much for your work!

maxgerhardt, 2023-04-09:
Love it, I've got the BMP firmware flashed onto my Bluepill now in accordance to that document and GDB can connect to the BMP. 
```
C:\Users\Max\.platformio\packages\toolchain-rp2040-earlephilhower\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
GNU gdb (GDB) 8.2.50.20190202-git
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
(gdb) target extended-remote \\.\COM15
Remote debugging using \\.\COM15
(gdb) monitor help
General commands:
        version -- Display firmware version info
        help -- Display help for monitor commands
        jtag_scan -- Scan JTAG chain for devices
        swdp_scan -- Scan SW-DP for devices
        auto_scan -- Automatically scan all chain types for devices
        frequency -- set minimum high and low times
        targets -- Display list of available targets
        morse -- Display morse error message
        halt_timeout -- Timeout (ms) to wait until Cortex-M is halted: (Default 2000)
        connect_rst -- Configure connect under reset: (enable|disable)
        reset -- Pulse the nRST line - disconnects target
        tdi_low_reset -- Pulse nRST with TDI set low to attempt to wake certain targets up (eg LPC82x)
        traceswo -- Start trace capture, NRZ mode: (baudrate) (decode channel ...)
        heapinfo -- Set semihosting heapinfo
(gdb) monitor version
Black Magic Probe (ST-Link/v2) v1.9.0-dirty, Hardware Version 0
Copyright (C) 2022 Black Magic Debug Project
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
```

Now let's see if it can attach to a Raspberry Pi Pico...

maxgerhardt, 2023-04-09:
```
(gdb) monitor swdp_scan
Target voltage: 3.51V
Available Targets:
No. Att Driver
 1      Raspberry RP2040 M0+
 2      Raspberry RP2040 M0+
 3      Raspberry RP2040 Rescue (Attach to reset!)
(gdb) att 1
Attaching to Remote target
warning: No executable has been specified and target does not support
determining executable automatically.  Try using the "file" command.
0x10008a2c in ?? ()
(gdb) i r
r0             0x0                 0
r1             0x1                 1
r2             0x2000a2bb          536912571
r3             0x100049f9          268454393
r4             0x0                 0
r5             0x0                 0
r6             0x0                 0
...
```
First try :) But.. we can't debug both cores at the same time, we have to select one? Let's try and get this into GDB.

maxgerhardt, 2023-04-09:
Flashing and debugging seems very unreliable with it. Sometimes it hangs up trying to connect to the probe, sometimes flashing fails, sometimes it passes. Had to power cycle it and press the reset button on the bluepill multiple times.

Same with debugging. Had to have two runs for it to be able to connect and flash properly.

But, when it does work.. 

[image]

maxgerhardt, 2023-04-09:
Per linked issue above, the debugging and flashing experience is better with the latest BMP commit's firmware, but still not hickup-free. For 3 debugging runs it failed on one, but recovered without pressing the reset button for the next run.

Let me know what your debugging experience is.

Riffer, 2023-04-10:
I see myself more as a pure hobbyist when it comes to debugging (most times I fix my problems with a serial out and use the BMP as a uniform and fast flashing tool for my small STM32 projects) and I can't rate your intensive participation in my request highly enough. Thank you very, very much!

I agree with all your observations, including the problem with every second start that it does not work.

Seeing the latest answer in https://github.com/blackmagic-debug/blackmagic/issues/1364#issuecomment-1501377076 I first tried that one but it did not work for me, the command "monitor swdp_scan" failed at all times.

After that, I flashed the version from 1.8.2 you mentioned before in that issue to my blue pill, which works exactly as you say like a flipflop.

But that is an issue of BMP, and I would see my request here as completely fulfilled. 

Take my warmest thanks for all of your work!

PR #1378 (2023-04-15): Enable BlackMagicProbe for PlatformIO
Fixes #1364.

By adding `"blackmagic"` in the protocols array in `makeboards.py`, all PlatformIO board definitions are marked to be usable with the BlackMagicProbe.

With the instability issues fixed in the latest versions of the BMP firmware per https://github.com/blackmagic-debug/blackmagic/issues/1364, this can be used now.

This does **not** add upload or debugging support for the Arduino IDE, that would have to be separate. Although it shouldn't be to difficult, just see my used flashing command in the linked issue.

This PR goes hand-in-hand with htt [... cut]
closed 2023-04-15 by earlephilhower