earlephilhower/arduino-pico #1369: Nano rp2040 connect extra analog pins

nakeze, 2023-04-11:
I just started with the nano connect, I was previously working on the nano 33 ble. I try rp2040 because of the dual core capacity but with the mbed_os library it is not convenient at all. 

I need to read 6 analog pins, so I need to read A6 and A7 which are on the Nina W102 module. To test this, I made a simple sketch that reads the analog pins on the board using analogRead() function (the one from the arduino library) : 
```
#include "WiFiNINA.h"
NinaPin A7NP = A7;
// the setup routine runs once when you press reset:
void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
}

// the loop routine runs over and over again forever:
void loop() {
  // read the input on analog pin 0:
  int sensorValue = analogRead(A0); Serial.print("A0 :");Serial.println(sensorValue); delay(1);        // delay in between reads for stability
  sensorValue = analogRead(A1); Serial.print("A1 :");Serial.println(sensorValue); delay(1);        // delay in between reads for stability
  sensorValue = analogRead(A2); Serial.print("A2 :");Serial.println(sensorValue); delay(1);        // delay in between reads for stability
  sensorValue = analogRead(A3); Serial.print("A3 :");Serial.println(sensorValue); delay(1);        // delay in between reads for stability
  //int sensorValue = analogRead(A4); Serial.print("A4 :");Serial.println(sensorValue); delay(1);        // delay in between reads for stability
  //int sensorValue = analogRead(A5); Serial.print("A5 :");Serial.println(sensorValue); delay(1);        // delay in between reads for stability
  sensorValue = analogRead(NinaPin(A6)); Serial.print("A6 :");Serial.println(sensorValue); delay(1);        // delay in between reads for stability
  sensorValue = analogRead(A7NP); Serial.print("A7 :");Serial.println(sensorValue); delay(1);        // delay in between reads for stability
  delay(1000);
}
```
I first tried from the official rp2040 core based on mbed os it compiles and works fine.

I tried to use the same sketch with this library as target and I have the compiling error "undefined reference to getAnalogReadResolution()".

I tried to replace the call for getAnalogReadResolution with an integer value (1024), it compiles but the analogRead() call returns 0 and is executed in a microsecond.
So it is not event trying to communicate with the nina module.

I used Kdiff to see differences in the files from the two libraries variants folder,.
I found in the nina_pins.h a difference with NINA_ATTRIBUTES :

```
#if __has_include("WiFiNINA.h")
 # define NINA_ATTRIBUTE
 #else
 #define NINA_ATTRIBUTE __attribute__ ((error("Please include WiFiNINA.h to use this pin")))
#endif
```

and the NINA_ATTRIBUTE set in the function declaration lines at the end of the file.

The pins_arduino.h files are also different.
There are also files that exist in one folder and not in the other.

I tried to add this part of code to the file and it gives this error :
error: call to 'analogRead' declared with attribute error: Please include WiFiNINA.h to use this pin

This error appears even with : #include "WifiNINA.h" in the sketch.

I saw that this "__has_include" creating compilation errors had been discussed in a previous issue that had been closed, but it did not made the analog read working. 

Do you have tips to dodge this issue ?

earlephilhower, 2023-04-11:
Sorry, I don't have an Arduino Nano to work on so this is going to need your or someone else's assistance.

The base problem, I imagine, is that the Arduino Nano board has some special ADC with additional pins and maybe better read resolution.  That means that the `variants/arduino_nano_connect` should have an override for `analogRead` (and associated calls) since the core's `analogRead` only works on the internal RP2040 pins.

You might want to look at the mbed core's implementation, and see if you can cut-n-paste it into the local variant here.

JAndrassy, 2023-04-11:
did you try `sensorValue = analogRead(A6);`?

JAndrassy, 2023-04-11:
in the Arduino Nano RP2040 core there is getAnalogReadResolution in wiring_analog.cpp. this core doesn't have getAnalogReadResolution anywhere.
it is needed here
```
return WiFiDrv::analogRead(adc_channel) >> (12 - pin.analogReadResolution());
```
@nakeze  1024 instead of analogReadResolution? 12 or 10 would be a good value

nakeze, 2023-04-12:
Thanks for taking time to answer.
@JAndrassy I tried with 10 or 12 and it worked ! I should have look after the calls of this function in WifiNina library... 10 gives the value more "arduino compatible" as the max value of analogRead for analog pins  is 1024.
 Thanks a lot for this patch.
I will now dive into the module's datasheet and in WifiNina library.

closed 2023-08-26 by earlephilhower