earlephilhower/arduino-pico #1370: LECentral and LEPeripheral not compiling

dpharris, 2023-04-11:
Hi --
First, thank you for the great work!
I am trying to compile the two examples.  I see that the BTstack was changed to BTstackLib, and I downloaded the directory.  The examples list shows BTStack, and the examples do have #include <BTStackLib>,
but they both complain: 

In file included from /Users/daveharris/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1/libraries/BTstackLib/src/BTstackLib.cpp:41:
/Users/daveharris/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1//pico-sdk/lib/btstack/src/ble/sm.h:228:1: error: 'irk_lookup_state_t' does not name a type
  228 | irk_lookup_state_t sm_identity_resolving_state(hci_con_handle_t con_handle);
      | ^~~~~~~~~~~~~~~~~~
Using library BTstackLib at version 1.0 in folder: /Users/daveharris/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1/libraries/BTstackLib 
Using library SPI at version 1.0 in folder: /Users/daveharris/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1/libraries/SPI 

I looked at hci.h and it appears to have an appropriate definition for irk_lookup_state_t

Can you help?

earlephilhower, 2023-04-12:
Try enabling Bluetooth in the menus.  It takes significant RAM and flash, so isn't on by default.  `Tools->IP/Bluetooth Stack-> IPv4 + BT`

dpharris, 2023-04-12:
Thanks, that worked a charm.
Should add a comment to do the same in the examples.

earlephilhower, 2023-04-12:
Good suggestion.  The new PR gives this error and aborts compile immediately:
````
arduino15/packages/rp2040/tools/pqt-python3/1.0.1-base-3a57aed/python3 -I /home/earle/Arduino/hardware/pico/rp2040/tools/signing.py --mode header --publickey /home/earle/Arduino/hardware/pico/rp2040/libraries/BTstackLib/examples/iBeacon/public.key --out /tmp/arduino_build_761369/core/Updater_Signing.h
/home/earle/.arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/arm-none-eabi-g++ -I /tmp/arduino_build_761369/core -c -Wall -Wextra -Werror=return-type -Wno-ignored-qualifiers -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0xf00a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico W\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipicow\"" -DTARGET_RP2040 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -MMD -iprefix/home/earle/Arduino/hardware/pico/rp2040/ @/home/earle/Arduino/hardware/pico/rp2040/lib/platform_inc.txt -I/home/earle/Arduino/hardware/pico/rp2040/include -fno-rtti -std=gnu++17 -g -pipe -DF_CPU=133000000L -DARDUINO=10816 -DARDUINO_RASPBERRY_PI_PICO_W "-DBOARD_NAME=\"RASPBERRY_PI_PICO_W\"" -DARDUINO_ARCH_RP2040 "-DSTASSID=\"NOBABIES\"" "-DSTAPSK=\"ElephantsAreGreat\"" -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -I/home/earle/Arduino/hardware/pico/rp2040/cores/rp2040 -I/home/earle/Arduino/hardware/pico/rp2040/variants/rpipicow -I/home/earle/Arduino/hardware/pico/rp2040/libraries/BTstackLib/src -I/home/earle/Arduino/hardware/pico/rp2040/libraries/SPI/src /tmp/arduino_build_761369/sketch/iBeacon.ino.cpp -o /tmp/arduino_build_761369/sketch/iBeacon.ino.cpp.o
In file included from /home/earle/Arduino/hardware/pico/rp2040/libraries/BTstackLib/src/BTstackLib.h:5,
                 from /home/earle/Arduino/hardware/pico/rp2040/libraries/BTstackLib/examples/iBeacon/iBeacon.ino:1:
/home/earle/Arduino/hardware/pico/rp2040/cores/rp2040/_needsbt.h:4:24: error: static assertion failed: This library needs Bluetooth enabled.  Use the 'Tools->IP/Bluetooth Stack' menu in the IDE to enable it.
    4 | #define ENABLE_CLASSIC 0
      |                        ^
/home/earle/Arduino/hardware/pico/rp2040/cores/rp2040/_needsbt.h:7:15: note: in expansion of macro 'ENABLE_CLASSIC'
    7 | static_assert(ENABLE_CLASSIC, "This library needs Bluetooth enabled.  Use the 'Tools->IP/Bluetooth Stack' menu in the IDE to enable it.");
      |               ^~~~~~~~~~~~~~
````

closed 2023-04-12 by earlephilhower

lu7did, 2023-05-01:
Hello, thank you for your work. I'd enabled the Tools->IP Bluetooth Stack->ipv4 + BT option, for example

iBeacon.ino fails with "Compilation error: 'UUID' does not name a type"

LEPeripheral fails with many errors, among them /private/var/folders/4b/ywr8xsfj2bqcqw2593cpc_rc0000gp/T/.arduinoIDE-unsaved202341-24882-5sks23.pohrd/LEPeripheral/LEPeripheral.ino:54:30: error: 'BLEStatus' was not declared in this scope; did you mean 'PinStatus'?
   54 | void deviceConnectedCallback(BLEStatus status, BLEDevice *device) {

iBeaconScanner.ino fails with
/private/var/folders/4b/ywr8xsfj2bqcqw2593cpc_rc0000gp/T/.arduinoIDE-unsaved202341-24882-1cfxnkk.c59e/iBeaconScanner/iBeaconScanner.ino:37:6: error: variable or field 'advertisementCallback' declared void
   37 | void advertisementCallback(BLEAdvertisement *adv) {

So it seems that I'm missing some small but crucial configuration step somewhere. I checked on the documentation and the previous issues but to no avail.

Running version 3.1.1 of the core.

Any guidance you can provide? 
Thanks in advance! Pedro.

mindeon, 2023-08-01:
wonderful work, thank you !
Any ideas how to solve the same problem in PlatformIO ?

PR #1371 (2023-04-12): Clear compile errors if BT needed but not on
Fixes #1370

Adds a simple helper assertion to tell the user how to enable BT if it's not enabled, instead of some odd compilation warnings about undefined functions.
commit 3dcb4bbf (2023-04-12): Clear compile errors if BT needed but not on (#1371)

Fixes #1370

Adds a simple helper assertion to tell the user how to enable BT if it's
not enabled, instead of some odd compilation warnings about undefined
functions.