earlephilhower/arduino-pico #1377: BTStack examples compile errors

hreintke, 2023-04-14:
When compiling an example from the BTStack library, the compile fails with : 

f.e. the ANCS 
```
In file included from C:\Users\Herman\AppData\Local\Temp\.arduinoIDE-unsaved2023314-23392-19w9qpp.uq3y\ANCS\ANCS.ino:6:
c:\users\herman\appdata\local\arduino15\packages\rp2040\hardware\rp2040\3.1.1\pico-sdk\lib\btstack\src\ble\sm.h:228:1: error: 'irk_lookup_state_t' does not name a type
  228 | irk_lookup_state_t sm_identity_resolving_state(hci_con_handle_t con_handle);
```

Do I need some additional configuration steps to use these ?

earlephilhower, 2023-04-14:
You need to enable Bluetooth in the menus. #1371 add an assertion that is a little clearer than the compiler warnings you showed.

closed 2023-04-14 by earlephilhower

hreintke, 2023-04-14:
Sorry, missed that option setting.

When setting the option correctly, all compiles OK.

Thanks

earlephilhower, 2023-04-14:
No worries.  No guarantees the BTstack library examples still work, but until someone writes and ArduinoBLE-compatible library it's a start...