earlephilhower/arduino-pico #1379: c++20 warnings fix

GFoxPM, 2023-04-17:
warning: compound assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]
[/cores/rp2040/RP2040Support.h#L361](https://github.com/earlephilhower/arduino-pico/blob/c276a36c9b33620df46b3dc0013b29656de55301/cores/rp2040/RP2040Support.h#L361)
```diff
-  rp2040._epoch += 1LL << 24;
+  rp2040._epoch = rp2040._epoch + (1LL << 24);
``` 
[/cores/rp2040/Bootsel.cpp#L38](https://github.com/earlephilhower/arduino-pico/blob/c276a36c9b33620df46b3dc0013b29656de55301/cores/rp2040/Bootsel.cpp#L38)
```diff
-  for (volatile int i = 0; i < 1000; ++i);
+  for (int i = 0; i < 1000; ++i); // ?
``` 
[/cores/rp2040/SerialPIO.cpp#L237](https://github.com/earlephilhower/arduino-pico/blob/c276a36c9b33620df46b3dc0013b29656de55301/cores/rp2040/SerialPIO.cpp#L237)
```diff
-  _rxPIO->sm[_rxSM].shiftctrl |= 0x80000000;
+  _rxPIO->sm[_rxSM].shiftctrl = _rxPIO->sm[_rxSM].shiftctrl | 0x80000000;
```

maxgerhardt, 2023-04-17:
The second change will optimize the loop away completeley, so its intended function of providing a delay is destroyed.

maxgerhardt, 2023-04-17:
Hm, using `i = i +1` instead of `++1` generates the `warning: using value of simple assignment with 'volatile'-qualified left operand is deprecated [-Wvolatile]ARM gcc 11.2.1 (none) #1` warning but then https://github.com/raspberrypi/pico-sdk/issues/1017#issuecomment-1242814904 says that this has already been deprecated as warning.

Other alternatives generated different microcode, so it would change the timing. Not sure if it's enough to break BOOTSEL functionality.

See https://godbolt.org/z/77TxzWM87

earlephilhower, 2023-04-17:
For the 2nd one I think the following is the safest:

````
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wno-volatile"
<for loop>
#pragma GCC diagnostic pop
````

That said, this is something that wouldn't be looked at until the move from GCC-10 to GCC-12, whenever it comes around.  Also, per @maxgerhardt 's linked issue the C++ guys seem to have dropped this as a warning going forward so maybe by then it will be a non-issue.

GFoxPM, 2023-04-17:
@earlephilhower 
warning: unknown option after '#pragma GCC diagnostic' kind

Is there more information about the lack of warnings in GCC12?
Because in ARM GCC 12.2 (Linux) the warnings are still there, and it's unlikely to be platform dependent...

Doesn't generate a warning:
[/cores/rp2040/Bootsel.cpp#L38](https://github.com/earlephilhower/arduino-pico/blob/c276a36c9b33620df46b3dc0013b29656de55301/cores/rp2040/Bootsel.cpp#L38)
```diff
-  for (volatile int i = 0; i < 1000; ++i);
+  volatile int i = 1000;
+  while(i) { i = i - 1; }
```

earlephilhower, 2023-04-19:
@GFoxPM are you using `--std=gnu++20` now?  Would you like to put in a PR with all your suggested changes?  We can pull them in now if there is no issue w/our current C++ level, although we may accidentally introduce other problems between now and when we go to C++20...

GFoxPM, 2023-04-19:
@earlephilhower Yes, I'm currently using `--std=gnu++20`
But I'm not sure about the completeness of my edits, because this is only what the compiler warned about.
A lot can also be hidden under `#ifdef`

earlephilhower, 2023-06-05:
Per https://github.com/esp8266/Arduino/issues/8916#issuecomment-1519095142 it seems these `volatile `warnings are de-deprecated in C++23.  So, closing this as a won't-fix.

closed 2023-06-05 by earlephilhower