earlephilhower/arduino-pico #1396: Can not use more than 8 Servo instances

aentinger, 2023-04-24:
Hi :coffee: :wave: 

I'm using the PIO based servo library bundled with this code and I'm running into an issue that I can not use more than 8 PIO based servo instances.

Here's a somewhat minimal code:
```C++
static size_t constexpr NUM_SERVOS = 12;
static std::array<Servo, NUM_SERVOS> servo_ctrl;
static std::array<pin_size_t, NUM_SERVOS> SERVO_PINS =
{
   2, /* GP2  = SERVO00 */
   3, /* GP3  = SERVO01 */
   6, /* GP6  = SERVO02 */
   7, /* GP7  = SERVO03 */
   8, /* GP8  = SERVO04 */
   9, /* GP9  = SERVO05 */
  10, /* GP10 = SERVO06 */
  11, /* GP11 = SERVO07 */
  12, /* GP12 = SERVO08 */
  13, /* GP13 = SERVO09 */
  14, /* GP14 = SERVO10 */
  15  /* GP15 = SERVO11 */
};
/* ... */
  for (size_t s = 0; s < NUM_SERVOS; s++)
    if (servo_ctrl[s].attach(SERVO_PINS[s], DEFAULT_MIN_PULSE_WIDTH, DEFAULT_MAX_PULSE_WIDTH, DEFAULT_NEUTRAL_PULSE_WIDTH) < 0)
      DBG_ERROR("Failed to attach servo pin %d", s);
```

What's the max amount of servos supported for this library? Can the limitation be circumvented using other pins?

Cheers, Alex

aentinger, 2023-04-24:
CC @generationmake

earlephilhower, 2023-04-24:
It uses the PIO state machine, so you get 8 max, assuming no other demand on the PIOs. That's all the hardware has.

I suppose you could write your own IRQ based servo control, but that would have much more jitter (vs 0 for the PIOs) and use CPU cycles (vs 0 for PIOs).

PWM doesn't give good resolution given that only ~5% of the cycle time and bits are meaningful for servos.

closed 2023-04-24 by earlephilhower

aentinger, 2023-04-24:
I've quickly glued something together, if more than 8 servos (obtainable via the PIO servo library shipped with this core) are needed - https://github.com/107-systems/107-Arduino-Servo-RP2040 .