earlephilhower/arduino-pico #1398: SD library compilation errors

obdevel, 2023-04-24:
I'm getting compilation errors when trying to use the SD library. The SD card reader is on the back of a TFT LCD display which is already working fine using the TFT_eSPI library.

Latest core version 3.1.1
Arduino IDE 1.8.19
macOS 12.3.1
Pico W

I've created a minimal test sketch based on the example given in a recent discussion (https://github.com/earlephilhower/arduino-pico/discussions/925):

```
#include <SPI.h>
#include <SD.h>

#define SPI1_MOSI 11
#define SPI1_MISO 8
#define SPI1_SCK 10
#define SD_CS 0

void setup() {

  Serial.begin(115200);
  while (!(Serial));

  Serial.printf("\n\n> SD card test\n\n");

  sd_card_test();

  Serial.printf("\n> setup complete\n");
}

void loop() {

}

bool sd_card_test(void) {

  bool card_present, ret = true;
  char data_buffer[1024];
  size_t bytes_read;
  File f;

  SPI1.setTX(SPI1_MOSI);
  SPI1.setRX(SPI1_MISO);
  SPI1.setSCK(SPI1_SCK);
  SPI1.setCS(SD_CS);

  card_present = SD.begin(SD_CS, SPI1);

  if (card_present) {
    Serial.printf("> SD card is present\n");

    f = SD.open("config.txt", "r");

    if (!f) {
      Serial.printf("> ** error opening config file\n");
      ret = false;
    }

    bytes_read = f.readBytes(data_buffer, sizeof(data_buffer));
    Serial.printf("> ** read %u bytes from config file\n", bytes_read);

    if (bytes_read == 0) {
      ret = false;
    }

    f.close();

  } else {
    Serial.printf("> ** SD card is not present\n");
    ret = false;
[... 101 lines cut ...]
      |                                                 ^
In file included from /Users/xxx/Documents/Arduino/libraries/SdFat/src/FatLib/FatVolume.h:27,
                 from /Users/xxx/Documents/Arduino/libraries/SdFat/src/FatLib/FatLib.h:28,
                 from /Users/xxx/Documents/Arduino/libraries/SdFat/src/SdFat.h:32,
                 from /Users/xxx/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1/libraries/SDFS/src/SDFS.h:35,
                 from /Users/xxx/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1/libraries/SDFS/src/SDFS.cpp:27:
/Users/xxx/Documents/Arduino/libraries/SdFat/src/FatLib/FatFile.h:1023:7: note: candidate: 'constexpr File32& File32::operator=(const File32&)'
 1023 | class File32 : public StreamFile<FatFile, uint32_t> {
      |       ^~~~~~
/Users/xxx/Documents/Arduino/libraries/SdFat/src/FatLib/FatFile.h:1023:7: note:   no known conversion for argument 1 from 'FsFile' to 'const File32&'
/Users/xxx/Documents/Arduino/libraries/SdFat/src/FatLib/FatFile.h:1023:7: note: candidate: 'constexpr File32& File32::operator=(File32&&)'
/Users/xxx/Documents/Arduino/libraries/SdFat/src/FatLib/FatFile.h:1023:7: note:   no known conversion for argument 1 from 'FsFile' to 'File32&&'
Multiple libraries were found for "SD.h"
 Used: /Users/xxx/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1/libraries/SD
 Not used: /Applications/Arduino.app/Contents/Java/libraries/SD
Multiple libraries were found for "SdFat.h"
 Used: /Users/xxx/Documents/Arduino/libraries/SdFat
 Not used: /Users/xxx/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1/libraries/ESP8266SdFat
Using library SPI at version 1.0 in folder: /Users/xxx/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1/libraries/SPI 
Using library SD at version 2.0.0 in folder: /Users/xxx/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1/libraries/SD 
Using library SDFS at version 0.1.0 in folder: /Users/xxx/Library/Arduino15/packages/rp2040/hardware/rp2040/3.1.1/libraries/SDFS 
Using library SdFat at version 2.2.2 in folder: /Users/xxx/Documents/Arduino/libraries/SdFat 
exit status 1
Error compiling for board Raspberry Pi Pico W.
```

earlephilhower, 2023-04-24:
Your trying to use the upstream SDFAT, not the one included with this core.  Remove that other copy and you should be good.

obdevel, 2023-04-24:
Thanks, that fixed it. I just need to remember to put the other one back when working with AVRs !

closed 2023-04-24 by obdevel