earlephilhower/arduino-pico #1402: LitteleFS and USBSerial cannot be used at the same time

MC-YasuhiroHirano, 2023-04-27:
### Discussed in https://github.com/earlephilhower/arduino-pico/discussions/1374

<div type='discussions-op-text'>

<sup>Originally posted by **MC-YasuhiroHirano** April 13, 2023</sup>
Hello, thanks for this useful software !
I'm using rp2040 with platformio.
Also, I'm using FreeRTOS.

I want to write files using LitteleFS and keyboard input using USB serial asynchronously.
However, when I try to perform this process, the program stuck on keyboard input.
I tried it in the main loop of core0 instead of asynchronous and it stuck as well.

Paste MCVE.

```cpp:test.cpp
#include <Arduino.h>
#include <FreeRTOS.h>
#include <LittleFS.h>

int cnt;

void setup() {
  Serial.begin(115200, true);
  cnt = 0;
  LittleFS.begin();
}

void loop() {
  char buf[64] = {0};
  File _lfs;
  int ret = 0;
  char c;
  delay(10);
  
  cnt++;
  if(!(cnt % 100))
  {
    sprintf(buf,"cnt[%d]\r\n",cnt);
    Serial.printf(buf);
    _lfs = LittleFS.open("/logfile.log", "a");
    if(_lfs){
      ret = _lfs.write(buf,strlen(buf));
    }
    _lfs.close();
  }

  while(Serial.available()>0) {
    c = Serial.read();
    Serial.print(c);
  }
}
```

MCVE only writes files periodically.

Please run MCVE, connect the USB cable and repeat keyboard input through the COM port.
The program will stuck.

* It will not stuck without keyboard input.
* If you comment out the LitteleFS process (open,write,close), the keyboard input does not stuck.
* The version of aruduino-pico is 3.1.0.
* The callstack when stuck varies for core0, but is always the same for core1. (see attached image)

[image]

Is there a problem with my usage?
Are there any restrictions or precautions I need to take when using LitteleFS?</div>

earlephilhower, 2023-05-02:
I was able to reproduce your failure while pasting large amounts of text into `minicom` in your MCVE.  It turns out the USB mutex used in the FreeRTOS USB stack was the wrong one, leading to a deadlock as you showed in the call stacks.  With #1419 I can's make it crash no matter how much I paste in.

closed 2023-05-02 by earlephilhower

MC-YasuhiroHirano, 2023-05-09:
@earlephilhower 

Thanks for the response.
I have checked in the latest environment and can confirm that it does not get stuck.

PR #1419 (2023-05-02): Fix USB crashes in FreeRTOS
Fixes #1402

The global USB mutex is auto-shadowed with a FreeRTOS semaphore while in FreeRTOS mode.  Unfortunately, while the core was using the proper FreeRTOS semaphore to lock access to the USB port, the actual FreeRTOS USB task was using the naked Pico SDK mutex, leading to cases where it could acquire the mutex even though some other FreeRTOS task actually owned the shadowed mutex.

Properly lock the shadowed Semaphore, not mutex_t, in the FreeRTOS USB periodic task.
commit 5a949443 (2023-05-02): Fix USB crashes in FreeRTOS (#1419)

Fixes #1402

The global USB mutex is auto-shadowed with a FreeRTOS semaphore while in
FreeRTOS mode.  Unfortunately, while the core was using the proper
FreeRTOS semaphore to lock access to the USB port, the actual FreeRTOS
USB task was using the naked Pico SDK mutex, leading to cases where it
could acquire the mutex even though some other FreeRTOS task actually
owned the shadowed mutex.

Properly lock the shadowed Semaphore, not mutex_t, in the FreeRTOS
USB periodic task.