earlephilhower/arduino-pico #1406: Compiling firmware via platformio does not output mouse coordinates. It works when firmware is compiled via Arduino IDE.

SocrateCrypto, 2023-04-27:
[env:pico]

platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
lib_archive = no
lib_deps = 
	
	
	adafruit/Adafruit TinyUSB Library@^2.1.0

 
board_build.f_cpu = 120000000L
build_flags = -fstack-protector 

       

     -D PIO_FRAMEWORK_ARDUINO_ENABLE_RTTI

   -D USE_TINYUSB
   -D DEBUG_RP2040_CORE

**********************************************************************************
/*********************************************************************
 Adafruit invests time and resources providing this open source code,
 please support Adafruit and open-source hardware by purchasing
 products from Adafruit!

 MIT license, check LICENSE for more information
 Copyright (c) 2019 Ha Thach for Adafruit Industries
 All text above, and the splash screen below must be included in
 any redistribution
*********************************************************************/

/* This example demonstrates use of both device and host, where
 * - Device run on native usb controller (controller0)
 * - Host run on bit-banging 2 GPIOs with the help of Pico-PIO-USB library (controller1)
 *
 * Example sketch receive keyboard report from host interface (from e.g consumer keyboard)
 * and remap it to another key and send it via device interface (to PC). For simplicity,
 * this example only toggle shift key to the report, effectively remap:
 * - all character key <-> upper case
 * - number <-> its symbol (with shift)
 *
 * Requirements:
 * - [Pico-PIO-USB](https://github.com/sekigon-gonnoc/Pico-PIO-USB) library
 * - 2 consecutive GPIOs: D+ is defined by PIN_PIO_USB_HOST_DP, D- = D+ +1
 * - Provide VBus (5v) and GND for peripheral
 * - CPU Speed must be either 120 or 240 Mhz. Selected via "Menu -> CPU Speed"
 */

// pio-usb is required for rp2040 host
#include "pio_usb.h"
#include "Adafruit_TinyUSB.h"

// Pin D+ for host, D- = D+ + 1
#ifndef PIN_USB_HOST_DP
#define PIN_USB_HOST_DP 5
#endif

// Pin for enabling Host VBUS. comment out if not used
// #ifndef PIN_5V_EN
// #define PIN_5V_EN        18
// #endif

#ifndef PIN_5V_EN_STATE
#define PIN_5V_EN_STATE 25
#endif
[... 366 lines cut ...]
.stack_dummy          2048   537137152

.ARM.attributes         40           0

.comment                18           0

.debug_info         286161           0

.debug_abbrev        56953           0

.debug_loc          153331           0

.debug_aranges        7128           0

.debug_ranges        28128           0

.debug_line         149405           0

.debug_str           34394           0

.debug_frame         15180           0

Total               847502
"C:\Users\38099\.platformio\penv\Scripts\python.exe" "C:\Users\38099\.platformio\packages\framework-arduinopico\tools\signing.py" --mode sign --privatekey "C:\Users\38099\OneDrive\Documents\PlatformIO\Projects\lvgl\src\private.key" --bin .pio\build\pico\firmware.bin --out .pio\build\pico\firmware.bin.signed
================================== [SUCCESS] Took 45.43 seconds

earlephilhower, 2023-04-27:
I think the easiest way to figure this out would be to capture a couple of the actual compilation commands that PIO is using and compare that to the ones the IDE is.  Probably a define missing or renamed somewhere.  Can you get the compilation command line for one file on both P.IO and Arduino IDE?  For the IDE there's a File->Prefs->Compilation->Verbose type option.  I unfortunately can't help find the equivalent for P.IO.

SocrateCrypto, 2023-04-28:
Arduino IDE 
*****************************************************
https://github.com/SocrateCrypto/mouse-coordinate/blob/main/arduinoIDE.txt

Platformio: (im usung comand:   pio run -vvv
https://github.com/SocrateCrypto/mouse-coordinate/blob/main/PlatformioCompil.txt

Arduino IDE work , PIO - dont work :))
Unfortunately, I cannot copy the output from Platformio entirely. Thank you for your help."

SocrateCrypto, 2023-04-28:
Perhaps this is related to these warnings, but unfortunately I don't know how to fix them.

Warning! Classic Pre Processor is used for C:\Users\38099\OneDrive\Documents\PlatformIO\Projects\lvgl\.pio\libdeps\pico\Adafruit TinyUSB Library\src\Adafruit_TinyUSB.h, advanced has failed with
Incorrect number of arguments to TU_CHECK_MCU'Warning! Classic Pre Processor is used forC:\Users\38099\OneDrive\Documents\PlatformIO\Projects\lvgl.pio\libdeps\pico\Adafruit TinyUSB Library\src\arduino\Adafruit_USBD_Device.h, advanced has  failed with Incorrect number of arguments to TU_CHECK_MCU'

maxgerhardt, 2023-04-29:
```
lib_deps = 
	adafruit/Adafruit TinyUSB Library@^2.1.0
```
^-- No. The core already has the right TinyUSB version built in -- remove that line, delete the `.pio` folder of the project and rery.

If still not working, we need to know with which stable version in the Arduino IDE you're using so you can use the exact same one in PlatformIO (right now it's using `5bf570c27d40cc2fc2b574661491ff4ccf37c2a´)

SocrateCrypto, 2023-04-30:
> ```
> lib_deps =
> 	adafruit/Adafruit TinyUSB Library@^2.1.0
> ```
>
> ^-- No. The core already has the right TinyUSB version built in -- remove that line, delete the `.pio` folder of the project and rery. [...]

"I did as you suggested, deleted the line adafruit/Adafruit TinyUSB Library@^2.1.0. Everything compiles, but the firmware doesn't work - the mouse doesn't output coordinates and binding to another port doesn't work. It still compiles and works as intended on Arduino IDE. Thank you for trying to help."   my Arduino IDE : 
 Version: 2.0.4-nightly-20230106
Date: 2023-01-06T03:06:40.103Z
CLI Version: 0.29.0 [76251df9]

Copyright © 2023 Arduino SA

SocrateCrypto, 2023-04-30:
I also tried through V Micro and it works too, Very strange behavior

maxgerhardt, 2023-04-30:
> my Arduino IDE :
>Version: 2.0.4-nightly-20230106
>Date: 2023-01-06T03:06:40.103Z
>CLI Version: 0.29.0 [76251df9]

No, I mean the version of the Arduino-Pico core installed in the Boards manager. Tools -> Boards -> Board Manager -> pico. Something like, version 3.1.1?

SocrateCrypto, 2023-04-30:
> > my Arduino IDE :
> > Version: 2.0.4-nightly-20230106
> > Date: 2023-01-06T03:06:40.103Z
> > CLI Version: 0.29.0 [76251df9]
> 
> No, I mean the version of the Arduino-Pico core installed in the Boards manager. Tools -> Boards -> Board Manager -> pico. Something like, version 3.1.1?

https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json

SocrateCrypto, 2023-04-30:
Using board 'rpipico' from platform in folder: C:\Users\38099\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.1.1
Using core 'rp2040' from platform in folder: C:\Users\38099\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.1.1
Detecting libraries used...
"C:\\Users\\38099\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.5.0-b-c7bab52/bin/arm-none-eabi-g++" -I "C:\\Users\\38099\\AppData\\Local\\Temp\\arduino-sketch-6DB3A557EC15AB075EC2FD43EC04B860/core" -c -Werror=return-type -DUSE_TINYUSB "-IC:\\Users\\38099\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\3.1.1/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipico\"" -DTARGET_RP2040 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -fstack-protector -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS "-iprefixC:\\Users\\38099\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\3.1.1/" "@C:\\Users\\38099\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\3.1.1/lib/platform_inc.txt" "-IC:\\Users\\38099\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\3.1.1/include" -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -O3 -DWIFICC=CYW43_COUNTRY_WORLDWIDE "-IC:\\Users\\38099\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\3.1.1\\cores\\rp2040" "-IC:\\Users\\38099\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\3.1.1\\variants\\rpipico" "C:\\Users\\38099\\AppData\\Local\\Temp\\arduino-sketch-6DB3A557EC15AB075EC2FD43EC04B860\\sketch\\sketch_apr19a.ino.cpp" -o nul
Alternatives for pio_usb.h: [Pico PIO USB@0.5.2]
ResolveLibrary(pio_usb.h)
  -> candidates: [Pico PIO USB@0.5.2]
"C:\\Users\\38099\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\1.5.0-b-c7bab52/bin/arm-none-eabi-g++" -I "C:\\Users\\38099\\AppData\\Local\\Temp\\arduino-sketch-6DB3A557EC15AB075EC2FD43EC04B860/core" -c -Werror=return-type -DUSE_TINYUSB "-IC:\\Users\\38099\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\3.1.1/libraries/Adafruit_TinyUSB_Arduino/src/arduino" -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico\"" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipico\"" -DTARGET_RP2040 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -fstack-protector -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS "-iprefixC:\\Users\\38099\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\3.1.1/" "@C:\\Users\\38099\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\3.1.1/lib/platform_inc.txt" "-IC:\\Users\\38099\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\3.1.1/include" -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO "-DBOARD_NAME=\"RASPBERRY_PI_PICO\"" -DARDUINO_ARCH_RP2040 -O3 -DWIFICC=CYW43_COUNTRY_WORLDWIDE "-IC:\\Users\\38099\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\3.1.1\\cores\\rp2040" "-IC:\\Users\\38099\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\3.1.1\\variants\\rpipico" "-Ic:\\Users\\38099\\OneDrive\\Documents\\Arduino\\libraries\\Pico_PIO_USB\\src" "C:\\Users\\38099\\AppData\\Local\\Temp\\arduino-sketch-6DB3A557EC15AB075EC2FD43EC04B860\\sketch\\sketch_apr19a.ino.cpp" -o nul
Alternatives for Adafruit_TinyUSB.h: [Adafruit TinyUSB Library@2.1.0 Adafruit TinyUSB Library@1.17.0]
ResolveLibrary(Adafruit_TinyUSB.h)
  -> candidates: [Adafruit TinyUSB Library@2.1.0 Adafruit TinyUSB Library@1.17.0]

maxgerhardt, 2023-04-30:
> Using core 'rp2040' from platform in folder: C:\Users\38099\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.1.1

Okay so it is 3.1.1.  Can you add this to your `platformio.ini`?
```ini
platform_packages =
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#3.1.1
``` 
(after saving the `platformio.ini`), the download of the new version **will take** a long time, wait until the update notifications disseappear).

SocrateCrypto, 2023-04-30:
> Okay so it is 3.1.1. Can you add this to your `platformio.ini`?
>
> ```ini
> platform_packages =
>    framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#3.1.1
> ```
>
> (after saving the [...]

thank you, I will try now".

SocrateCrypto, 2023-04-30:
> ```ini
> platform_packages =
>    framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#3.1.1
> ```

[env:pico]

platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages =
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#3.1.1
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
lib_archive = no
lib_deps = 
	
	

  
  sekigon-gonnoc/Pico PIO USB@

 
board_build.f_cpu = 120000000L
build_flags = -fstack-protector 

       

     -D PIO_FRAMEWORK_ARDUINO_ENABLE_RTTI

   -D USE_TINYUSB
   -D DEBUG_RP2040_CORE
   This ok?

maxgerhardt, 2023-04-30:
Are you sure you need `lib_deps =sekigon-gonnoc/Pico PIO USB@` for this sketch? I thought it would only use libraries that are already built-into this core?

If you can't reduce it down, keep it and try it.

SocrateCrypto, 2023-04-30:
> Are you sure you need `lib_deps =sekigon-gonnoc/Pico PIO USB@` for this sketch? I thought it would only use libraries that are already built-into this core?
> 
> If you can't reduce it down, keep it and try it.

yes, it's needed

// pio-usb is required for rp2040 host
#include "pio_usb.h"
#include "Adafruit_TinyUSB.h"

// Pin D+ for host, D- = D+ + 1
#ifndef PIN_USB_HOST_DP
#define PIN_USB_HOST_DP 5
#endif

// Pin for enabling Host VBUS. comment out if not used
// #ifndef PIN_5V_EN
// #define PIN_5V_EN        18
// #endif

for includ  #include "pio_usb.h"

SocrateCrypto, 2023-04-30:
> adafruit/Adafruit TinyUSB Library@

After many struggles with this configuration, it started working! Thank you very much! I don't know why, but it works."

[env:pico]

platform = https://github.com/maxgerhardt/platform-raspberrypi.git
platform_packages =
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#3.1.1
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
lib_archive = no
lib_deps = 
	
	

  
  sekigon-gonnoc/Pico PIO USB@
  adafruit/Adafruit TinyUSB Library@1.17.0
 
board_build.f_cpu = 120000000L
build_flags = -fstack-protector 

       

     -D PIO_FRAMEWORK_ARDUINO_ENABLE_RTTI

   -D USE_TINYUSB
   -D DEBUG_RP2040_CORE

closed 2023-04-30 by earlephilhower