earlephilhower/arduino-pico #1409: Reading internal temperature sensor on RPiPico (Analog 4) in Arduino/Platformio

SJegorovs, 2023-04-30:
Dear community,

I'm new with programming the Raspberry Pi Pico with the Arduino IDE 2.1.0 (Platformio planed later). Now I've the problem to read the internel temperature sensor of my pico. In MicroPython it has being working. How has an hint for me.

Best regards
Svens from Germany

maxgerhardt, 2023-04-30:
What code are you using in Micropython and what code are you using in the Arduino IDE?

earlephilhower, 2023-04-30:
This describes the temperature read call:  https://arduino-pico.readthedocs.io/en/latest/analog.html#float-analogreadtemp-float-vref-3-3f

closed 2023-04-30 by earlephilhower

SJegorovs, 2023-05-01:
In MycroPython I use sensor = machine.ADC(4) -> sensor.read_u16() with a conversion factor of 3.3 / 65535 and (temperatur = 27 - (voltage - 0.706) / 0.001721) to get the temperatur from the onboard sensor.

In Arduino IDE I've tried:
- analogReadTemp(3.3f) -> to get the tempeatur in °C
- adc_init(4); adc_read(); calculation of the received voltage like in MicroPython
But on compiling I get error: adc_init not declared in this scope.

maxgerhardt, 2023-05-01:
The first way should be right. You're observing differences? Mind that the reference is not accurate to begin with.

>This reading is not exceedingly accurate and of relatively low resolution, so it is not a replacement for an external temperature sensor in many cases.

SJegorovs, 2023-05-01:
With the 1st methode I get an temperature of 17.31 °C, wich seams to be to low in my office

maxgerhardt, 2023-05-01:
And in the exact same spot, can you reflash the Micropython on it, what does it read?

SJegorovs, 2023-05-01:
On my pi Pico I don't have a microphone