earlephilhower/arduino-pico #1410: RPi Pico HID Mouse and Keyboard libraries together

Ray-Zl, 2023-04-30:
Hi, 
I've used arduino-pico library sometime ago to create a macro keypad with rotary encoder as mouse wheel which is working fine.
I wanted to make some changes and now it's not working
I narrowed down the issue to this:
when I include ```keyboard.h``` the mouse function stops working
I'm using:
**Arduino IDE 2.1.0
arduino-pico 3.1.1
keyboard 1.0.4
mouse 1.0.1**
here is a simplified sketch with two push buttons attached to pin 4 and 5
as soon as uncommenting ```//#include <Keyboard.h>``` the mouse stops working
(the sketch works fine on Leonardo with both mouse and keyboard included)

```C
#include <Mouse.h>
//#include <Keyboard.h>

const int button1 = 4;
const int button2 = 5;

void setup()
{
  pinMode(button1  , INPUT_PULLUP);
  pinMode(button2  , INPUT_PULLUP);  
  Mouse.begin();
}

void loop()
{
  if (digitalRead(button1) == LOW) {
    Mouse.move(0,0,1);
    delay(100);
  }
  if (digitalRead(button2) == LOW) {
    Mouse.move(0,0,-1);
    delay(100);
  }
}
```

earlephilhower, 2023-05-01:
Thanks, can reproduce it here.  The HID code was refactored and it looks like something got broken in the process.

earlephilhower, 2023-05-02:
@Ray-Zl Can you please try #1418 and report back?

Ray-Zl, 2023-05-02:
> @Ray-Zl Can you please try #1418 and report back?

I tried it. All working fine.
Thanks for quick response.

commit ca7cd6a0 (2023-05-02): Fix USB Mouse/Joystick HID report ID

Fixes #1410

The USB Keyboard now has 2 reports (keyboard + consumer control), so when both
Keyboard and Mouse libraries are included, the Mouse must be report 3, not 2.

Update the Mouse and Joystick report IDs appropriately.
PR #1418 (2023-05-02): Fix USB Mouse/Joystick HID report ID
Fixes #1410

The USB Keyboard now has 2 reports (keyboard + consumer control), so when both Keyboard and Mouse libraries are included, the Mouse must be report 3, not 2.

Update the Mouse and Joystick report IDs appropriately.
commit 89fe754a (2023-05-02): Fix USB Mouse/Joystick HID report ID (#1418)

Fixes #1410

The USB Keyboard now has 2 reports (keyboard + consumer control), so when both
Keyboard and Mouse libraries are included, the Mouse must be report 3, not 2.

Update the Mouse and Joystick report IDs appropriately.
closed 2023-05-02 by earlephilhower