earlephilhower/arduino-pico #1417: Core 3.1.1 BT examples aren't compiling on a Mac

lu7did, 2023-05-01:
Hi,

Placed the same message as a follow on on other thread but perhaps it's better to keep it separated as I'm sure the problem lies on something I'm not doing and thus can be closed after discussing.

Installed the 3.1.1 latest version, selected Tools/IP Bluetooth stack as ip v4 and BT. Took the examples (iBeacon, iBeaconScanner, ANCS, etc) and none of them compiles. The sketch seems to be unable to find the proper libraries.

The following copy of the messages occurs with iBeacon.ino but similar errors occurs with any of the others:

/private/var/folders/4b/ywr8xsfj2bqcqw2593cpc_rc0000gp/T/.arduinoIDE-unsaved202341-24882-1nnra2o.i8dbg/iBeacon/iBeacon.ino:14:1: error: 'UUID' does not name a type
   14 | UUID uuid("E2C56DB5-DFFB-48D2-B060-D0F5A71096E0");
      | ^~~~
/private/var/folders/4b/ywr8xsfj2bqcqw2593cpc_rc0000gp/T/.arduinoIDE-unsaved202341-24882-1nnra2o.i8dbg/iBeacon/iBeacon.ino: In function 'void setup()':
/private/var/folders/4b/ywr8xsfj2bqcqw2593cpc_rc0000gp/T/.arduinoIDE-unsaved202341-24882-1nnra2o.i8dbg/iBeacon/iBeacon.ino:17:3: error: 'BTstack' was not declared in this scope
   17 |   BTstack.setup();
      |   ^~~~~~~
/private/var/folders/4b/ywr8xsfj2bqcqw2593cpc_rc0000gp/T/.arduinoIDE-unsaved202341-24882-1nnra2o.i8dbg/iBeacon/iBeacon.ino:18:29: error: 'uuid' was not declared in this scope
   18 |   BTstack.iBeaconConfigure(&uuid, 4711, 2);
      |                             ^~~~
/private/var/folders/4b/ywr8xsfj2bqcqw2593cpc_rc0000gp/T/.arduinoIDE-unsaved202341-24882-1nnra2o.i8dbg/iBeacon/iBeacon.ino: In function 'void loop()':
/private/var/folders/4b/ywr8xsfj2bqcqw2593cpc_rc0000gp/T/.arduinoIDE-unsaved202341-24882-1nnra2o.i8dbg/iBeacon/iBeacon.ino:24:3: error: 'BTstack' was not declared in this scope
   24 |   BTstack.loop();
      |   ^~~~~~~

exit status 1

Compilation error: 'UUID' does not name a type

[image]

lu7did, 2023-05-01:
Just to document the core level I'm using

[image]

lu7did, 2023-05-01:
[image]

earlephilhower, 2023-05-01:
This was fixed in #1360.  The problem was that Mac and Windows aren't case-sensitive for filenames, and there is a both a `BTstack.h` and a `btstack.h` file in the path.  Under Linux or Windows + Case Sensitive NTFS this is not a problem, but otherwise you can end up including the wrong file and missing the library's definition.

Please use `git` master for now, until a new release is published.  Probably later this week since some minor fixes have come in.

closed 2023-05-01 by earlephilhower