earlephilhower/arduino-pico #1422: Waveshare RP2040-Plus 4MB I2C Issues

Dameon87, 2023-05-03:
Have been having trouble with all of the waveshare boards. I'm attempting to access/use I2C on them all but only get Write Fail:4 messages. 
I am attempting this with the Radio library (Which works just fine on other rp2040 boards).

Si4703 breakout board connected to SDA=8, SCL=9, and Reset Pin = 1

The Code in question is:
`///
/// \file  TestSI4703.ino
/// \brief An Arduino sketch to operate a SI4703 chip based radio using the Radio library.

#include <Arduino.h>
#include <Wire.h>
#include <radio.h>
#include <SI4703.h>

#define RESET_PIN 1
#define MODE_PIN 8
// ----- Fixed settings here. -----

#define FIX_BAND RADIO_BAND_FM  ///< The band that will be tuned by this sketch is FM.
#define FIX_STATION 8930        ///< The station that will be tuned by this sketch is 89.30 MHz.
#define FIX_VOLUME 8            ///< The volume that will be set by this sketch is level 4.

SI4703 radio;  // Create an instance of Class for Si4703 Chip

/// Setup a FM only radio configuration
/// with some debugging on the Serial port
void setup() {
  // open the Serial port
  Serial.begin(115200);
  Serial.println("Radio::TestSI4703...");
  delay(200);

  radio.setup(RADIO_RESETPIN, RESET_PIN);
  radio.setup(RADIO_SDAPIN, MODE_PIN);

  // Enable information to the Serial port
  radio.debugEnable(true);
  radio._wireDebug(true);

  // Set FM Options for Europe
  radio.setup(RADIO_FMSPACING, RADIO_FMSPACING_100);   // for EUROPE
  radio.setup(RADIO_DEEMPHASIS, RADIO_DEEMPHASIS_50);  // for EUROPE

  // Initialize the Radio
  radio.initWire(Wire);

  // Set all radio setting to the fixed values.
  radio.setBandFrequency(FIX_BAND, FIX_STATION);
  radio.setVolume(FIX_VOLUME);
  radio.setMono(false);
  radio.setMute(false);

}  // setup

/// show the current chip data every 3 seconds.
void loop() {
  char s[12];
  radio.formatFrequency(s, sizeof(s));
  Serial.print("Station:");
  Serial.println(s);

  Serial.print("Radio:");
  radio.debugRadioInfo();

  Serial.print("Audio:");
  radio.debugAudioInfo();

  delay(3000);
}  // loop

// End.
`

maxgerhardt, 2023-05-03:
>Si4703 breakout board connected to SDA=8, SCL=9, and Reset Pin = 1 (This has also been tried with 4,5)

Good, this is correct for `Wire` per

https://github.com/earlephilhower/arduino-pico/blob/2c0ce6f416a1847a5f9687e32141589fe7ae11e6/variants/waveshare_rp2040_plus_4mb/pins_arduino.h#L60-L62

Did you run [the I2C Scanner sketch](https://playground.arduino.cc/Main/I2cScanner/) beforehand and was it able to find an active address?

Dameon87, 2023-05-03:
I did run the I2CScan test as well. It did not display any addresses at all.
The SI4703 requires setting the reset pin to high to trigger 2 wire mode. So I added that to the sketch as well. However, it simply doesn't return any addresses. 
In general, I've just not had much luck with these boards for some reason.

maxgerhardt, 2023-05-03:
If you have a logic probe (like a Salae clone) you can trace what's going on the reset and I2C pins and compare the Micropython vs Arduino-Pico behavior.

earlephilhower, 2023-05-10:
No feedback in a week.  Closing for now.

closed 2023-05-10 by earlephilhower