earlephilhower/arduino-pico #1425: Help needed with Wifi library..

abdalaosama, 2023-05-04:
Hey, I have been trying to run the ScanNetworks example on my new pico.
been getting compile errors.
it seems to me like I might be missing a setup step .. or that my enviroment is missing some library, hence the  "undefined refrence to" errors.. 

Here is the compiler output:
```cpp
/home/abdo/.arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/abdo/.arduino15/packages/rp2040/hardware/rp2040/3.1.1/lib/libpicow-noipv6-nobtc-noble.a(cyw43_ctrl.c.obj): in function `cyw43_ensure_up':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:164: undefined reference to `__wrap_cyw43_cb_tcpip_deinit'
/home/abdo/.arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:165: undefined reference to `__wrap_cyw43_cb_tcpip_deinit'
/home/abdo/.arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/abdo/.arduino15/packages/rp2040/hardware/rp2040/3.1.1/lib/libpicow-noipv6-nobtc-noble.a(cyw43_ctrl.c.obj): in function `cyw43_cb_process_async_event':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:429: undefined reference to `__wrap_cyw43_cb_tcpip_set_link_up'
/home/abdo/.arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:350: undefined reference to `__wrap_cyw43_cb_tcpip_set_link_down'
/home/abdo/.arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:405: undefined reference to `__wrap_cyw43_cb_tcpip_set_link_up'
/home/abdo/.arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:409: undefined reference to `__wrap_cyw43_cb_tcpip_set_link_down'
/home/abdo/.arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/abdo/.arduino15/packages/rp2040/hardware/rp2040/3.1.1/lib/libpicow-noipv6-nobtc-noble.a(cyw43_ctrl.c.obj): in function `cyw43_wifi_set_up':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:560: undefined reference to `__wrap_cyw43_cb_tcpip_deinit'
/home/abdo/.arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:561: undefined reference to `__wrap_cyw43_cb_tcpip_init'
/home/abdo/.arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/abdo/.arduino15/packages/rp2040/hardware/rp2040/3.1.1/lib/libpicow-noipv6-nobtc-noble.a(cyw43_ll.c.obj): in function `cyw43_ll_process_packets':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ll.c:1185: undefined reference to `__wrap_cyw43_cb_process_ethernet'
/home/abdo/.arduino15/packages/rp2040/tools/pqt-gcc/1.5.0-b-c7bab52/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: /home/abdo/.arduino15/packages/rp2040/hardware/rp2040/3.1.1/lib/libpicow-noipv6-nobtc-noble.a(cyw43_ll.c.obj): in function `cyw43_do_ioctl':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ll.c:1214: undefined reference to `__wrap_cyw43_cb_process_ethernet'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```
I have tried googling some of the errors but seems like no results for this, or the missing libs names.

any help or insights would be greatly appreciated.. thanks.

earlephilhower, 2023-05-04:
Looks like you haven't set the board.  Use the Tools->Board->Raspberry Pi Pico W and rebuild.

abdalaosama, 2023-05-05:
well, 
did as you advised worked like a charm..
I genuinely embarrassed, I actually had selected the Pi Pico (Without 'W') and it has been working for me on anything 
except wireless, sorry for wasting your time.. 

this field is still new to me and I am still learning... this has take more time than it should have..
once more, thank you :)