earlephilhower/arduino-pico #1430: SD support broken.

tbillion, 2023-05-08:
I have tried the base examples and any other example based on reading or writing to the sd card all are broken and fault with the same error:

```
In file included from C:\Users\LocalAdmin\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.1.1\libraries\SDFS\src/SDFS.h:35,
                 from C:\Users\LocalAdmin\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.1.1\libraries\SD\src/SD.h:24,
                 from C:\Users\LocalAdmin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:87:
C:\Users\LocalAdmin\Documents\Arduino\libraries\SdFat\src/SdFat.h:448:2: warning: #warning File not defined because __has_include(FS.h) [-Wcpp]
  448 | #warning File not defined because __has_include(FS.h)
      |  ^~~~~~~
In file included from C:\Users\LocalAdmin\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.1.1\libraries\SD\src/SD.h:24,
                 from C:\Users\LocalAdmin\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_tft_espi.cpp:87:
C:\Users\LocalAdmin\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.1.1\libraries\SDFS\src/SDFS.h: In member function 'virtual int sdfs::SDFSFileImpl::availableForWrite()':
C:\Users\LocalAdmin\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.1.1\libraries\SDFS\src/SDFS.h:279:31: error: 'using element_type = class File32' {aka 'class File32'} has no member named 'availableSpaceForWrite'; did you mean 'availableForWrite'?
  279 |         return _opened ? _fd->availableSpaceForWrite() : 0;
      |                               ^~~~~~~~~~~~~~~~~~~~~~
      |                               availableForWrite

```

before i bothered anyone here the AI recommended this , but im not sure of its solution:

```
The error message you are getting is telling you that there is no member function called availableSpaceForWrite() in the File32 class. This is likely because the SDFSFileImpl class is trying to call this function, but it is not defined.

To fix this error, you will need to either define the availableSpaceForWrite() function in the File32 class, or you will need to change the code in the SDFSFileImpl class so that it does not call this function.

Here is an example of how you could define the availableSpaceForWrite() function in the File32 class:
C++

class File32 {
  public:
    int availableSpaceForWrite() {
      // This function returns the number of bytes that are available to be written to the file.
      return _fileSize - _filePosition;
    }
  private:
    int _fileSize;
    int _filePosition;
};

Use code with caution. [Learn more](https://bard.google.com/faq#coding)

Once you have defined this function, you can change the code in the SDFSFileImpl class so that it calls this function instead of the undefined availableSpaceForWrite() function.
C++

class SDFSFileImpl {
  public:
    int availableForWrite() {
      return _opened ? _fd->availableSpaceForWrite() : 0;
    }
  private:
    bool _opened;
    File32* _fd;
};

Use code with caution. [Learn more](https://bard.google.com/faq#coding)

Once you have made these changes, the error should be resolved.
```

Guidance?

Thanks!

earlephilhower, 2023-05-08:
> C:\Users\LocalAdmin\Documents\Arduino\libraries\SdFat\src/SdFat.h:

This is the problem.  The IDE is using the wrong SDFat.

You need to remove the `SDFat` in your `Arduino/libraries` directory and use the one that comes distributed here in the repo itself.  That has both this add'l call as well as some fixes included that are not in upstream.

closed 2023-05-08 by earlephilhower

tbillion, 2023-05-08:
will this not break other installations that use SDfat on other devices? just a Q, because i don't want to break my other Dev environments within Arduino IDE.

maxgerhardt, 2023-05-08:
Arduino IDE's library management with respect to conflicting with built-in libraries is not the best :( Also hard to find a generic solution -- I think usually you want to prefer the core built-in libraries (or most cases), but in some cases you want an explicitly downloaded library to override it (because it's newer, has a bugfix, or whatever).