earlephilhower/arduino-pico #1433: print (stdio) linker errors.

OzmoOzmo, 2023-05-09:
Hi - I am having linker errors - reduced to its most basic - the code below fails to compile in both Platformio and Arduino 2.0 IDE
Using latest version from git: Arduino boards manager reports it as V3.2.0

#include <stdio.h> 
#include "pico/stdlib.h"  
int main() {  
  stdio_init_all(); //compiles if this is omitted - but then pico will not actually send anything over usb/uart  
  printf("Hello") ;  
}  

`Error:  
sketch_may9a/sketch_may9a.ino:5: undefined reference to 'stdio_init_all'  
collect2.exe: error: ld returned 1 exit status  
`

I hack/fixed the issue for myself by including these missing? files from Pico SDK 1.5 in my src folder... Im not sure if there is other code or libs missing - but this allows the above to compile and run and print to Terminal from the pico.

files: stdio_uart.h, stdio_usb.h, stdio.c, stdio_uart.c, stdio_usb.c, stdlib.c 

I need to use pico sdk - not arduino equivalents.
ps. Thanks for a great library.

earlephilhower, 2023-05-09:
You can't use `stdio_init_all`, the USB port is owned by `Serial` in the Arduino core.  Just use `Serial.printf()` or standard library `printf` and set the `Tools->Debug Port->Serial` to send all STDOUT to the USB (or UART) port.

closed 2023-05-09 by earlephilhower

OzmoOzmo, 2023-05-14:
"or standard library printf"

Thanks - but this just doesnt work unfortunately.
All the rpi pico sdk examples need this line of code .
Without the call to "stdio_init_all(); " printf does nothing. No compiler error - printf calls without issue - but nothing gets sent over usb serial - nothing sent on Pin 1&2.

When I call "stdio_init_all(); " it works - I can include it and it compiles - but will not link unless I add these files to my project stdio.c, stdio_uart.c, stdio_usb.c, stdlib.c
(I also have to build_flags = -D PICO_STDIO_USB -D PICO_STDIO_UART in platformio.ini)
I'm not using Arduino ide - but I tested and found same issue.

Then it all works. Its like these files were omitted by accident since the code compiles - 
- just will not link as headers are present - but c files missing.

I am following a Course from Cornell Uni - and they do not use Arduino - only official SDK calls - so "Serial" is out as that Arduino .
Issue is fixed by including the above files from official SDK.

This library/tool is the only one I found that does all 4 of:
-Pico Uploader using cmsis-dap
-Debugger using official rpi debug probe debugger
-Coding in RPI SDK
-All within Platformio/Visual Studio.

So thanks very much for a great tool and thanks very much for replying :)

earlephilhower, 2023-05-14:
You need to set the debug port in the `Tools->Debug Port->Serial` or (Serial1/2) because `::printf` is not Arduino standard (the `Serial`-type objects are used).

Under P.IO use `-DDEBUG_RP2040_PORT=Serial` (or Serial1/2 for UARTs).