earlephilhower/arduino-pico #1439: Deadlock with LittleFs

mc-takashi-sato, 2023-05-11:
### Discussed in https://github.com/earlephilhower/arduino-pico/discussions/1357

I've tried writing files with LittleFS using Release 3.2.0 and the fix below seems to lead to a different error.
->Fix USB crashes in FreeRTOS (#1419)

The MCVE used for operation confirmation is as follows.

```

#include <Arduino.h>
#include <FreeRTOS.h>
#include <task.h>
#include <LittleFS.h>
#include <queue.h>
#include "hardware/watchdog.h"

int cnt;

#define DEF_LOGGERTASK_QUE_MAX 20
#define DEF_LOGGERTASK_MSG_LEN 64
#define DEF_LOGGERTASK_STKSIZE 2048

typedef struct tagLoggerMsg
{
    unsigned long       m_dwMessage;                        
    int                 m_nLength;                          
    char                m_chBuff[DEF_LOGGERTASK_MSG_LEN];   
} LoggerMsg;

#define DEF_LOGGERTASK_BUFFSIZE sizeof(LoggerMsg)
#define INTERNAL_LOG_SIZE_MAX   (1024 * 5) 
#define INTERNAL_LOG_FILE_MAX   10 

enum LoggerEventId{
  LoggerEvent_UNKNOWN = -1,
  LoggerEvent_WRITE = 0,           
  LoggerEvent_MAX                  
} ;

class LoggerTaskClass
{
public:
  TaskHandle_t _chkTaskHandleLoggerTask ;
  QueueHandle_t _xLoggerTaskQueue;
  void begin();
private:
  // static void createTask(void *pvParameters) ;
  static void LoggerTaskMainLoop(void *pvParameters) ;
};

LoggerTaskClass LOGGER ;

/******************************************************************************/
// logger main loop
/******************************************************************************/
void LoggerTaskClass::LoggerTaskMainLoop(void *pvParameters)
{
  int filesize = 0;
  LoggerMsg RcvMsg = {0};
  char *buffer;
  File _lfs;
  int ret = 0;

  LittleFS.begin();
  while(1)
  {
    if (xQueueReceive(LOGGER._xLoggerTaskQueue, &RcvMsg, portMAX_DELAY ) == pdTRUE)
    {
      
      switch(RcvMsg.m_dwMessage)
      {
        case LoggerEvent_WRITE:           
          buffer = RcvMsg.m_chBuff;
          Serial.printf(buffer);

          _lfs = LittleFS.open("/logfile.log", "a");
          if(_lfs){
            ret = _lfs.write(RcvMsg.m_chBuff,RcvMsg.m_nLength);
          }
          _lfs.close();
          break;
        default:
          break;
      }
    }
    
  }
}

/******************************************************************************/
// logger begin api
/******************************************************************************/
void LoggerTaskClass::begin(void)
{
  LOGGER._xLoggerTaskQueue = xQueueCreate(DEF_LOGGERTASK_QUE_MAX, DEF_LOGGERTASK_BUFFSIZE);

  _chkTaskHandleLoggerTask = NULL;
  xTaskCreate(LoggerTaskMainLoop, "LoggerTask", DEF_LOGGERTASK_STKSIZE, this, 3, &_chkTaskHandleLoggerTask);
  if (!_chkTaskHandleLoggerTask)
  {
    ;
  }
}

/******************************************************************************/
// log que send
/******************************************************************************/
void SaveLogFile(const char *buffer,int len)
{
  LoggerMsg SndMsg ={0};
  if(NULL == LOGGER._xLoggerTaskQueue){
    return;
  }
  
  SndMsg.m_dwMessage = LoggerEvent_WRITE;
  
  SndMsg.m_nLength = len;
  memcpy(SndMsg.m_chBuff, buffer, len);

  if(xQueueSend(LOGGER._xLoggerTaskQueue, &SndMsg, 0) == pdFALSE)
  {
    ;
  }
}

/******************************************************************************/
// main setup
/******************************************************************************/
void setup() {
  Serial.begin(115200, true);
  delay(500);

#if !defined(DEBUG)
  watchdog_enable(3000, 1);  
#endif

  cnt = 0;

  LOGGER.begin();
}

/******************************************************************************/
// main loop
/******************************************************************************/
void loop() {
  char buffer[64] = {0};
  File _lfs;
  delay(1000);
  
  cnt++;

  sprintf(buffer,"(%ld)loopCount[%d]\r\n",millis(),cnt);
  SaveLogFile(buffer,strlen(buffer));

#if !defined(DEBUG)
  watchdog_update();
#endif
}

```

mc-takashi-sato, 2023-05-11:
As soon as I start it, only one line of log is displayed and the operation deadlocks.

The capture at that time was like below.

I would appreciate it if you could confirm

[image]

earlephilhower, 2023-05-12:
I can reproduce with this sketch.  The original MCVE passes (Serial related) no matter what I do, though, same as you reported.

mc-takashi-sato, 2023-05-12:
@earlephilhower 

thank you for making sure
However, it seems to be reproducible here...

One thing I know is that if I comment out the following, the phenomenon will not occur.
`board_build.filesystem_size = 0.5m`
-> Because Open of LittleFs fails, so it does not write. .

Can you show me your platform.ini?
My settings are below.

```

[platformio]
; default_envs = appli_release
default_envs = appli_debug

[env]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
; platform = raspberrypi
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
board_build.mcu = rp2040
board_build.f_cpu = 133000000L
build_flags = 
	-Wno-unknown-pragmas

[env:appli_release]
build_flags = 
	${env.build_flags}
	-Og
build_type = release
platform_packages = 
	${env.platform_packages}
upload_protocol = picoprobe
debug_tool = picoprobe
monitor_port = COM10
upload_speed = 115200

[env:appli_debug]
build_flags = 
	${env.build_flags}
	-D DEBUG=1
	-D DEBUG_PORT=0
build_type = debug
platform_packages = 
	${env.platform_packages}
upload_protocol = picoprobe
debug_tool = picoprobe
monitor_port = COM10
upload_speed = 115200
debug_init_break = tbreak setup

```

earlephilhower, 2023-05-12:
I don't use Platform.IO, but the original MCVE was related to the USB stack being re-entrantly called (a no-no) due to using an SDK mutex instead of a FreeRTOS mutex in the USB task.  The deadlock there was deep in TinyUSB.

I haven't looked into it yet, but here it looks like it's related to a global SDK mutex, and your trace has no TinyUSB shown.  So it's something different, possibly uncovered by the other fix letting things run longer.

earlephilhower, 2023-05-14:
Please give #1444 a try out.  It seems stable on this MCVE and the earlier one.

earlephilhower, 2023-05-16:
@mc-takashi-sato #1444 has undergone some refactoring, so please be sure to pull the latest changes for your testing.

mc-takashi-sato, 2023-05-16:
@earlephilhower 
I have tested it with the latest version and confirmed that there are no issues.
->cheetos (SHA-1: 28e200feb68aa67b8ea6d40a32e7bcdfa25918ed)

I also tried reading the serial simultaneously and confirmed that it functions properly without any deadlock.

This problem has been resolved. Thank you very much.

commit e8a155e7 (2023-05-14): Use real FreeRTOS tasks to idle core during flash

Fixes #1439

Use a real FreeRTOS task, at the highest priority, to idle the other core
while doing flash accesses.
commit ae908c8e (2023-05-16): Use real FreeRTOS tasks to idle core during flash (#1444)

Fixes #1439

Use a real FreeRTOS task, at the highest priority, to idle the other core
while doing flash accesses.

The USB stack seems to have some timing dependent bits which get broken
if FreeRTOS switches out the current task when it's running.  Avoid any
issue by disabling preemption on the core for all tud_task calls.

The PendSV handler disable flag can live completely inside the FreeRTOS
variant port, so remove any reference to it in the main core.

Tested using Multicode-FreeRTOS, #1402 and #1441

The USB FIFO interrupts were  [... cut]
closed 2023-05-16 by mc-takashi-sato